MichaIng/DietPi #5936: Installing MyMPD failed

jalsco, 2022-12-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Kernel version | `uname -a`
Linux batopi 5.15.76-rockchip64 #22.08.8 SMP PREEMPT Sun Oct 30 10:57:32 CET 2022 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
ROCK Pi 4 (aarch64)

#### Steps to reproduce

Tried to install myMPD from `dietpi-software` 

It fails and gives these messages : 
[image]

Command: usermod -g dietpi mympd                                                                                  │
[image]

                          │  - Exit code: 6                                                                                                      │
                          │  - DietPi version: v8.11.2 (MichaIng/master) | HW_MODEL: 72 | HW_ARCH: 3 | DISTRO: 6                                 │
                          │  - Image creator: DietPi Core Team                                                                                   │
                          │  - Pre-image: Armbian                                                                                                │
                          │  - Error log:                                                                                                        │
                          │ usermod: user 'mympd' does not exist

jalsco, 2022-12-01:
```
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  liblua5.4-0
The following NEW packages will be installed:
  liblua5.4-0 mympd
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 725 kB of archives.
After this operation, 1634 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 liblua5.4-0 arm64 5.4.2-2 [125 kB]
Get:2 https://download.opensuse.org/repositories/home:/jcorporation/Debian_11  mympd 10.1.3-1 [600 kB]
Fetched 725 kB in 10s (76.0 kB/s)
                                 Selecting previously unselected package liblua5.4-0:arm64.
(Reading database ... 56979 files and directories currently installed.)
Preparing to unpack .../liblua5.4-0_5.4.2-2_arm64.deb ...
Unpacking liblua5.4-0:arm64 (5.4.2-2) ...
Selecting previously unselected package mympd.
Preparing to unpack .../mympd_10.1.3-1_arm64.deb ...
Unpacking mympd (10.1.3-1) ...
Setting up liblua5.4-0:arm64 (5.4.2-2) ...
Setting up mympd (10.1.3-1) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+deb11u5) ...
[  OK  ] DietPi-Software | APT install for: mympd
[  OK  ] DietPi-Software | systemctl stop mympd
[FAILED] DietPi-Software | usermod -g dietpi mympd
[FAILED] DietPi-Software | usermod -g dietpi mympd
[FAILED] DietPi-Software | usermod -g dietpi mympd
[FAILED] DietPi-Software | usermod -g dietpi mympd

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Thu Dec  1 05:42:05 EST 2022
- DietPi version | v8.11.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCK Pi 4 (aarch64) (ID=72)
- Kernel version | `Linux batopi 5.15.76-rockchip64 #22.08.8 SMP PREEMPT Sun Oct 30 10:57:32 CET 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `usermod -g dietpi mympd`
- Exit code      | 6
- Software title | DietPi-Software
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
**usermod: user 'mympd' does not exist**
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
 - Press any key to return to DietPi-Launcher...
```

Joulinar, 2022-12-01:
probably an upstream issue. Usually, the user `mympd` should have been created by the `apt` package 🤔 

I see as well that config directories being moved. Probably we would need to adjust our install procedure.

jalsco, 2022-12-01:
Thanks @Joulinar  - just let me know if anything I can do to help 🙏🏻

Joulinar, 2022-12-01:
@MichaIng 
we would need to have a deeper look into MyMPD install process. Looks like existing configurations are not excepted currently. Strange behavior for an app to fail just because a config file exists. 🤔 Seems like a bug? 

```
Dec 01 14:33:19 DietPiR5S systemd[1]: Started myMPD server daemon.
Dec 01 14:33:19 DietPiR5S mympd[6091]: NOTICE   mympd     Cache dir: "/var/cache/mympd"
Dec 01 14:33:19 DietPiR5S mympd[6091]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Dec 01 14:33:19 DietPiR5S mympd[6091]: ERROR    mympd     Config dir: creating "/var/lib/mympd/config" failed
Dec 01 14:33:19 DietPiR5S mympd[6091]: ERROR    mympd     File exists
Dec 01 14:33:19 DietPiR5S mympd[6091]: Exiting erroneous, thank you for using myMPD
```

Reported https://github.com/jcorporation/myMPD/issues/912

MichaIng, 2022-12-01:
@Joulinar 
Yes this is already known, see my post and linked issue above 😉. A dynamic service user basically makes pre-creation of that directory impossible, so let's see how it's intended by Jürgen.

MichaIng, 2022-12-02:
PR up to solve this: #5944

It will work after next myMPD package release. Currently `mympd -c` will still fail, but after next release it won't. We do now use the dynamic service user to benefit from the theoretical security benefits.

jalsco, 2022-12-03:
> PR up to solve this: #5944
> 
> It will work after next myMPD package release. Currently `mympd -c` will still fail, but after next release it won't. We do now use the dynamic service user to benefit from the theoretical security benefits.

Thanks so much !!  You guys really rock 😁😁✨💖

closed 2022-12-04 by MichaIng

jcorporation, 2022-12-04:
myMPD v10.1.4 is out!

jalsco, 2022-12-05:
Thanks to @jcorporation , @MichaIng  and @Joulinar  for getting this updated so quickly ! 🙏🏻

I did : ` sudo apt update && sudo apt full-upgrade`
It did inidcate new version of `mympd`

**But I am still having problems to get `mympd` to run on my Dietpi 😢**

Maybe it is still half in and half out - because of previous failured attempts to install previous version of `mympd` ? Anyway , I have tried all the options of - uninstall install, reinstall... But I still have this status: 
```
[  OK  ] DietPi-Services | mpd                   active (running) since Sun 2022-12-04 20:11:22 EST; 2min 20s ago
[FAILED] DietPi-Services | ● mympd.service - myMPD server daemon
     Loaded: loaded (/lib/systemd/system/mympd.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2022-12-04 20:11:22 EST; 2min 20s ago
       Docs: man:mympd(1)
    Process: 871 ExecStart=/usr/bin/mympd (code=exited, status=1/FAILURE)
   Main PID: 871 (code=exited, status=1/FAILURE)
        CPU: 103ms

Dec 04 20:11:22 batopi mympd[871]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Dec 04 20:11:22 batopi mympd[871]: NOTICE   mympd     Config dir: "/var/lib/mympd/config"
Dec 04 20:11:22 batopi mympd[871]: NOTICE   mympd     Setting loglevel to NOTICE
Dec 04 20:11:22 batopi mympd[871]: NOTICE   mympd     Starting myMPD 10.1.4
Dec 04 20:11:22 batopi mympd[871]: 4d29 1 mongoose.c:4201:mg_open_listener bind: 98
Dec 04 20:11:22 batopi mympd[871]: 4d29 1 mongoose.c:3458:mg_listen        Failed: http://0.0.0.0:80, errno 98
Dec 04 20:11:22 batopi mympd[871]: EMERG    mympd     Can't bind to http://0.0.0.0:80
Dec 04 20:11:22 batopi mympd[871]: Exiting erroneous, thank you for using myMPD
Dec 04 20:11:22 batopi systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILURE
Dec 04 20:11:22 batopi systemd[1]: mympd.service: Failed with result 'exit-code'.
```

Note also - mympd - does NOT show in the - installed - list on `dietpi-software`, and installing via `dietpi-software` always fails with error message - as before

Joulinar, 2022-12-05:
Yes configuration directory would need to be removed before trying again

```
rm -Rf /var/lib/mympd
rm -Rf /var/lib/private/mympd
```

jalsco, 2022-12-05:
> Yes configuration directory would need to be removed before trying again
> 
> rm -Rf /var/lib/mympd
> rm -Rf /var/lib/private/mympd

I just did that - and then tried about to install using `dietpi-software` , but still errors

```
  - Command: usermod -g dietpi mympd   
  - Exit code: 6
  - DietPi version: v8.11.2 (MichaIng/master) | HW_MODEL: 72 | HW_ARCH: 3 | DISTRO: 6
  -  Image creator: DietPi Core Team 
  - Pre-image: Armbian
  -  Error log:
  -  usermod: user 'mympd' does not exist   
```

Joulinar, 2022-12-05:
probably wait un our upcoming Beta version. There we changed the install process according new mympd version.

jalsco, 2022-12-05:
ok - is that very close ?

Joulinar, 2022-12-05:
probably next weekend https://github.com/MichaIng/DietPi/issues/5898

In urgent case, you could switch to development branch already.

jalsco, 2022-12-05:
next week is good

MichaIng, 2022-12-05:
Just in case, dev branch can be applied via:
```sh
G_DEV_BRANCH dev
```
And to switch back to master, once it gets released:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master
```