MichaIng/DietPi #5941: No sound from 3.5mm jack on a RockPi4

jalsco, 2022-12-02:
Any assistance to help troubleshoot this - greatly appreciated 🙏🏻 

No sound via the 3.5mm jack. I have tested with a pair of ordinary headphones and also with powered speakers - just silence. 
```
DietPi v8.11.2
Device model : ROCK Pi 4 (aarch64)
[  OK  ] DietPi-Services | mpd                   active (running) 
[  OK  ] DietPi-Services | ympd                  active (running) 
```

```
[Fri Dec 02] root@batopi: ~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: hdmisound [hdmi-sound], device 0: ff8a0000.i2s-i2s-hifi i2s-hifi-0 [ff8a0000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
On port 1337 `ympd` starts up ✔, finds the folder of songs✔, and starts to play them✔...  except the room stays silent 🤣🤷‍♂️   `ympd` reports no problems, and keeps happily "playing" the songs.... 

In Launcher: 

```
   Sound card      : [hw:0,0-eq]                                                   
   Auto-conversion : [On] Automatic software conversions
```

And while `ympd` is playing those songs...  JustBoom even shows , via `ALSA Output Info` that the songs are really being played...
```
[ SUB1 ] DietPi-JustBoom > ALSA Output Info
access: MMAP_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 1024
buffer_size: 4096
```
I feel that I am very close - and probably something dumb and basic that I am failing to see...

What have I done wrong ?

MichaIng, 2022-12-02:
Many thanks for reporting.

The `aplay -l` output shows a HDMI audio device only, so 3.5mm jack cannot work. Attach an audio-capable HDMI screen and you'll hear the audio you're seeing in ALSA output info.

Which image/kernel version do you use? Basically it's always helpful if the bug report template is filled 😉:
```sh
uname -a
```

jalsco, 2022-12-03:
```
[Fri Dec 02] root@batopi: /$ uname -a
Linux batopi 5.15.76-rockchip64 #22.08.8 SMP PREEMPT Sun Oct 30 10:57:32 CET 2022 aarch64 GNU/Linux
```

jalsco, 2022-12-03:
Apparently it is common problem - with other distros (e.g.  Xubuntu and Armbian ) also reporting same that the audio hardware is not detected in Rockpi4.  But - apparently - the "offical Debian image" does contain the necessary driver : 

https://forum.radxa.com/t/rock-pi4b-audio-not-working/1737/2

If that is true - can the driver just be added into DietPi from Debian ?

MichaIng, 2022-12-03:
In a perfect world, Radxa would try to provide the driver or patch upstream to be merged into mainline kernel. But this is sadly usually not done, but it's OSS/Linux/Armbian kernel developers which try to backwards engineer or manually hack things, to get it working.

Btw, thanks for showing your disinterest for Igor's rant. Ancient topic. While he writes a lot of false or outdated things about DietPi, he is right that we do not contribute to their hard kernel development. The reason (from my end) is this welcoming atmosphere, a vicious cycle. And it's not only us, but most of their users who do bug reports but no significant donations face long posts about why it costs so much time/money and in case why their small donation does not cover it. IMO, accepting donations and blaming everyone how does not ~donate~ **pay**, isn't a healthy business model. How one person can be main driver and saboteur of the same project at the same time is sad. Watching this over years is one of the reasons I never tried to make DietPi a business or financially depend on it.

Back to topic. Here are the relevant device tree node and kernel driver:
- https://github.com/radxa/kernel/blob/release-4.4-rockpi4/arch/arm64/boot/dts/rockchip/rockpi-4-linux.dtsi#L155-L174
- https://github.com/radxa/kernel/blob/release-4.4-rockpi4/sound/soc/codecs/es8316.c

To compare with upstream Linux:
- https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4.dtsi#L54-L58 (here is a significant difference)
- https://github.com/torvalds/linux/blob/master/sound/soc/codecs/es8316.c

Ah, and here is the fix, it seems:
- https://github.com/armbian/build/commit/c3db31b
- https://github.com/armbian/build/pull/4433

Based on the version tag of this commit it might be already included with the recent kernel/dtb package upgrade. Can you try:
```sh
apt update
apt full-upgrade
```
In case this is Armbian v22.11.1 and Linux v5.15.80 already.

jalsco, 2022-12-03:
wow ! really...  I am amazed ! 
-------------
```
[Sat Dec 03] root@batopi: /mnt/pibax$ apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease
Get:3 https://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:4 https://dl.cloudsmith.io/public/caddy/stable/deb/debian any-version InRelease [7505 B]
Get:5 https://download.opensuse.org/repositories/home:/jcorporation/Debian_11  InRelease [1532 B]
Hit:6 https://deb.debian.org/debian-security bullseye-security InRelease
Get:7 https://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Hit:2 https://mirror-us-stl1.armbian.airframes.io/apt bullseye InRelease
Get:8 https://deb.debian.org/debian bullseye-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:9 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-12-03-0803.53-F-2022-12-03-0803.53.pdiff [609 B]
Get:9 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-12-03-0803.53-F-2022-12-03-0803.53.pdiff [609 B]
Fetched 166 kB in 7s (23.8 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
5 packages can be upgraded. Run 'apt list --upgradable' to see them.
[Sat Dec 03] root@batopi: /mnt/pibax$ apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  armbian-firmware linux-dtb-current-rockchip64 linux-image-current-rockchip64 linux-libc-dev linux-u-boot-rockpi-4b-current
5 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 64.7 MB of archives.
After this operation, 316 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 https://armbian.chi.auroradev.org/apt bullseye/main arm64 armbian-firmware all 22.11.1 [9442 kB]
Get:2 https://mirror-us-stl1.armbian.airframes.io/apt bullseye/main arm64 linux-dtb-current-rockchip64 arm64 22.11.1 [383 kB]
Get:4 https://armbian.tnahosting.net/apt bullseye/main arm64 linux-libc-dev arm64 22.11.1 [1247 kB]
Get:5 https://mirror-us-stl1.armbian.airframes.io/apt bullseye/main arm64 linux-u-boot-rockpi-4b-current arm64 22.11.1 [471 kB]
Get:3 https://armbian.chi.auroradev.org/apt bullseye/main arm64 linux-image-current-rockchip64 arm64 22.11.1 [53.2 MB]
Fetched 64.7 MB in 15s (4424 kB/s)
(Reading database ... 57366 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_22.11.1_all.deb ...
Unpacking armbian-firmware (22.11.1) over (22.08.6) ...
Preparing to unpack .../linux-dtb-current-rockchip64_22.11.1_arm64.deb ...
Unpacking linux-dtb-current-rockchip64 (22.11.1) over (22.08.8) ...
dpkg: warning: unable to delete old directory '/boot/dtb-5.15.76-rockchip64/rockchip': Directory not empty
dpkg: warning: unable to delete old directory '/boot/dtb-5.15.76-rockchip64': Directory not empty
Preparing to unpack .../linux-image-current-rockchip64_22.11.1_arm64.deb ...
Unpacking linux-image-current-rockchip64 (22.11.1) over (22.08.8) ...
Removing obsolete initramfs images
Preparing to unpack .../linux-libc-dev_22.11.1_arm64.deb ...
Unpacking linux-libc-dev:arm64 (22.11.1) over (22.08.6) ...
Preparing to unpack .../linux-u-boot-rockpi-4b-current_22.11.1_arm64.deb ...
Unpacking linux-u-boot-rockpi-4b-current (22.11.1) over (22.08.6) ...
Setting up linux-image-current-rockchip64 (22.11.1) ...
Removing obsolete initramfs images
removed '/boot/uInitrd-5.15.76-rockchip64'
removed '/boot/initrd.img-5.15.76-rockchip64'
update-initramfs: Generating /boot/initrd.img-5.15.80-rockchip64
update-initramfs: Converting to U-Boot format
Image Name:   uInitrd
Created:      Sat Dec  3 07:38:56 2022
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    11339189 Bytes = 11073.43 KiB = 10.81 MiB
Load Address: 00000000
Entry Point:  00000000
'/boot/uInitrd' -> 'uInitrd-5.15.80-rockchip64'
Setting up linux-u-boot-rockpi-4b-current (22.11.1) ...
Setting up linux-libc-dev:arm64 (22.11.1) ...
Setting up linux-dtb-current-rockchip64 (22.11.1) ...
Setting up armbian-firmware (22.11.1) ...
```
Then a reboot

Then - yes - see it 

```
usb-dac : USB Audio DAC (any)                                                          │
hw:0,0  : Analog ff880000.i2s-ES8316 HiFi ES8316 HiFi-0                                │
hw:1,0  : hdmisound ff8a0000.i2s-i2s-hifi i2s-hifi-0
```
And selected it - the analog option...

The EQ seems more active now.  **But I am still not hearing anything**.  I am still checking 
Seems very very close now .

jalsco, 2022-12-03:
Um - I had rebuilt some of the those  raxda DTB files - to get the 2x speed in the NVMe disk... I will check that later - maybe the update has overwritten what I did there. I will check it later - just still trying on the sound

MichaIng, 2022-12-03:
> maybe the update has overwritten what I did there.

Always do such as device tree overlays, like [here](https://github.com/MichaIng/DietPi/issues/5931#issuecomment-1335996007) done for enabling USB gadget mode on Radxa Zero.

Depending on how you try to play audio (the audio format), the auto-conversion plugin might be required. Also check `alsamixer` for mute flags or low volume sliders.

jalsco, 2022-12-03:
I am not worried about those dtb things - I can think about that later... 

Very strange - I even power off rebooted... Possibly it is my player that is not working. I am using `ympd` the volume slider keeps going to zero.  But I am confident I have tried it ... As u know I cannot install `mympd` because the problem installing.  I would like to try with a different player.  I already enabled "autoconvert". The files I am trying to play are mp3.  I am checking the `alsamixer` again..  Lots of pretty colors on the screen.....  Maybe it will work after a night's sleep. I am pretty tired.

jalsco, 2022-12-03:
I found some errors about ALSA - not sure how to interpret meaning ? 
```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing ALSA: Advanced Linux Sound Architecture

[ SUB1 ] DietPi-Set_hardware > soundcard (hw:0,0-eq)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
[  OK  ] DietPi-Set_hardware | rm /etc/asound.conf
[  OK  ] DietPi-Set_hardware | rm /var/lib/alsa/asound.state
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "Analog" "" "" "" ""
Hardware is initialized using a generic method
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "simple-card" "" "" "" ""
Hardware is initialized using a generic method
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: CONFIG_SOUNDCARD=hw:0,0-eq
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: libasound2-plugin-equal
[  OK  ] DietPi-Set_hardware | chmod 0775 /var/lib/dietpi/dietpi-config
[  OK  ] DietPi-Set_hardware | chown dietpi:dietpi /var/lib/dietpi/dietpi-config
[  OK  ] DietPi-Set_hardware | alsactl -g store
[  OK  ] soundcard hw:0,0 | Completed
```
Is anything there significant ?

MichaIng, 2022-12-03:
Those are not really errors. There are no default configs for this sound card shipped by ALSA, so it initialises configs in a generic way. This is also the reason why `alsamixer` changes may be required to make it work. We can set this as part of `dietpi-config` < `dietpi-set_hardware`, in case.

For reference, similar bug report on Debian with links to ALSA project GitHub repo: https://bugs.debian.org/994179

jalsco, 2022-12-03:
I will fell better about it when I test again using a different playing client - I think `ympd` is strange and may be the problem. If ever `ympd's` Browser-tab loses focus, its volume slider goes to zero. I would be more confident testing with a different player... I honestly cannot see any reason that music is not filling the room here.... 🤷🏻‍♂️ 

[image]

MichaIng, 2022-12-03:
The first volume could be higher. Often it's not linear but lower values cannot be heard at all. To rule out a player issue:
```sh
speaker-test
```

jalsco, 2022-12-03:
[image]

It is 100 now -- one tap on down arrow takes it to 56.

MichaIng, 2022-12-03:
Did you also try to toggle other controls, like the `Playback`. A full list of mixer control can be printed via:
```sh
amixer scontents
```

jalsco, 2022-12-03:
```
[Sat Dec 03] root@batopi: ~$ amixer scontents
Simple mixer control 'Headphone',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 3
  Mono:
  Front Left: Playback 3 [100%] [0.00dB]
  Front Right: Playback 3 [100%] [0.00dB]
Simple mixer control 'Headphone Mixer',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 11
  Front Left: 9 [82%] [-3.00dB]
  Front Right: 9 [82%] [-3.00dB]
Simple mixer control 'Mic Boost',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Playback Polarity',0
  Capabilities: enum
  Items: 'Normal' 'R Invert' 'L Invert' 'L + R Invert'
  Item0: 'Normal'
Simple mixer control 'Capture Polarity',0
  Capabilities: enum
  Items: 'Normal' 'Invert'
  Item0: 'Normal'
Simple mixer control 'ADC',0
  Capabilities: cvolume cvolume-joined
  Capture channels: Mono
  Limits: Capture 0 - 192
  Mono: Capture 192 [100%] [0.00dB]
Simple mixer control 'ADC Double Fs',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'ADC PGA Gain',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 10
  Mono: 9 [90%] [21.00dB]
Simple mixer control 'ADC Soft Ramp',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'ALC',0
  Capabilities: cswitch cswitch-joined
  Capture channels: Mono
  Mono: Capture [off]
Simple mixer control 'ALC Capture Attack Time',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 10
  Mono: 7 [70%]
Simple mixer control 'ALC Capture Decay Time',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 10
  Mono: 7 [70%]
Simple mixer control 'ALC Capture Hold Time',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 10
  Mono: 7 [70%]
Simple mixer control 'ALC Capture Max',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 28
  Mono: 25 [89%] [31.00dB]
Simple mixer control 'ALC Capture Min',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 28
  Mono: 25 [89%] [25.50dB]
Simple mixer control 'ALC Capture Noise Gate',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'ALC Capture Noise Gate Threshold',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 31
  Mono: 24 [77%]
Simple mixer control 'ALC Capture Noise Gate Type',0
  Capabilities: enum
  Items: 'Constant PGA Gain' 'Mute ADC Output'
  Item0: 'Constant PGA Gain'
Simple mixer control 'ALC Capture Target',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 10
  Mono: 8 [80%] [-4.50dB]
Simple mixer control 'DAC',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 192
  Mono:
  Front Left: Playback 152 [79%] [-20.00dB]
  Front Right: Playback 152 [79%] [-20.00dB]
Simple mixer control 'DAC Double Fs',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'DAC Mono Mix',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'DAC Notch Filter',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'DAC Soft Ramp',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'DAC Soft Ramp Rate',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 4
  Mono: 3 [75%]
Simple mixer control 'DAC Source Mux',0
  Capabilities: enum
  Items: 'LDATA TO LDAC, RDATA TO RDAC' 'LDATA TO LDAC, LDATA TO RDAC' 'RDATA TO LDAC, RDATA TO RDAC' 'RDATA TO LDAC, LDATA TO RDAC'
  Item0: 'LDATA TO LDAC, RDATA TO RDAC'
Simple mixer control 'DAC Stereo Enhancement',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 7
  Mono: 6 [86%]
Simple mixer control 'Differential Mux',0
  Capabilities: enum
  Items: 'lin1-rin1' 'lin2-rin2' 'lin1-rin1 with 20db Boost' 'lin2-rin2 with 20db Boost'
  Item0: 'lin1-rin1'
Simple mixer control 'Digital Mic Mux',0
  Capabilities: enum
  Items: 'dmic disable' 'dmic data at high level' 'dmic data at low level'
  Item0: 'dmic disable'
Simple mixer control 'Left Headphone Mixer LLIN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Left Headphone Mixer Left DAC',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Left Headphone Mux',0
  Capabilities: enum
  Items: 'lin1-rin1' 'lin2-rin2' 'lin-rin with Boost' 'lin-rin with Boost and PGA'
  Item0: 'lin1-rin1'
Simple mixer control 'Right Headphone Mixer RLIN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Right Headphone Mixer Right DAC',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Right Headphone Mux',0
  Capabilities: enum
  Items: 'lin1-rin1' 'lin2-rin2' 'lin-rin with Boost' 'lin-rin with Boost and PGA'
  Item0: 'lin1-rin1'
```

MichaIng, 2022-12-03:
Can you try to enable one-by-one the following toggles in `alsamixer`:
- ADC Double Fs
- DAC Mono Mix
- DAC Notch Filter
- Left Headphone Mixer LLIN
- Left Headphone Mixer Left DAC
- Right Headphone Mixer RLIN
- Right Headphone Mixer Right DAC

I do only partly understand what is what, some are related to digital-analogue/analogue-digital converter, others for analogue line-in. Correct should be the DAC ones, the `Left Headphone Mixer Left DAC` and `Right Headphone Mixer Right DAC` sound most promising, but I lack knowledge and insights about how all is internally connected or whether these mixers are required at all.

jalsco, 2022-12-04:
ok thanks - I think that I understand even less that you what all those items are doing !
I will try doing that (enable one-by-one) a bit later today. Thanks all your time on this  🙏🏻

jalsco, 2022-12-04:
@MichaIng you are a freaking GENIUS ! 💖💖💖💖💪😊😊😊🌟

MichaIng, 2022-12-04:
Great. I'll add a dedicated 3.5mm onboard audio selector for ROCK Pi 4 which sets these two automatically.

When you know the control names, `amixer` is indeed great, and mandatory for scripts. `alsamixer` is good to get an overview of available controls, and some might like the immediate graphical feedback.

> Also - when using `ympd` with `alsamixer` the volume controls in `ympd` always go to zero

Interesting. Does the volume control in `ympd` work? Probably it uses a false ALSA control.

jalsco, 2022-12-04:
volume in `ympd` does not work at all - on my system.  
But I only have it temporarily, until `mympd` is fixed and able to be installed 😎 

thx again 😊

MichaIng, 2022-12-04:
Will be interesting if it works in myMPD, as I can imagine that it's MPD itself. I guess they use some generic set of control names for the volume, which does not imply "DAC" or "Headphone". Are those actually both effective? I see the headphones control has 4 steps (0-3) only, which is not great, but the "DAC" control can be very fine grained with 193 steps (0 - 192).

jalsco, 2022-12-04:
Am just using "DAC"  for volume - it works either 0-192 , or as a percentage %.

The headphone master volume control - does work on volume inside `alsamixer` . It probably should work also with the commandline using `amixer`,  but I could not find the exact syntax to check it ...  `amixer` command not fully intuitive. .. and `DAC` is quicker to type anyway  😁

```
[Sun Dec 04] root@batopi: ~$ amixer set headphone 50%
amixer: Unable to find simple control 'headphone',0

[Sun Dec 04] root@batopi: ~$ amixer set headphones 50%
amixer: Unable to find simple control 'headphones',0

[Sun Dec 04] root@batopi: ~$ amixer set DAC 87%
Simple mixer control 'DAC',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 192
  Mono:
  Front Left: Playback 167 [87%] [-12.50dB]
  Front Right: Playback 167 [87%] [-12.50dB]
[Sun Dec 04] root@batopi: ~$
```

MichaIng, 2022-12-04:
In my case, both, myMPD and ympd use the "PCM" control as volume slider, while there is a "Master" control as well. So this is all done by MPD itself.

Try to respect capital letters:
```sh
amixer set Headphones 50%
```
With
```sh
amixer scontrols
```
you get the full list of control names. In `alsamixer` all names above 9 characters are truncated.

jalsco, 2022-12-04:
```
[Sun Dec 04] root@batopi: ~$ amixer set Headphones 50%
amixer: Unable to find simple control 'Headphones',0

[Sun Dec 04] root@batopi: ~$
```

jalsco, 2022-12-04:
```
[Sun Dec 04] root@batopi: ~$ amixer scontrols
Simple mixer control 'Headphone',0
Simple mixer control 'Headphone Mixer',0
Simple mixer control 'Mic Boost',0
Simple mixer control 'Playback Polarity',0
Simple mixer control 'Capture Polarity',0
Simple mixer control 'ADC',0
Simple mixer control 'ADC Double Fs',0
Simple mixer control 'ADC PGA Gain',0
Simple mixer control 'ADC Soft Ramp',0
Simple mixer control 'ALC',0
Simple mixer control 'ALC Capture Attack Time',0
Simple mixer control 'ALC Capture Decay Time',0
Simple mixer control 'ALC Capture Hold Time',0
Simple mixer control 'ALC Capture Max',0
Simple mixer control 'ALC Capture Min',0
Simple mixer control 'ALC Capture Noise Gate',0
Simple mixer control 'ALC Capture Noise Gate Threshold',0
Simple mixer control 'ALC Capture Noise Gate Type',0
Simple mixer control 'ALC Capture Target',0
Simple mixer control 'DAC',0
Simple mixer control 'DAC Double Fs',0
Simple mixer control 'DAC Mono Mix',0
Simple mixer control 'DAC Notch Filter',0
Simple mixer control 'DAC Soft Ramp',0
Simple mixer control 'DAC Soft Ramp Rate',0
Simple mixer control 'DAC Source Mux',0
Simple mixer control 'DAC Stereo Enhancement',0
Simple mixer control 'Differential Mux',0
Simple mixer control 'Digital Mic Mux',0
Simple mixer control 'Left Headphone Mixer LLIN',0
Simple mixer control 'Left Headphone Mixer Left DAC',0
Simple mixer control 'Left Headphone Mux',0
Simple mixer control 'Right Headphone Mixer RLIN',0
Simple mixer control 'Right Headphone Mixer Right DAC',0
Simple mixer control 'Right Headphone Mux',0
[Sun Dec 04] root@batopi: ~$ amixer set Headphone 50%
Simple mixer control 'Headphone',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 3
  Mono:
  Front Left: Playback 2 [67%] [-12.00dB]
  Front Right: Playback 2 [67%] [-12.00dB]
[Sun Dec 04] root@batopi: ~$
```

jalsco, 2022-12-04:
Is there some obvious command to change it from `mono` ?

MichaIng, 2022-12-04:
You mean it plays stereo sources in mono currently?

jalsco, 2022-12-04:
I cannot be sure. I have only attached ONE single speaker to it (not headphones) . I am just going by the responses from `amixer` that specifically say " `Mono` " .. It seems to say that it is playing in "mono" ... It is not important for me at this time.. But in the future, if ever I attached it to a pair of (L-R) speakers, or to headphones... then it would be nice to hear stereo. .  

[image]

MichaIng, 2022-12-04:
The relevant toggles look like it's stereo:
```
Simple mixer control 'Headphone',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 3
  Mono:
  Front Left: Playback 3 [100%] [0.00dB]
  Front Right: Playback 3 [100%] [0.00dB]
Simple mixer control 'DAC',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 192
  Mono:
  Front Left: Playback 152 [79%] [-20.00dB]
  Front Right: Playback 152 [79%] [-20.00dB]
Left Headphone Mixer Left DAC
Right Headphone Mixer Right DAC
```
It's other controls, the captures and analogue-digital converter, which have mono volumes/toggles defined. Could be tested, but I think it's fine.

MichaIng, 2022-12-04:
The two controls to enable 3.5mm output are not set by default on ROCK Pi 4: https://github.com/MichaIng/DietPi/commit/e616453

closed 2022-12-04 by MichaIng

wahono77, 2023-10-26:
This problem also in ROCK 3A with the latest Dietpi (8.23.3)

```
aplay -l

**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDMI], device 0: fe400000.i2s-i2s-hifi i2s-hifi-0 [fe400000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```
Any solution for Rock 3A? Thanks in advance

MichaIng, 2023-10-26:
@wahono77 
Could you please open a new issue about this, and fill the bug report template?