MichaIng/DietPi #5945: Tried to update PIhole and got this....

blindtrust, 2022-12-02:
#### Details:
- Date           | Fri Dec  2 14:24:58 EST 2022
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync

#### Steps to reproduce:
 typed DietPi-Update and got an error instantly, see below. related to timesync???

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd
[FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd

if you could tell me how to fix that would be great!
thanks.

#### Expected behaviour:
it should have updated to new verion of dietpi
- ...
#### Actual behaviour:
error
- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```
---------------------------------------------------------------------

Joulinar, 2022-12-02:
Did you have done a Debian destro update before?

You can install the missing time sync service as follows

`apt install systemd-timesyncd`

blindtrust, 2022-12-02:
That seemed to do the trick. thanks.

Scott Seifert
TEAMSPIKE
216-338-1486   Mobile

[TeamSpike Signture.jpg]

Joulinar, 2022-12-02:
Going to close the issue. Feel free to reopen if needed.

closed 2022-12-02 by Joulinar