MichaIng/DietPi #5946: How to know if there is an update for individual optimized software?

lutfor-diu, 2022-12-03:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
How to know if there is an update for individual optimized software?
- ...

#### Describe the solution you'd like:
How to know if there is an update for individual optimized software?
- ...

#### Describe alternatives you've considered:
none
- ...

#### Additional context
example: how to know if here is a new update of qbittorent? It would be great if there is list we can check and find which software recently updated.

Joulinar, 2022-12-03:
Usually in many UI there you will be informed about individual updates available. However for us this is nearly impossible to host a site with all versions and updates. He support over 200 software title and it's simply out of our score.

MichaIng, 2022-12-03:
In case of qBittorrent any many others:
```sh
apt update
```
By default this is done on a daily basis and should show you in the login banner when updates are available for packages. Other software have internal updaters or update checkers, and again others are packaged by us, in which case the DietPi release notes contain an info about updates and how to apply, usually also a prompt is shown on DietPi update. But indeed there are others where we simply download a latest release, checked via GitHub releases API.

It would be a too huge development and maintenance effort to code update checks for all those into DietPi scripts, but it would be possible to add a note to our docs how to check for updates manually. E.g. [here](https://dietpi.com/docs/software/cloud/#file-browser) for File Browser, adding an info to the "Update to latest version" tab that its GitHub releases page can be checked for available updates: https://github.com/filebrowser/filebrowser/releases

lutfor-diu, 2022-12-05:
> In case of qBittorrent any many others:
> 
> ```shell
> apt update
> ```
> 
> By default this is done on a daily basis and should show you in the login banner when updates are available for packages. Other software have internal updaters or update checkers, and again others are packaged by us, in which case the DietPi release notes contain an info about updates and how to apply, usually also a prompt is shown on DietPi update. But indeed there are others where we simply download a latest release, checked via GitHub releases API.
> 
> It would be a too huge development and maintenance effort to code update checks for all those into DietPi scripts, but it would be possible to add a note to our docs how to check for updates manually. E.g. [here](https://dietpi.com/docs/software/cloud/#file-browser) for File Browser, adding an info to the "Update to latest version" tab that its GitHub releases page can be checked for available updates: https://github.com/filebrowser/filebrowser/releases

thank you so much for details answer. i got it. :)

closed 2022-12-05 by lutfor-diu