MichaIng/DietPi #5949: Significant audio distortion with USB DAC

Kreeblah, 2022-12-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.11.2
- Distro version | bullseye
- Kernel version | Linux DietPi 5.19.17-meson64 #22.11.1 SMP PREEMPT Wed Nov 30 11:05:42 UTC 2022 aarch64 GNU/Linux
- SBC model | Odroid N2 (aarch64) - Actuallny an N2+, if it matters
- Power supply used | 12V/2A from AmeriDroid
- SD card used | 8GB eMMC

#### Additional Information (if applicable)
- Software title | Roon Bridge
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes.

- Bug report ID | 89a3073f-311b-421b-8bd3-162dc9172e46
- USB DAC | Hidizs S9 Pro

#### Steps to reproduce

1. Connect the USB DAC
2. Set ALSA to use either "any USB DAC" or the specific hardware ID for the USB DAC (both yield the same results)
3. Play something from Roon

#### Expected behaviour

- The audio should be clear

#### Actual behaviour

- There's significant audio distortion

#### Extra details

- This same hardware setup works fine with the official [ODROID minimal images](https://wiki.odroid.com/odroid-n2/os_images/ubuntu).  I don't experience audio distortion with those.

MichaIng, 2022-12-04:
Many thanks for your report.

Probably a limitation of the mainline kernel, but I didn't hear of this until now.

3.5mm jack and HDMI audio work fine? And you did try all USB ports, whether it makes a difference?

I can test it when I find time, having N2+ and a (cheap) USB DAC here as well.

Kreeblah, 2022-12-05:
I did some more testing based on your question, and I have a more specific condition for this to happen.  While I don't have a good way of testing the HDMI audio (so I don't know whether there's an issue there or not), I was able to test the internal 3.5mm output, and it worked fine.  So did my USB DAC afterwards.

I was able to replicate this on another fresh install, too.  It doesn't matter which USB port I have the DAC plugged into.  It sounds extremely distorted until I first use `dietpi-config` to switch to the 3.5mm output and then back to the DAC.  I don't even have to play anything through the 3.5mm jack.  Just switch to it.  After that, my DAC sounds fine on any USB port, and it persists across reboots (whereas prior to doing that, the distortion persisted across reboots).

Is it possible that something might not be initialized properly except when switching to the internal 3.5mm output?

MichaIng, 2022-12-05:
Probably it's related to hardware interrupts all done on the same CPU. E.g. on legacy kernel C2 images we apply SMP affinities. Can you show:
```sh
cat /proc/interrupts
```

MichaIng, 2022-12-05:
Looks good, most IRQs are the same. The graphics/panfrost ones are different, somehow, and you do not seem to use the serial console, while I use no Ethernet (a USB WiFi adapter instead). Not sure whether the IDs depend on the order the devices are detected by the kernel, e.g. `ff808000.ir` and `ff809000.adc` have switched IDs in our cases.

Test whether changing the affinity helps to solve the USB sound quirks. If so, I'll think about how to implement this into our Odroid N2 images OOTB.

MichaIng, 2022-12-18:
Strange indeed. A I see you used Linux 5.19. Probably you switched to edge kernel to solve or test another issue? Then with edge Linux the issue may still exist.

Yours is Linux 5.10, right?
```sh
uname -a
```

Kreeblah, 2022-12-18:
No, I'm actually on 5.19.17.

```
# uname -a
Linux DietPi 5.19.17-meson64 #22.11.1 SMP PREEMPT Wed Nov 30 11:05:42 UTC 2022 aarch64 GNU/Linux
```

This is just from the [DietPi image](https://dietpi.com/downloads/images/DietPi_OdroidN2-ARMv8-Bullseye.7z) with whatever updates it does (so, no switching kernels or branches) and installing the Roon bridge and dependencies/enabling the USB DAC.  No other changes.

I just double-checked the image I've been using to make sure it's the same as what's currently, and it's the same.  Both have a SHA256 sum of `8b6bb04307086d91bba4966025272637d15b71a9e6ce060dbfeba66ab6614a58` for the 7zip file.

MichaIng, 2022-12-18:
Ah okay, I just didn't recognise yet that the "current" kernel branch for meson64 obviously moved to Linux 5.19 already. Seems to have been done with Armbian v22.11.

Let me keep this issue open, so when I'm back home from holiday I'll check whether/how Armbian's boot scripts do tweak SMP affinities on meson64/Odroid N2 and how. I actually also read somewhere else that it doesn't have any benefit nowadays, but it doesn't sound logic to me to let a single core dealing with all hardware interrupts.

MichaIng, 2023-02-12:
Okay, lacking a better method, this should do it:
```sh
cat << '_EOF_' > /var/lib/dietpi/postboot.d/smp_affinities.sh
#!/bin/dash
while read -r line
do
	[ "$skipped" ] || { skipped=1; continue; } # Skip first line
	irq=${line%%:*}
	expr "$irq" : '^[0-9]\+$' > /dev/null || break # Stop when reaching non-numeric IRQs
	case "$line" in
		*mmc*) echo 1 > "/proc/irq/$irq/smp_affinity_list";;
		*xhci-hcd*) echo 2 > "/proc/irq/$irq/smp_affinity_list";;
		*eth0*) echo 3 > "/proc/irq/$irq/smp_affinity_list";;
	esac
done < /proc/interrupts
_EOF_
```
- MMC to CPU1
- USB3 to CPU2
- Ethernet to CPU3

MichaIng, 2023-03-11:
@Kreeblah 
Are you still bugged with the issue and able to test the above script?

MichaIng, 2023-03-11:
I tested it here with a cheap LogiLink USB DAC and test OGG and MP3 audio files and could not face any audio distortion on different ports with a USB WiFi adapter attached as well. Hence I cannot test whether above SMP affinity tweak is helping.

We are talking about the digital audio stream to the DAC, so the USB data transfer must be significantly disturbed for such effect, but if anyone can verify it, and that the above tweak helps, I'll add this OOTB to our Odroid N2 images, as other USB-related tasks would then benefit as well.

Kreeblah, 2023-03-11:
Adding that script to set affinities works fine for me, though there was a point at which I stopped being able to replicate this issue on my hardware without it, and I'm still not sure why, or what was causing it originally.

Still, it's probably worth including this in the OOTB images since the N2 has several cores to work with anyway.

MichaIng, 2023-03-11:
Okay thanks for reporting back. I'll reread current information about SMP affinities and whether they really make still sense with current Linux. I remember some discussion which stated it wouldn't have benefits today, and that this is also a reason why it isn't possible on Raspberry Pi (the handles are write-protected). We shouldn't add a script if it has no actual benefit 😉.

Kreeblah, 2023-03-15:
Now that I've had that script running for a few days, I've noticed something else.  Up until a few days ago, using the volume slider in Roon would sometimes cause one channel (left or right) to cut out until I changed the volume again.  I figured that was just an issue with my DAC, but it hasn't happened since then.

I'll need to do some more testing, though, since I updated to v8.16 shortly after it became available, so I'll need to isolate whether it's a result of the new version or the CPU affinity script.

Kreeblah, 2023-04-02:
I took a more in-depth look, and it doesn't seem that the CPU affinity script is what improved things there.  Removing the script doesn't seem to make a difference.  I think the new version resolved things for me.

It turns out I _do_ still occasionally get some odd behavior when adjusting the volume (occasionally it'll drop to 0 instead of the value it was supposed to go to), but that seems to be extremely rare.  I've only gotten it to happen a couple of times despite trying to make it happen.  That's a separate issue, though, so if it ends up being a big thing, I'll open a new one.  As far as what I originally reported here, though, I haven't been able to replicate it in a while, so I think it's solved out of the box now.

MichaIng, 2023-04-03:
Okay, thanks for testing. I was reading a little about SMP IRQ affinities and generally it is possible that this enhances performance or lower latency on high load systems or real-time applications. Sometimes it is not about balancing IRQs between CPU cores but to have interrupts for a hardware device and the related process on the same core for cache sharing.

The cache is btw also a reason why letting multiple cores handling the same interrupt is usually not beneficial. Also power saving considerations play a role as you do not want all core to wake up regularly if actually one core would be able to deal with all interrupts. However, there is a daemon for balancing IRQs among cores which is interesting to have a look at when looking deeper into this topic:
- https://packages.debian.org/bookworm/irqbalance
- https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/performance_tuning_guide/appe-red_hat_enterprise_linux-performance_tuning_guide-tool_reference#sect-Red_Hat_Enterprise_Linux-Performance_Tuning_Guide-Tool_Reference-irqbalance

I think the cases where this makes a real difference are low, otherwise the default wouldn't be that CPU0 handles all hardware interrupts by default. Marking this issue as closed but added to archive to remember and in case have another look when we find evidence that it helps.

closed 2023-04-03 by MichaIng