MichaIng/DietPi #5950: unable to upgrade v8.10.2 to v8.11.2

eddydc, 2022-12-04:
- Date           | Sun Dec  4 09:23:57 GMT 2022
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `apt-get -y upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:
sudo dietpi-update on a v8.10.2

#### Expected behaviour:
update to v8.11.2 as it worked on my other machine, which is configured the same way

#### Actual behaviour:
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2022-12-04 09:23:57 GMT; 38ms ago
     Docs: man:nginx(8)
  Process: 1241 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
  Process: 1242 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=1/FAILURE)

Dec 04 09:23:55 DietPi nginx[1242]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:23:55 DietPi nginx[1242]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:23:56 DietPi nginx[1242]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:23:56 DietPi nginx[1242]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:23:56 DietPi nginx[1242]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:23:56 DietPi nginx[1242]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:23:57 DietPi nginx[1242]: nginx: [emerg] still could not bind()
Dec 04 09:23:57 DietPi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Dec 04 09:23:57 DietPi systemd[1]: nginx.service: Failed with result 'exit-code'.
Dec 04 09:23:57 DietPi systemd[1]: Failed to start A high performance web server and a reverse proxy server.
dpkg: error processing package nginx-full (--configure):
 installed nginx-full package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of nginx:
 nginx depends on nginx-full (<< 1.14.2-2+deb10u5.1~) | nginx-light (<< 1.14.2-2+deb10u5.1~) | nginx-extras (<< 1.14.2-2+deb10u5.1~); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.
 nginx depends on nginx-full (>= 1.14.2-2+deb10u5) | nginx-light (>= 1.14.2-2+deb10u5) | nginx-extras (>= 1.14.2-2+deb10u5); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.

dpkg: error processing package nginx (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 nginx-full
 nginx
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

#### Extra details:
Tried this without any difference: https://github.com/MichaIng/DietPi/issues/175

#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up nginx-full (1.14.2-2+deb10u5) ...
Job for nginx.service failed because the control process exited with error code.
See "systemctl status nginx.service" and "journalctl -xe" for details.
invoke-rc.d: initscript nginx, action "start" failed.

sudo systemctl status nginx.service
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2022-12-04 09:30:35 GMT; 37s ago
     Docs: man:nginx(8)
  Process: 1847 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
  Process: 1848 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=1/FAILURE)

Dec 04 09:30:33 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:33 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:35 DietPi nginx[1848]: nginx: [emerg] still could not bind()
Dec 04 09:30:35 DietPi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Dec 04 09:30:35 DietPi systemd[1]: nginx.service: Failed with result 'exit-code'.
Dec 04 09:30:35 DietPi systemd[1]: Failed to start A high performance web server and a reverse proxy server.

sudo journalctl -xe
-- Subject: A start job for unit nginx.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit nginx.service has begun execution.
--
-- The job identifier is 323.
Dec 04 09:30:32 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:32 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:33 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:33 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:33 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:33 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 04 09:30:34 DietPi nginx[1848]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Dec 04 09:30:35 DietPi nginx[1848]: nginx: [emerg] still could not bind()
Dec 04 09:30:35 DietPi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit nginx.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Dec 04 09:30:35 DietPi systemd[1]: nginx.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit nginx.service has entered the 'failed' state with result 'exit-code'.
Dec 04 09:30:35 DietPi systemd[1]: Failed to start A high performance web server and a reverse proxy server.
-- Subject: A start job for unit nginx.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit nginx.service has finished with a failure.
--
-- The job identifier is 323 and the job result is failed.
Dec 04 09:30:40 DietPi sudo[1279]: pam_unix(sudo:session): session closed for user root
Dec 04 09:31:12 DietPi sudo[1876]:   dietpi : TTY=pts/0 ; PWD=XXX ; USER=XXX ; COMMAND=/bin/systemctl status nginx.service
Dec 04 09:31:12 DietPi sudo[1876]: pam_unix(sudo:session): session opened for user XXX by XXX(uid=0)
Dec 04 09:31:12 DietPi sudo[1876]: pam_unix(sudo:session): session closed for user XXX
Dec 04 09:31:53 DietPi sudo[1882]:   dietpi : TTY=pts/0 ; PWD=XXX ; USER=root ; COMMAND=/bin/journalctl -xe
Dec 04 09:31:53 DietPi sudo[1882]: pam_unix(sudo:session): session opened for user XXX by XXX(uid=0)

Joulinar, 2022-12-04:
Nginx is not able to start as something else is already blocking the port. Can you share following

```
ss -tulpn | grep LISTEN
```

eddydc, 2022-12-04:
tcp     LISTEN   0        128              0.0.0.0:111           0.0.0.0:*
tcp     LISTEN   0        128              0.0.0.0:22            0.0.0.0:*
tcp     LISTEN   0        256            127.0.0.1:5335          0.0.0.0:*
tcp     LISTEN   0        100              0.0.0.0:25            0.0.0.0:*
tcp     LISTEN   0        128                 [::]:111              [::]:*
tcp     LISTEN   0        4096                   *:80                  *:*
tcp     LISTEN   0        4096                   *:53                  *:*
tcp     LISTEN   0        128                 [::]:22               [::]:*
tcp     LISTEN   0        100                 [::]:25               [::]:*

Strange thing is that all previous dietpi updates were successful and the config didn't change for 2 years now.

MichaIng, 2022-12-04:
Please copy and paste the whole content. Interesting is the last column which shows the process names.

> Strange thing is that all previous dietpi updates were successful and the config didn't change for 2 years now.

It is a conflicting software which got installed in the meantime on your system, not related to the DietPi update.

eddydc, 2022-12-04:
That's the whole content. No process names.
No additional software has been installed. The only thing that runs on that machine is adguard and once a week a BU script

MichaIng, 2022-12-04:
Then you omitted the `-p` option? Please try again:
```sh
ss -tlpn
```

eddydc, 2022-12-04:
I did provide to -p option
[image]

Joulinar, 2022-12-04:
Let's try following

```
ps -ef | grep www
```

MichaIng, 2022-12-04:
What the... This is what the `-p` is supposed to do. This doesn't work either?
```sh
ss -nl -p | grep ':80'
```

eddydc, 2022-12-04:
Sorry, my mistake. I wasn't aware that the names only show with sudo
[image]

It seems that Adguard is blocking things. This is strange, since it was there all the time since the machine is setup. It is it sole purpose of the machine

MichaIng, 2022-12-04:
Ah okay, I wasn't aware of this either 😅. So AdGuard Home is for some reason listening on port 80.

Joulinar, 2022-12-04:
Question is how AGH was installed? Manually or using our scrips?

eddydc, 2022-12-04:
Okay. Killed Adguard and the update was succesfull.
Restarted AGH without error
Thing is that Nginx took port 80 now and not AGH. The latter doesn't seem to listen
[image]

I'm wondering why that's an issue now and not in the past. 

A such, I don't need nginx, do I?

Honestly I don't recall anymore if I did a manual install or via the scripts. Suggestion how to fix this and/or convert to the scripted install without losing the AGH config?

MichaIng, 2022-12-04:
> The latter doesn't seem to listen

It isn't started automatically until reboot when you killed it, and it would now fail most likely because Nginx blocks the port.

Please still show
```sh
grep bind_port /mnt/dietpi_userdata/adguardhome/AdGuardHome.yaml
```
which, if it fails or not, also is an indicator how you installed AGH.

eddydc, 2022-12-04:
AGH isn't installed under /mnt/dietpi_userdata but the home dir of a user, so that clarifies that.
Meanwhile I've stopped Nginx and started AGH, which runs on 80 again.

So what's the best way to proceed in order to prevent this issue  for future upgrade? 
I don't need nginx, but is it mandatory for the future? Will a dietpi script install prevent these conflicts? 
If so, how can I migrate without losing the config. Keeping the yaml, unstall AGH manually, install it via the scripts and moving the yaml? Or isn't it that simple?

Joulinar, 2022-12-04:
Let's check what is installed on your system

```
dietpi-software list | grep " =2 "
```

eddydc, 2022-12-04:
[image]

Apart from that AGH and raspiBackup. 

Goal was always a barebone system with as less things installed as possible so I'm paining my memory if I did install nginx for some reason, but I can't find a reason.

Joulinar, 2022-12-04:
Looks like Nginx has been installed by yourself outside our script. If not needed, you could remove it

```
apt purge nginx
```

MichaIng, 2022-12-04:
Generally you can simply set/change the `bind_port` setting in `AdGuardHome.yaml` to change the port to something that is more individual. But it also means that you need to define that port explicitly to access its web interface. We use port `8083` for AGH by default.

However, if you're not using the web server (Nginx) for anything, you can just purge it and leave AGH as it is.

To uninstall Nginx safely, to rule out something else depends on it, try this:
```sh
apt-mark auto $(dpkg --get-selections 'nginx' 'nginx-*' 2> /dev/null | mawk '{print $1}')
apt autopurge
```
If nothing is removed, something else depends on Nginx.

Joulinar, 2022-12-04:
AHH and we have an own backup tool called `dietpi-backup`. Usually no need for additional software like raspiBackup. Basically you complicated thinkings a bit. Theoretically you could have used our tools directly to install AGH and to do backups 😉

eddydc, 2022-12-04:
OK Lessons learned. Thank you all for the swift responses.

Joulinar, 2022-12-04:
OK I'm going to close the issue for now. Feel free to reopen if needed.

closed 2022-12-05 by MichaIng