MichaIng/DietPi #5953: DietPi-Software | Reinstall Baikal: MariaDB vs. SQLite

twikedk, 2022-12-04:
### Creating a feature request
I'm not sure if this should go here or as a 'bug'

#### Is your feature request related to a problem? Please describe:
I'm running Baikal with SQLite and as I'm the only user it should be fine. But when I do a  reinstall/update I always end up with having MariaDB installed. This is a little frustrating as I always end up uninstalling it again :-)

#### Describe the solution you'd like:
The solution I can think of is when reinstalling/updating apllication/software the system should only update what is already installed unless of course due to .g. dependency new software is needed.

I could find there had been some thought about something similare in #4420

Joulinar, 2022-12-04:
We have a hard code dependency for MariaDB (software ID 88)

https://github.com/MichaIng/DietPi/blob/98873288ede2c05f4a3b60201759c519da82ed04/dietpi/dietpi-software#L983-L988

Therefore it is going to be installed always.

MichaIng, 2022-12-04:
Thanks for reporting.

Hmm, our Baïkal software option did always depend on and setup with MariaDB, and since also for a single person calendars and contact lists can grow far beyond of what I'd ever store with SQLite, I think this is the right thing. Do you have any particular issue with MariaDB in this context?

But yes, you already linked the issue which is about giving users more choice in this regards.

twikedk, 2023-12-30:
I never got back to this one.
No I don't have any issues with using MariaDB here but my setup was based on older manual settings prior to my knowledge of DietPi's existence. And for both they were to begin with running on their own SBC, now running on just one SBC and it was easy to move it just take the sqlite DB and move it to the common unit without having to think about how to migrate it to MariaDB. I will keep my running setup and live with that when updating I can decide if I want to remove the newly installed MariaDB. :-)

You may close this one.

Joulinar, 2023-12-30:
@MichaIng I doubt we are going to implement this one. Should we close it?

MichaIng, 2023-12-30:
Right, and we still have #4420 as a general idea for a database choice system.

closed 2023-12-30 by MichaIng