MichaIng/DietPi #5962: QUARTZ64: microk8s could not installed

BlackByteDE, 2022-12-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8, G_DIETPI_VERSION_SUB=11, G_DIETPI_VERSION_RC=2
- Distro version | bullseye
- Kernel version | 6.1.0-rc1
- SBC model | Quartz64 (aarch64)
- Power supply used | 12V 5A
- SD card used | 32GB EMMC

#### Additional Information (if applicable)
- Software title | microk8s
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

On a new installed Quartz64 with latest image i've tried to activate microk8s. It gives me :
`- Command: snap install microk8s --classic
- Exit code: 1
- DietPi version: v8.11.2 (MichaIng/master) | HW_MODEL: 49 | HW_ARCH: 3 | DISTRO: 6 
- Image creator: DietPi Core Team 
- Pre-image: from scratch
- Error log: 
error: 
 system does not fully support snapd: cannot mount squashfs image using 
 "squashfs": mount: /tmp/sanity-mountpoint-912337423: wrong fs type, bad 
 option, bad superblock on /dev/loop0, missing codepage or helper 
 program, or other error. `

#### Steps to reproduce
1. use graphical installer for module microk8s

#### Expected behaviour
working

#### Actual behaviour
not working

#### Extra details
with dmesg i got some infos
` 11.379292] Filesystem uses "xz" compression. This is not supported`
further analysis shows me that the kernel is not with CONFIG_SQUASHFS_XZ compiled 
`~$ cat /proc/config.gz | gunzip > running.config
~$ cat running.config | grep _XZ
CONFIG_RD_XZ=y
# CONFIG_MODULE_COMPRESS_XZ is not set
# CONFIG_SQUASHFS_XZ is not set
CONFIG_XZ_DEC=y
...`

MichaIng, 2022-12-09:
Many thanks for reporting.

Indeed the kernel has not set this explicitly: https://gitlab.com/pgwipeout/quartz64_ci/-/blob/main/quartz64_defconfig

Sadly Peter is not reacting to my merge requests to add further common features: https://gitlab.com/pgwipeout/quartz64_ci/-/merge_requests

I'll try to generate the image with Debian's generic arm64 kernel in January.

MichaIng, 2023-02-11:
FYI: #6151

MichaIng, 2023-02-18:
`CONFIG_SQUASHFS_XZ=y` itself isn't sufficient, `CONFIG_SQUASHFS=m` was missing 🙂. But new kernel builds work very well.

MichaIng, 2023-02-18:
MicroK8s works now with new kernel: https://dietpi.com/downloads/images/testing/
I'm during further fine tuning and testing before moving new images to stable builds.

The kernel itself can be upgraded as well with the matching firmware package from here: https://dietpi.com/downloads/binaries/

Will ship this upgrade with next DietPi update, after some more testing and probably little enhancements to the kernel config.

closed 2023-02-19 by MichaIng