MichaIng/DietPi #5965: /var/lib/logrotate/pihole is world-readable after upgrade to bullseye

Joulinar, 2022-12-09:
### Discussed in https://github.com/MichaIng/DietPi/discussions/5964

<div type='discussions-op-text'>

<sup>Originally posted by **eddydc** December  9, 2022</sup>
#### Required Information
cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
echo $G_DISTRO_NAME $G_RASPBIAN
bullseye 0
uname -a
Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
echo $G_HW_MODEL_NAME
RPi 4 Model B (aarch64)

Upgraded this week two machines to bullseye, by following each instruction in https://dietpi.com/blog/?p=811: an AGH and pihole machine.

All went well expect for the latter where I get the error: state file /var/lib/logrotate/pihole is world-readable and thus can be locked from other unprivileged users. Skipping lock acquisition...

Is this serious and how can this be fixed?</div>

MichaIng, 2022-12-09:
Look like an error from `logrotate` itself. No idea why the file would be world-readable. @eddydc can you show the output of:
```sh
ls -al /var/lib/logrotate
```

eddydc, 2022-12-09:
Thx for the very quick response. Here it is

ls -al /var/lib/logrotate
total 16
drwxr-xr-x  2 root root 4096 Dec  9 20:51 .
drwxr-xr-x 22 root root 4096 Dec  9 20:50 ..
-rw-r-----  1 root root  198 Dec  9 20:51 pihole
-rw-r--r--  1 root root  344 Dec  9 00:00 status

MichaIng, 2022-12-09:
So it is not world-readable 🤔. Does the error still show up?

Joulinar, 2022-12-10:
What DNS resolver has been set locally on the system running PiHole?

```
cat /etc/resolv.conf
```

eddydc, 2022-12-10:
The pihole 
nameserver 127.0.0.1
nameserver 9.9.9.9
AGH has only
nameserver 9.9.9.9

Both have unbound installed, so I guess that the 9.9.9.9 shouldn't be there at all (and is the AGH is not correctly configured. Maybe this remark is stupid, but I'm not a linux expert. I find my way in things, but that's about it.

eddydc, 2022-12-10:
Just read [this](https://www.reddit.com/r/pihole/comments/ksi37v/do_i_need_to_disable_resolvconf_for_unbound/), it's the other way around. pihole uses unbound and not the system. Again, not being expert, things sometimes gets confusing.
So I deleted the 127.0.0.1 and rebooted.
Problem vanished.

Joulinar, 2022-12-10:
Yes there is a difference between PiHole and the operating system themselves. PiHole uses unbound as it's configured within PiHole settings. However the operating system has an own configuration how to resolve DNS queries. From our side we always recommend not to use PiHole or AGH as upstream DNS on os level. Just to avoid the DNS issue you where facing 😉

MichaIng, 2022-12-10:
About the original issue, probably it is `logrotate` itself which behaves/expects differently and fixed it on next boot session. The Pi-hole scripts just pass the file path to `logrotate` and do not pre-create it at all.

Yes, better to have the Pi-hole/Unbound host system not relying on Pi-hole/Unbound itself.

eddydc, 2022-12-10:
Guys, again thanks for the quick and great help. Much appreciated. I'd which that everybody would react as such.
Issue can be closed.

closed 2022-12-10 by MichaIng