MichaIng/DietPi #5967: qbittorrent WebUI not accepting any passsword after fresh install

alucard87pl, 2022-12-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye 0`
- Kernel version | `uname -a`
`Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
Official RPi 4 3A 5V Adapter
- SD card used | (EG: SanDisk ultra)
USB Thumbdrive boot, Sandisk 32GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) `qbittorrent`
- Was the software title installed freshly or updated/migrated? fresh update on a fresh DietPi install
- Can this issue be replicated on a fresh installation of DietPi? should be

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Install fresh DietPi, change software and user passwords as recommended
2. install `qbittorrent`
3. Go to `http://PI_IP:1340/`
4. enter suggested login information (username `qbittorrent`, password should still initially be `dietpi` as said by the docs due to the hashing issue)

#### Expected behaviour

Login successful

#### Actual behaviour

Invalid Username or Password.
I tried all possible combinations of username (`dietpi`, `root`, `qbittorrent`) and password (from the docs)

#### Extra details

- ...

Joulinar, 2022-12-10:
i just did a test with my R5S demo system. For me user `qbittorrent` and password `dietpi` is working fine. Just ensure that there are no spaces on the user name / password if you copy them from web site.

MichaIng, 2023-02-12:
Marking as closed due to outstanding reply. Feel free to reopen if the issue persists.

closed 2023-02-12 by MichaIng