MichaIng/DietPi #5971: [Dietpi-ddns] show token instead of hiding it behind password char (*)

TDuffinNTU, 2022-12-11:
Since freedns.afraid.org's token is particularly long and includes numbers, characters, and upper case characters, it can be quite easy to mess up inputting the access token to the DDNS provider when the token is obscured and requires to be input twice to work.

I appreciate the security aspect of obscuring the token, but I'd argue that in the context of a home webserver (the most common usecase for DietPi/DDNS setup) and the fact I'm already manually copying said access token from another webpage unobscured, it's likely a redundant security measure in the first place at best. At worst, it makes setup of DDNS cronjobs more cumbersome than necessary. It also makes it necessary to write the token twice in an equally cumbersome manner!

I'm not too aware of the features in the UI API used dietpi shell scripts, but a potential option could be to allow users to unobscure the password with a toggle, and then removing the repeated token entry :)

MichaIng, 2022-12-12:
Quite an argument. Probably we can differenciate between passwords and tokens for using password input or regular input box. Sadly with whiptail we cannot add a show/hide toggle, which would be very handy.

But in case of tokens, generally it's great to copy&paste. With SSH at least this is simple.

TDuffinNTU, 2023-10-06:
Brilliant point RE: SSH. Since I was doing this setup directly rather than SSH I had to type it manually and didn't think of any alternatives!

I'll close this for now since realistically the solution is to use copy-paste to avoid human error entirely. Appreciate the reply!

closed 2023-10-06 by TDuffinNTU