MichaIng/DietPi #5973: Nanopi R2s first boot setup error

cloud08, 2022-12-12:
#### Details:
- Date           | Mon Dec 12 04:11:26 GMT 2022
- DietPi version | v8.11.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | NanoPi R2S (aarch64) (ID=55)
- Kernel version | `Linux DietPi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `modprobe ledtrig-netdev`
- Exit code      | 1
- Software title | DietPi-Patch
#### Steps to reproduce:
Flashing the latest Nanopi R2s image and logging in root for the first time
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
modprobe: FATAL: Module ledtrig-netdev not found in directory /lib/modules/5.10.63-rockchip64
```

cloud08, 2022-12-12:
- Reference code: be89b46b-1856-45df-9841-5fbc980b4d87

MichaIng, 2022-12-12:
Yes I recognised this, when a kernel upgrade is done as well and fixed it in beta already. A reboot will also fix it.

I wonder why a DietPi update is done, since the image should ship with latest DietPi release v8.11 already. Was this image downloaded more than 3 weeks ago?

cloud08, 2022-12-12:
I got the image yesterday from the official dietpi.com website under Nanopi R2s
https://dietpi.com/downloads/images/DietPi_NanoPiR2S-ARMv8-Bullseye.7z

link probably needs to be updated.

Joulinar, 2022-12-12:
It's a permanent link and will point to latest image always

MichaIng, 2022-12-30:
Solved with DietPi v8.12 and new image.

closed 2022-12-30 by MichaIng