MichaIng/DietPi #5976: xRDP – cannot read /etc/xrdp/key.pem. Permission denied error

amibumping, 2022-12-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.11.2
- Distro version | bullseye
- Kernel version | Linux DietPi 5.19.17-meson64 #22.11.1 SMP PREEMPT Wed Nov 30 11:05:42 UTC 2022 aarch64 GNU/Linux
- SBC model | Odroid HC4

Randomly I've found this error on xRDP log:

[20221214-09:43:09] [ERROR] Cannot read private key file /etc/xrdp/key.pem: Permission denied

As I can read, the problem is because the user xrdp don't have access to ssl folder to read the real cert that is linked.

[https://c-nergy.be/blog/?p=13708](https://c-nergy.be/blog/?p=13708)

The solution is to add user xrdp into ssl-cert group

MichaIng, 2022-12-18:
I recognised this but thought it is only relevant if you do key authentication instead of password authentication, isn't it? Sadly the guide you linked as well doesn't say anything about what it's used for. However, that those are symlinks to the dummy/snakeoil certificate (shipped by `ssl-cert` package, not pre-installed on DietPi), shows that it should be properly configured instead and is optional.

So instead of setting it up to use snakeoil, we should find out what exactly it is, how to use it properly, document that and disable it completely by default (so that it does not try to load the snakeoil cert).

MichaIng, 2023-02-12:
Okay its about TLS encryption for client connections via TLS. A key and a cert needs to be set both: https://manpages.debian.org/bullseye/xrdp/xrdp.ini.5.en.html#certificate

This generally makes sense to have enabled OOTB, but I'm not sure what the Windows RDP client does when not trusting the cert, which would be the case with the default snake oil one or any self-signed one, unless going through the process or adding it to the trusted certs/CA store.

MichaIng, 2023-02-12:
Found the Debian bug report about this: https://bugs.debian.org/860890

> Control: tags -1 + wontfix
> 
> This is entirely normal, common to many Debian packages and basic knowledge for a Debian administrator.
> 
> -nik

How I hate such kind of answers. This is one of the reasons so many "How to start with Linux" guides recommend Ubuntu over Debian, which is otherwise nonsense...

Why is the `ssl-cert` package a strict dependency if it cannot be used by XRDP without manual setup. If access permissions is such common admin knowledge, then adding a proper certificate, or at least installing the snakeoil certificate is just the same common knowledge. But the bigger problem is that it degrades security completely unnecessarily until this setup is done.

We'll do this in `dietpi-software` now.

MichaIng, 2023-02-12:
Done with: https://github.com/MichaIng/DietPi/commit/f8a272a

closed 2023-02-15 by MichaIng

amibumping, 2023-09-19:
I dont' know if it's correct to answer on a closed issue, but xrdp has received and update, and after this I can no login with root&dietpi users. I tried to reinstall xrdp but again the same. I run `journalctl -u xrdp` and I see the same error of this issue:
`Cannot read private key file /etc/xrdp/key.pem: Permission denied`

Thank you @MichaIng

MichaIng, 2023-09-19:
This error did never break anything but was only an information that XRDP had no permissions to access the snakeoil certificate, and hence only non-encrypted traffic was possible. This is however negotiated automatically between client and server, unless you enforce TLS at one side.

The last update of XRDP was last December, hence before this issue was solved, so since then nothing about the matter has changed.

We  did never apply the change for running systems, only for fresh XRDP installs or reinstall. You can manually apply it:
```sh
sudo usermod -aG ssl-cert xrdp
```
While this mutes the warning, I guess it does not solve your connection issue. Could you paste the full service log?

amibumping, 2023-09-19:
I am not saying is a dietpi problem but xrdp. I use it everyday, yesterday was working, and today after the update and restarting the service it broke.

This is the full log, and an attempt to log.

`Sep 19 23:08:19 DietPi systemd[1]: Stopping xrdp daemon...
Sep 19 23:08:19 DietPi xrdp[713028]: stopping xrdp
Sep 19 23:08:19 DietPi xrdp[713028]: stopping process id 57798
Sep 19 23:08:19 DietPi xrdp[57798]: [INFO ] Received termination signal, stopping the server accept new connections thread
Sep 19 23:08:19 DietPi xrdp[57798]: [DEBUG] Closed socket 11 (AF_INET6 :: port 62826)
Sep 19 23:08:19 DietPi systemd[1]: xrdp.service: Succeeded.
Sep 19 23:08:19 DietPi systemd[1]: Stopped xrdp daemon.
Sep 19 23:08:19 DietPi systemd[1]: Starting xrdp daemon...
Sep 19 23:08:19 DietPi xrdp[713042]: [INFO ] address [0.0.0.0] port [62826] mode 1
Sep 19 23:08:19 DietPi xrdp[713042]: [INFO ] listening to port 62826 on 0.0.0.0
Sep 19 23:08:19 DietPi xrdp[713042]: [INFO ] xrdp_listen_pp done
Sep 19 23:08:19 DietPi xrdp[713042]: [DEBUG] Closed socket 7 (AF_INET6 :: port 62826)
Sep 19 23:08:19 DietPi systemd[1]: xrdp.service: Can't open PID file /run/xrdp/xrdp.pid (yet?) after start: Operation not permitted
Sep 19 23:08:20 DietPi systemd[1]: Started xrdp daemon.
Sep 19 23:08:21 DietPi xrdp[713044]: [INFO ] starting xrdp with pid 713044
Sep 19 23:08:21 DietPi xrdp[713044]: [INFO ] address [0.0.0.0] port [62826] mode 1
Sep 19 23:08:21 DietPi xrdp[713044]: [INFO ] listening to port 62826 on 0.0.0.0
Sep 19 23:08:21 DietPi xrdp[713044]: [INFO ] xrdp_listen_pp done
Sep 19 23:09:40 DietPi xrdp[713044]: [INFO ] Socket 12: AF_INET6 connection received from ::ffff:192.168.1.18 port 38942
Sep 19 23:09:40 DietPi xrdp[713044]: [DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.1.3 port 62826)
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] Closed socket 11 (AF_INET6 :: port 62826)Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ini_version, value 1
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item fork, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item port, value 62826
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item use_vsock, value false
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item tcp_nodelay, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item tcp_keepalive, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item security_layer, value negotiate
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item crypt_level, value high
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item certificate, value
Sep 19 23:09:40 DietPi xrdp[714465]: [INFO ] Using default X.509 certificate: /etc/xrdp/cert.pem
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item key_file, value
Sep 19 23:09:40 DietPi xrdp[714465]: [INFO ] Using default X.509 key file: /etc/xrdp/key.pem
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ssl_protocols, value TLSv1.2, TLSv1.3
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] TLSv1.3 enabled
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] TLSv1.2 enabled
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item autorun, value
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item allow_channels, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item allow_multimon, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item bitmap_cache, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item bitmap_compression, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item bulk_compression, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item max_bpp, value 32
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item new_cursors, value true
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item use_fastpath, value both
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item blue, value 009cb5
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item grey, value dedede
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_top_window_bg_color, value e0ecf0Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_width, value 350
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_height, value 430
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_bg_color, value 59667d
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_logo_filename, value
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_logo_x_pos, value 55
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_logo_y_pos, value 50
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_label_x_pos, value 30
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_label_width, value 65
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_input_x_pos, value 110
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_input_width, value 210
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_input_y_pos, value 220
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_ok_x_pos, value 142
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_ok_y_pos, value 370
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_ok_width, value 85
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_ok_height, value 30
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_cancel_x_pos, value 237
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_cancel_y_pos, value 370
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_cancel_width, value 85
Sep 19 23:09:40 DietPi xrdp[714465]: [DEBUG] item ls_btn_cancel_height, value 30
Sep 19 23:09:40 DietPi xrdp[714465]: [INFO ] Security protocol: configured [SSL|RDP], requested [SSL|HYBRID|HYBRID_EX|RDP], selected [SSL]
[... 138 lines cut ...]
[... cut ...]
Sep 19 23:10:30 DietPi xrdp[714465]: [ERROR] xrdp_rdp_recv: xrdp_sec_recv failed
Sep 19 23:10:30 DietPi xrdp[714465]: [ERROR] libxrdp_process_data: xrdp_rdp_recv failed
Sep 19 23:10:30 DietPi xrdp[714465]: [ERROR] xrdp_process_data_in: xrdp_process_loop failed
Sep 19 23:10:30 DietPi xrdp[714465]: [ERROR] SSL_shutdown: I/O error
Sep 19 23:10:30 DietPi xrdp[714465]: [DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.1.3 port 62826)
Sep 19 23:10:30 DietPi xrdp[714465]: [DEBUG] xrdp_mm_module_cleanup
root@DietPi:~#`

[image]

Joulinar, 2023-09-20:
We don't build these packages ourselves. We use upstream packages provided by Debian repository.

MichaIng, 2023-09-20:
Oh, this version was not yet visible in the online database yesterday. Let me see whether I can replicate the issue.

So the issue is that XRDP is now our of sync with `xorgxrdp` and the letter needs to be updated by Debian.

amibumping, 2023-09-20:
That's correct @MichaIng 2 options, waiting for Debian to update it or rebuilding it manually.

MichaIng, 2023-09-20:
Jep, or installing TigerVNC and issue Xvnc login method.

I'll try to re-assign the Debian bug report to the xorgxrdp package as the xrdp maintainer does not seem to be aware that xorgxrdp is provided by Debian as well.

amibumping, 2023-09-21:
Already fixed :+1:

MichaIng, 2023-09-24:
That was fast indeed. They did not pushed a new (upstream) version but backported the particular needed patch.