MichaIng/DietPi #5980: Problema after upgrade to 8.12 with HA last versione

maury77, 2022-12-15:
### Creating a bug report/issue

#### Required Information
DietPi version | 8.12
Distro version | bullseye 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
dic 15 12:43:41 DietPi Home Assistant[1952]: Importing the numpy C-extensions failed. This error can happen for
dic 15 12:43:41 DietPi Home Assistant[1952]: many reasons, often due to issues with your setup or how NumPy was
dic 15 12:43:41 DietPi Home Assistant[1952]: installed.
dic 15 12:43:41 DietPi Home Assistant[1952]: We have compiled some common reasons and troubleshooting tips at:
dic 15 12:43:41 DietPi Home Assistant[1952]:     https://numpy.org/devdocs/user/troubleshooting-importerror.html
dic 15 12:43:41 DietPi Home Assistant[1952]: Please note and check the following:
dic 15 12:43:41 DietPi Home Assistant[1952]:   * The Python version is: Python3.9 from "/home/homeassistant/.pyenv/versions/3.9.10/bin/python3.9"
dic 15 12:43:41 DietPi Home Assistant[1952]:   * The NumPy version is: "1.23.2"
dic 15 12:43:41 DietPi Home Assistant[1952]: and make sure that they are the versions you expect.
dic 15 12:43:41 DietPi Home Assistant[1952]: Please carefully study the documentation linked above for further help.

Joulinar, 2022-12-15:
Did you followed instructions given by the error? 

```
dic 15 12:43:41 DietPi Home Assistant[1952]: Importing the numpy C-extensions failed. This error can happen for
dic 15 12:43:41 DietPi Home Assistant[1952]: many reasons, often due to issues with your setup or how NumPy was
dic 15 12:43:41 DietPi Home Assistant[1952]: installed.
dic 15 12:43:41 DietPi Home Assistant[1952]: We have compiled some common reasons and troubleshooting tips at:
dic 15 12:43:41 DietPi Home Assistant[1952]: https://numpy.org/devdocs/user/troubleshooting-importerror.html
...
dic 15 12:43:41 DietPi Home Assistant[1952]: Please carefully study the documentation linked above for further help.
```

maury77, 2022-12-16:
I solved with this procedure

https://dietpi.com/forum/t/after-the-update-homeassistant-mqtt-broker-integration-broken/14719/17

login as user root
**apt install libatlas-base-dev**
switch to user homeassistant
**sudo -u homeassistant bash
. /home/homeassistant/pyenv-activate.sh
pip3 uninstall numpy**
exit and switch back to user root
**apt install python3-numpy**

closed 2022-12-16 by maury77