MichaIng/DietPi #5982: DietPi PHP 7.4 (is EOL) , current PHP is 8.x

jalsco, 2022-12-16:
### Update PHP to current levels 
Currently using: **DietPi v8.11.2**
#### PHP is at version 7.4.33
Warning (from Matomo): 

The PHP version 7.4 you are using has reached its End of Life (EOL). You are strongly urged to upgrade to a current version, as using this version may expose you to security vulnerabilities and bugs that have been fixed in more recent versions of PHP.

```
[Thu Dec 15] root@batopi: ~$ php --version
PHP 7.4.33 (cli) (built: Nov  8 2022 11:40:37) ( NTS )
Copyright (c) The PHP Group
Zend Engine v3.4.0, Copyright (c) Zend Technologies
    with Zend OPcache v7.4.33, Copyright (c), by Zend Technologies
```

Joulinar, 2022-12-16:
We use PHP version provided by global Debian apt package manager/ repository. Unfortunately this is the version available on Debian Bullseye. PHP 8 will become available using Debian Bookworm. 

Basically you have 3 options.

1. Stay with PHP 7and wait for the release of Debian Bookworm in summer 2023
2. Like to be on bleeding edge and upgrade to Bookworm, even it's still testing repository
3. Install PHP from 3rd party repository

jalsco, 2022-12-16:
Hi @Joulinar , thanks for explaining. I will stay with the default PHP from the distro. 

Just was made a little nervous by that Warning Message from Matomo (see above).
Seems like going to PHP v8 is already in the (Debian) plan - so now is a time for patience.

closed 2022-12-16 by jalsco

Joulinar, 2022-12-16:
Yes PHP 8 is available within Debian Bookworm.