MichaIng/DietPi #5983: conbee II tray 2 implemented

palviehvandesarre, 2022-12-16:
Dietpi has encountered an error

Please create a ticket: https://github.com/MichaIng/DietP1/Issues Copy and paste only the BLUE lines below into the ticket

#### Detalls:

Fri Dec 16 11:42:14 CET 2022

Date

7510814d-1168-411d-a2f0-3d595a688987

Bug report

DietPI version v8.11.2 (MichaIng/master) DletPi Core Team

Pre-image

Hardware

from scratch

Image creator RPI 4 Model B (aarch64) (ID=4) 100

Kernel version Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/LI

Distro

Command

Exit code

bullseye (ID=6, RASPBIAN=0) apt-get-y-allow-change-held-packages autopurge python3-rpi.gpio

Software title | DietPi-Software

#### Steps to reproduce:

1.

#### Expected behaviour:

What SHOULD happen? -->

#### Actual behaviour:

#### Extra details:

#### Additional logs:

E: Malformed entry i in list file /etc/apt/sources.list.d/deconz. List (Component

F: The list of sources could not be read.

E: Malformed entry 1 in list file /etc/apt/sources.list.d/deconz.list (Component) E: The list of sources could not be read.

Tram (0) Dietei-Software | Unable to continue, DietPi-Software will now terminate.

root@Dietpl:"# 9376.059121] usb 1-1.3: USB disconnect, device number 10

palviehvandesarre, 2022-12-16:
currently I can't install any software. It seems that the source is tied to deconz

Joulinar, 2022-12-16:
Did you add source configuration file `/etc/apt/sources.list.d/deconz.list` yourself? Because this is not something we do on our software script. You would need to correct the configuration file or remove it.

palviehvandesarre, 2022-12-16:
yes, I oriented 
[image]
yes, I followed this link

palviehvandesarre, 2022-12-16:
In the meantime I have reflashed my SD card.
What is the ideal way to implement the ZIGBEE dongle ConBee II in my DietPi?
Tanx

palviehvandesarre, 2022-12-16:
there is probably a problem with "lsb_release" and/or "lsb-release" ???

Joulinar, 2022-12-16:
> there is probably a problem with "lsb_release" and/or "lsb-release" ???

Yes, because this tool is not installed by default.

> In the meantime I have reflashed my SD card.

This was not necessary. Simple correcting the source file would have been enough.

```
sudo sh -c "echo 'deb http://phoscon.de/apt/deconz bullseye main' > /etc/apt/sources.list.d/deconz.list"
apt update
apt install deconz -y
```

palviehvandesarre, 2022-12-16:
subba vielen DANK :-) StepByStep

Joulinar, 2022-12-16:
These are the very same steps as on the web site linked above. Just replaced `lsb-release` with current Debian version 😉 

I'm going to close the issue for now.

closed 2022-12-16 by Joulinar