MichaIng/DietPi #5987: fail2ban not banning IP adresses and not noticing failed logins over SSH (similar to #3697)

aszurnasirpal, 2022-12-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8, G_DIETPI_VERSION_SUB=11, G_DIETPI_VERSION_RC=2, G_GITBRANCH='master'`
- Distro version | `bullseye 0`
- Kernel version | `Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)` 
- Power supply used | standard 
- SD card used | SanDisk 256 GB

#### Additional Information (if applicable)
- Software title | fail2ban
- Was the software title installed freshly or updated/migrated? restored from backup (fresh installation then restored by dietpi-backup)
- Can this issue be replicated on a fresh installation of DietPi? Unkown - will try once I have physical access to the SBC again

#### Steps to reproduce
- Install fail2ban with the dietpi-software command
- Deliberately fail the login (with root user and an extra added user)

#### Expected behaviour
- IP address should be banned after three failed attempts (using the default jail)
- failed logins should be shown when using "fail2ban-client status sshd" command

#### Actual behaviour
- fail2ban doesn't show any fail attempts

#### Extra details
```
dietpi@DietPi:~$ sudo journalctl -u sshd
-- Journal begins at Sat 2022-12-17 15:50:48 WET, ends at Sun 2022-12-18 09:50:54 WET. --
-- No entries --
```
```
dietpi@DietPi:~$ sudo fail2ban-regex /var/log/auth.log /etc/fail2ban/filter.d/sshd.conf

Running tests
=============

Use   failregex filter file : sshd, basedir: /etc/fail2ban
Use         maxlines : 1
Use      datepattern : {^LN-BEG} : Default Detectors
Use         log file : /var/log/auth.log
Use         encoding : UTF-8

Results
=======

Prefregex: 5226 total
|  ^(?P<mlfid>(?:\[\])?\s*(?:<[^.]+\.[^.]+>\s+)?(?:\S+\s+)?(?:kernel:\s?\[ *\d+\.\d+\]:?\s+)?(?:@vserver_\S+\s+)?(?:(?:(?:\[\d+\])?:\s+[\[\(]?sshd(?:\(\S+\))?[\]\)]?:?|[\[\(]?sshd(?:\(\S+\))?[\]\)]?:?(?:\[\d+\])?:?)\s+)?(?:\[ID \d+ \S+\]\s+)?)(?:(?:error|fatal): (?:PAM: )?)?(?P<content>.+)$
`-

Failregex: 0 total

Ignoreregex: 0 total

Date template hits:
|- [# of hits] date format
|  [5226] {^LN-BEG}(?:DAY )?MON Day %k:Minute:Second(?:\.Microseconds)?(?: ExYear)?
`-

Lines: 5226 lines, 0 ignored, 0 matched, 5226 missed
[processed in 1.07 sec]

Missed line(s): too many to print.  Use --print-all-missed to print all 5226 lines
```

jail.conf
```

[DEFAULT]
enabled = true
ignoreip = 127.0.0.1/8 192.168.88.1/8
ignorecommand =
backend = systemd
mode = normal
filter = %(__name__)s[mode=%(mode)s]
findtime = 600
maxretry = 3
bantime = 600
banaction = route
action = %(banaction)s[blocktype=blackhole]

[sshd]
# Mode: normal (default), ddos, extra or aggressive (combines all)
# See "filter.d/sshd.conf" for details.
mode = extra
backend = systemd
enabled = true
filter = sshd
action = ufw
findtime = 600
maxretry = 6
bantime = 86400

```

MichaIng, 2022-12-18:
Thanks for reporting.

First of all, you have a customised `jail.conf`. Please try it with the original file. The missing OpenSSH logs in your journal indicate that you did not install OpenSSH, but likely that Dropbear is installed as default SSH server. But you removed the Dropbear jail from `jail.conf`, breaking Fail2Ban for it. It is part of the default config, so that should work.

aszurnasirpal, 2022-12-18:
Hi!
Thanks for the prompt replay.
1) yes, I have OpenSSH enabled and running:
```
dietpi@DietPi:~$ ssh -V
OpenSSH_8.4p1 Debian-5+deb11u1, OpenSSL 1.1.1n  15 Mar 2022
```
2) I purged fail2ban and deleted manually /etc/fail2ban.  Then I Installed it again with default jail.conf
but still the output is the same (regex doesn't catch anything in /var/log/auth.log)
3) yes, I have ufw installed - (but now I have moved to default) 
```

● fail2ban.service - Fail2Ban Service
     Loaded: loaded (/lib/systemd/system/fail2ban.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-12-18 10:39:31 WET; 5min ago
       Docs: man:fail2ban(1)
    Process: 318880 ExecStartPre=/bin/mkdir -p /run/fail2ban (code=exited, status=0/SUCCESS)
   Main PID: 318881 (fail2ban-server)
      Tasks: 5 (limit: 4531)
     Memory: 28.7M
        CPU: 2.793s
     CGroup: /system.slice/fail2ban.service
             └─318881 /usr/bin/python3 /usr/bin/fail2ban-server -xf start

Dec 18 10:39:31 DietPi systemd[1]: Starting Fail2Ban Service...
Dec 18 10:39:31 DietPi systemd[1]: Started Fail2Ban Service.
Dec 18 10:39:33 DietPi fail2ban-server[318881]: Server ready

```
```

dietpi@DietPi:~$ sudo fail2ban-client status sshd
Status for the jail: sshd
|- Filter
|  |- Currently failed: 0
|  |- Total failed:     0
|  `- File list:        /var/log/auth.log
`- Actions
   |- Currently banned: 0
   |- Total banned:     0
   `- Banned IP list:

```

Joulinar, 2022-12-18:
> I purged fail2ban and deleted manually /etc/fail2ban. Then I Installed it again

How do you installed F2B?

aszurnasirpal, 2022-12-18:
apt-get install fail2ban

Joulinar, 2022-12-18:
try to remove and use `dietpi-software` for new install

aszurnasirpal, 2022-12-18:
sudo apt-get purge fail2ban,
then installed via dietpi-software
I modified only slightly jail.conf (added ignore to local subnet)

```
dietpi@DietPi:~$ cat /etc/fail2ban/jail.conf
[DEFAULT]
enabled = true
ignoreip = 127.0.0.1/8 192.168.88.1/8
ignorecommand =
backend = systemd
mode = normal
filter = %(__name__)s[mode=%(mode)s]
findtime = 600
maxretry = 3
bantime = 600
banaction = route
action = %(banaction)s[blocktype=blackhole]

[dropbear]

[sshd]
# Mode: normal (default), ddos, extra or aggressive (combines all)
# See "filter.d/sshd.conf" for details.
mode = normal
```
Unfortunately still zero detection

Joulinar, 2022-12-18:
using defailt installation, without any modification is working well on my demo system

```
root@DietPiR5S:~# fail2ban-client status sshd
Status for the jail: sshd
|- Filter
|  |- Currently failed: 1
|  |- Total failed:     1
|  `- Journal matches:  _SYSTEMD_UNIT=sshd.service + _COMM=sshd
`- Actions
   |- Currently banned: 0
   |- Total banned:     0
   `- Banned IP list:
root@DietPiR5S:~#
```

aszurnasirpal, 2022-12-18:
On my system, it worked well until I needed to restore the backup.
I spotted however strange thing 
in my auth.log. I see "dropbear" entries
```
Dec 18 13:24:16 DietPi dropbear[371953]: Exit before auth from <156.59.103.197:43966>: (user 'root', 1 fails): Disconnect received
Dec 18 13:24:17 DietPi dropbear[371981]: Child connection from 161.35.124.45:34692
Dec 18 13:24:17 DietPi dropbear[371982]: Child connection from 197.155.234.157:47814
Dec 18 13:24:18 DietPi dropbear[371981]: Login attempt for nonexistent user
Dec 18 13:24:18 DietPi dropbear[371981]: Exit before auth from <161.35.124.45:34692>: Disconnect received
Dec 18 13:24:19 DietPi dropbear[371982]: Login attempt for nonexistent user
```
But.. I don't have dropbear installed. 

```
dietpi@DietPi:~$ sudo apt purge --auto-remove dropbear
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'dropbear' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

also, I don't have any idea why regex is not working on this log (all lines are ignored)

aszurnasirpal, 2022-12-18:
```
dietpi@DietPi:~$ sudo journalctl -u ssh.service
-- Journal begins at Sat 2022-12-17 15:50:48 WET, ends at Sun 2022-12-18 13:35:51 WET. --
Dec 17 15:50:53 DietPi systemd[1]: Starting OpenBSD Secure Shell server...
Dec 17 15:50:54 DietPi sshd[626]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Dec 17 15:50:54 DietPi sshd[626]: error: Bind to port 22 on :: failed: Address already in use.
Dec 17 15:50:54 DietPi sshd[626]: fatal: Cannot bind any address.
Dec 17 15:50:54 DietPi systemd[1]: ssh.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 17 15:50:54 DietPi systemd[1]: ssh.service: Failed with result 'exit-code'.
Dec 17 15:50:54 DietPi systemd[1]: Failed to start OpenBSD Secure Shell server.
```

And yes I am in jail.conf uncommented #mode = normal

Joulinar, 2022-12-18:
At least on a default DietPi system logging is not done to `/var/log/auth.log`. SSH server logging is done towards `journalctl`

```
root@DietPiR5S:~# journalctl -u ssh.service
-- Journal begins at Sun 2022-12-18 14:21:46 CET, ends at Sun 2022-12-18 14:24:41 CET. --
Dec 18 14:21:52 DietPiR5S systemd[1]: Starting OpenBSD Secure Shell server...
Dec 18 14:21:52 DietPiR5S sshd[510]: Server listening on 0.0.0.0 port 22.
Dec 18 14:21:52 DietPiR5S sshd[510]: Server listening on :: port 22.
Dec 18 14:21:52 DietPiR5S systemd[1]: Started OpenBSD Secure Shell server.
Dec 18 14:22:16 DietPiR5S sshd[912]: Accepted publickey for root from 192.168.0.92 port 55751 ssh2: RSA SHA256...
Dec 18 14:22:16 DietPiR5S sshd[912]: pam_unix(sshd:session): session opened for user root(uid=0) by (uid=0)
Dec 18 14:24:39 DietPiR5S sshd[1975]: pam_unix(sshd:auth): authentication failure; logname= uid=0 euid=0 tty=ssh ruser= rhost=192.168.0.11  user=root
Dec 18 14:24:41 DietPiR5S sshd[1975]: Failed password for root from 192.168.0.11 port 35336 ssh2
root@DietPiR5S:~#
```

Joulinar, 2022-12-18:
> Dec 17 15:50:54 DietPi sshd[626]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Dec 17 15:50:54 DietPi sshd[626]: error: Bind to port 22 on :: failed: Address already in use.
Dec 17 15:50:54 DietPi sshd[626]: fatal: Cannot bind any address.

There seems to be 2 SSH server running. Can be checked as follow

```
ss -tulpn | grep LISTEN
```

aszurnasirpal, 2022-12-18:
```
dietpi@DietPi:~$ sudo lsof -i :22
COMMAND     PID USER   FD   TYPE  DEVICE SIZE/OFF NODE NAME
dropbear    455 root    4u  IPv4   11477      0t0  TCP *:ssh (LISTEN)
dropbear    455 root    5u  IPv6   11478      0t0  TCP *:ssh (LISTEN)
dropbear 371397 root    6u  IPv4 2420432      0t0  TCP 192.168.88.23:ssh->192.168.88.21:50703 (ESTABLISHED)
```

aszurnasirpal, 2022-12-18:
Strange, I don't see dropbear in my installed packages in dietpi-software. How can I remove it?

Joulinar, 2022-12-18:
Dropbear is running. Can you share

```
dietpi-software list | grep "ssh"
```

aszurnasirpal, 2022-12-18:
```
dietpi@DietPi:~$ sudo dietpi-software list | grep "ssh"
ID 104 | =0 | Dropbear: Lightweight SSH server | | https://dietpi.com/docs/software/ssh/#dropbear
ID 105 | =2 | OpenSSH Server: Feature-rich SSH server with SFTP and SCP support | +OpenSSH Client | https://dietpi.com/docs/software/ssh/#openssh
```

But in dietpi-software, on the list of applications that I can remove, there is no Dropbear listed

I will stop service manually and disable it

Joulinar, 2022-12-18:
wait

aszurnasirpal, 2022-12-18:
Hi, thanks will do that. It could be a root cause for sure. 
I will let you know in a couple of hours if this solves the problem. Many, many thanks, and I appreciate your help very much

aszurnasirpal, 2022-12-19:
Hi,

I can confirm the problem is solved!! You guys are amazing !

closed 2022-12-19 by aszurnasirpal