MichaIng/DietPi #5988: v8.12 - myMPD -fails to install with error

jalsco, 2022-12-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
 ```
 G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=12
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Kernel version | `uname -a`
```
Linux batopi 5.15.80-rockchip64 #22.11.1 SMP PREEMPT Wed Nov 30 11:12:47 UTC 2022 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
ROCK Pi 4 (aarch64)
```
#### Additional Information (if applicable)
- Software title |  MyMPD
- Was the software title installed freshly or updated/migrated? | fresh install

- Bug report ID | `echo $G_HW_UUID`
```
37e1c77c-d2fc-49f0-8b90-1499727ef9da
```
#### Steps to reproduce
1. ... Install myMPD from Dietpi-software

#### Actual behaviour

```
Command: systemctl stop mympd                                                                                     │
                        │  - Exit code: 5                                                                                                      │
                        │  - DietPi version: v8.12.1 (MichaIng/master) | HW_MODEL: 72 | HW_ARCH: 3 | DISTRO: 6                                 │
                        │  - Image creator: DietPi Core Team                                                                                   │
                        │  - Pre-image: Armbian                                                                                                │
                        │  - Error log:                                                                                                        │
                        │ Failed to stop mympd.service: Unit mympd.service not loaded
```

related to issue #5936

MichaIng, 2022-12-18:
Thanks for reporting. The two issues are not related at all 😄.

No idea why this never was an issue before, since the packages `postinst` script never had any command which would reload the service files. But it does exist, doesn't it?
```sh
systemctl daemon-reload
systemctl status mympd
```

jalsco, 2022-12-18:
```
[Sun Dec 18] root@batopi: ~$ systemctl daemon-reload
[Sun Dec 18] root@batopi: ~$ systemctl status mympd
Unit mympd.service could not be found.
```
Sorry to be always reporting problems.  Hope you understand that -overall - just very very grateful to be able to use such a wonderful system, and appreciative of all your hard work to make it exist. thank you 🙏🏻

MichaIng, 2022-12-18:
No problem. But that the service file is not present at all is indeed a problem. Can you show the output of:
```sh
dpkg -L mympd
```

jalsco, 2022-12-18:
```
[Sun Dec 18] root@batopi: ~$ dpkg -L mympd
/.
/lib
/lib/systemd
/lib/systemd/system
/lib/systemd/system/mympd.service
/usr
/usr/bin
/usr/bin/mympd
/usr/bin/mympd-script
/usr/share
/usr/share/doc
/usr/share/doc/mympd
/usr/share/doc/mympd/changelog.Debian.gz
/usr/share/doc/mympd/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/mympd-script.1.gz
/usr/share/man/man1/mympd.1.gz
[Sun Dec 18] root@batopi: ~$
```

```
[Sun Dec 18] root@batopi: ~$ apt install --reinstall mympd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 612 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://download.opensuse.org/repositories/home:/jcorporation/Debian_11  mympd 10.1.6-1 [612 kB]
Fetched 612 kB in 8s (73.8 kB/s)
(Reading database ... 56691 files and directories currently installed.)
Preparing to unpack .../mympd_10.1.6-1_arm64.deb ...
Unpacking mympd (10.1.6-1) over (10.1.6-1) ...
Setting up mympd (10.1.6-1) ...
Removing obsolet mympd user and group
Processing triggers for man-db (2.9.4-2) ...
[Sun Dec 18] root@batopi: ~$
```
```
[Sun Dec 18] root@batopi: ~$ ls -l /lib/systemd/system/mympd.serice
ls: cannot access '/lib/systemd/system/mympd.serice': No such file or directory
[Sun Dec 18] root@batopi: ~$
```

jalsco, 2022-12-18:
I think I left out the `update` - but got nothing new...

```
[Sun Dec 18] root@batopi: ~$ apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease
Get:2 https://dl.cloudsmith.io/public/caddy/stable/deb/debian any-version InRelease [7505 B]
Hit:4 https://deb.debian.org/debian bullseye-updates InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Get:6 https://download.opensuse.org/repositories/home:/jcorporation/Debian_11  InRelease [1532 B]
Hit:7 https://deb.debian.org/debian bullseye-backports InRelease
Hit:3 https://mirrors.jevincanders.net/armbian/apt bullseye InRelease
Fetched 9037 B in 5s (1910 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
[Sun Dec 18] root@batopi: ~$
```

jalsco, 2022-12-18:
I tried to restart the service 
```
[FAILED] DietPi-Services | ● mympd.service - myMPD server daemon
     Loaded: loaded (/lib/systemd/system/mympd.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2022-12-18 05:42:53 EST; 15s ago
       Docs: man:mympd(1)
    Process: 6225 ExecStart=/usr/bin/mympd (code=exited, status=1/FAILURE)
   Main PID: 6225 (code=exited, status=1/FAILURE)
        CPU: 141ms

Dec 18 05:42:53 batopi mympd[6225]: NOTICE   mympd     Work dir: "/var/lib/mympd"
Dec 18 05:42:53 batopi mympd[6225]: NOTICE   mympd     Config dir: "/var/lib/mympd/config"
Dec 18 05:42:53 batopi mympd[6225]: NOTICE   mympd     Setting loglevel to NOTICE
Dec 18 05:42:53 batopi mympd[6225]: NOTICE   mympd     Starting myMPD 10.1.6
Dec 18 05:42:53 batopi mympd[6225]: 3bbe7e 1 mongoose.c:4201:mg_open_listen bind: 98
Dec 18 05:42:53 batopi mympd[6225]: 3bbe7e 1 mongoose.c:3458:mg_listen      Failed: http://0.0.0.0:80, errno 98
Dec 18 05:42:53 batopi mympd[6225]: EMERG    mympd     Can't bind to http://0.0.0.0:80
Dec 18 05:42:53 batopi mympd[6225]: Exiting erroneous, thank you for using myMPD
Dec 18 05:42:53 batopi systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILURE
Dec 18 05:42:53 batopi systemd[1]: mympd.service: Failed with result 'exit-code'.
```

jalsco, 2022-12-18:
Exactly which  files and folders (and anything else ?)  should I remove ? How can I determine what should be removed ?  I did actually ask about this a couple weeks ago, when the "updated mympd" would not install successfully onto my 8.11 system. I asked if any extra steps were required. The short answer given ( @Joulinar ) was to just wait for 8.12 and it would all work 🤷🏻‍♂️ 

`mympd` has never been successfully installed on my system. My system does not think it is installed.   Anyway - please advise - which files / folders / and anything-else ?  that I need to remove. 

I will complete all removal steps before trying to install again.

jalsco, 2022-12-18:
```
[Sun Dec 18] root@batopi: ~$ rm -Rf /var/lib/mympd
[Sun Dec 18] root@batopi: ~$ rm -Rf /var/lib/private/mympd
```
```
[Sun Dec 18] root@batopi: ~$ systemctl stop mympd.service
[Sun Dec 18] root@batopi: ~$ systemctl disable mympd.service
```
```
[Sun Dec 18] root@batopi: ~$ rm /usr/lib/systemd/system/mympd.service
rm: remove regular file '/usr/lib/systemd/system/mympd.service'? y
[Sun Dec 18] root@batopi: ~$ systemctl daemon-reload
```

Anything else needed before attempt to install `mympd` ?

Joulinar, 2022-12-18:
hopefully following will clean it

```
systemctl unmask mympd
systemctl disable --now mympd
apt purge mympd
rm /etc/apt/sources.list.d/dietpi-mympd.list
rm /etc/apt/trusted.gpg.d/dietpi-mympd.gpg
rm -Rf /var/lib/mympd
rm -Rf /var/lib/private/mympd
rm -Rf /var/cache/private/mympd
rm -Rf /var/cache/mympd
rm -f /usr/share/man/man1/mympd* /etc/mympd.conf*
```

MichaIng, 2022-12-19:
Ah right, if the `mympd -c` (config creation during install) didn't succeed, but the service did start, default configs are created. And the install won't overwrite existing configs. Joulinars cleanup steps and a new install attempt should finally work.

jalsco, 2022-12-19:
I did all those removal steps.
Then I installed it - using dietpi-software.
This time - installed successfully - and now the mympd service is Running [ ok ] 

However - nothing showing up on port 1333
Also there is no config file at:  `/etc/mympd.conf`

MichaIng, 2022-12-19:
Port 1333 should be the HTTPS port, 1332 the plain HTTP port. It does not listen to it? `ss -tlpn`

`mympd -c` runs as part of the installation already. `/etc/mympd.conf` does not exist anymore but all settings can be found below `/var/lib/mympd/config`, resp. `/var/lib/private/mympd/config`.

Yes, docs need an update. Since I'm on vacation, I only have sporadic time/access to work on this.

Joulinar, 2022-12-19:
I started working to update docs https://github.com/MichaIng/DietPi-Docs/pull/810

MichaIng, 2022-12-23:
I'll mark this as closed. Feel free to reopen if required.

closed 2022-12-23 by MichaIng

jcorporation, 2022-12-23:
For v10.2.0 the behavior will chang back to old behavior if the user does not exist. 

You should use `systemd-run` to create the initial configuration. This should also work with current v10.1.x versions.

```
systemd-run -t -p DynamicUser=yes -p StateDirectory=mympd -p CacheDirectory=mympd debug/bin/mympd -c
```