MichaIng/DietPi #5992: Readarr fails to install

lambolighting, 2022-12-19:
#### Details:
- Date           | Mon Dec 19 01:08:02 GMT 2022
- DietPi version | v8.11.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux Atlas 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `mv Readarr /opt/readarr`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Installing readarr through dietpi-software
#### Expected behaviour:

- Readarr get installed.
#### Actual behaviour:

- It exits with error code that no such file or directory is found.
#### Extra details:

- ...
#### Additional logs:
```
mv: cannot stat 'Readarr': No such file or directory
```

Joulinar, 2022-12-19:
Hi,

unfortunately, a bug on our end. We added an incorrect download link. 🙄

MichaIng, 2022-12-20:
Live patch merged.

closed 2022-12-20 by MichaIng