MichaIng/DietPi #5994: Radxa Zero Update to v8.12.1 fails (w/ possible solution)

justlep, 2022-12-19:
#### Required Information
- Date           | Mon Dec 19 18:48:50 CET 2022
- DietPi version | v8.12.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Radxa Zero (aarch64) (ID=74)
- Kernel version | `Linux zero 5.19.17-meson64 #22.11.1 SMP PREEMPT Wed Nov 30 11:05:42 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `dtc -I dts -O dtb -o /boot/overlay-user/dietpi-usb-gadget.dtbo /boot/overlay-user/dietpi-usb-gadget.dts`
- Exit code      | 1
- Software title | DietPi-Patch

#### Steps to reproduce

Run `sudo dietpi-update`

#### Expected behaviour

Dietpi updates to v8.12.1

#### Actual behaviour

Update fails with 
```
FATAL ERROR: Couldn't open "/boot/overlay-user/dietpi-usb-gadget.dts": No such file or directory
```

#### Possible Solution?

There is a file called `dietpi-usb-otg.dts` instead of `dietpi-usb-gadget.dts` in that folder, so I tried

```shell
cp /boot/overlay-user/dietpi-usb-otg.dts /boot/overlay-user/dietpi-usb-gadget.dts
```

Dietpi update then succeeded, USB still works fine after reboot in v8.12.1.

So, was this really just a filename issue?

Joulinar, 2022-12-19:
Probably, @MichaIng would need to have a look

MichaIng, 2022-12-20:
Fix merged. @justlep the file was intended to be called `dietpi-usb-otg.dts`. Could you retest whether the update goes not through by reapplying it?
```sh
sed -i 's/gadget/otg/' /boot/dietpiEnv.txt
mv /boot/overlay-user/dietpi-usb-{gadget,otg}.dts
mv /boot/overlay-user/dietpi-usb-{gadget,otg}.dtbo
dietpi-update -1
```

justlep, 2022-12-20:
I have retestested with your steps - all looking good now. Big thanks @MichaIng , enjoy your holidays :)

MichaIng, 2022-12-21:
Great, many thanks for testing. I'll mark this as closed then.

closed 2022-12-21 by MichaIng