MichaIng/DietPi #5995: Trying to update to from 8.11.2 to 8.12.1

Psych85, 2022-12-19:
#### Details:
- Date           | Mon Dec 19 21:22:32 GMT 2022
- DietPi version | v8.11.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. Run dietpi-update
2. ...
#### Expected behaviour:

- update to next version. 
#### Actual behaviour:

- fails when attempting to suspend time syncs.service which is not present. 
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```

Joulinar, 2022-12-19:
Question: did you have done an upgrade from Buster to Bullseye recently?

To fix:

```
apt install systemd-timesyncd
```

Psych85, 2022-12-20:
Thank you for your response. You are correct in that I have recently upgraded to Bullseye. I have installed the service and the update commences but now there appears to be another issue to do with some of the repositories:

#### Details:
- Date           | Tue Dec 20 08:46:34 GMT 2022
- DietPi version | v8.11.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Ign:1 https://apt.sonarr.tv/debian bullseye InRelease
Err:2 https://apt.sonarr.tv/debian bullseye Release
  404  Not Found [IP: 37.187.255.43 443]
Get:3 https://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian bullseye InRelease [15.0 kB]
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Ign:5 https://download.mono-project.com/repo/debian raspbianbullseye InRelease
Err:6 https://download.mono-project.com/repo/debian raspbianbullseye Release
  404  Not Found [IP: 152.199.19.161 443]
Reading package lists...
E: The repository 'https://apt.sonarr.tv/debian bullseye Release' does not have a Release file.
E: The repository 'https://download.mono-project.com/repo/debian raspbianbullseye Release' does not have a Release file.
```

Joulinar, 2022-12-20:
Did you follow our upgrade guide? https://dietpi.com/blog/?p=811
Or something else?

Psych85, 2022-12-20:
That's the one I followed and used the manual steps. Perhaps I did something wrong regarding the package sources and need to look at it again?

MichaIng, 2022-12-20:
It seems you missed that one 🙂:
```sh
apt -y install fdisk systemd-timesyncd
```

Joulinar, 2022-12-20:
> Perhaps I did something wrong regarding the package sources and need to look at it again?

Well, how to say 🤣 

- 2.2.3 Upgrade the system
> As time of writing, the following repositories are known to not provide Bullseye packages yet: Mono & Sonarr.

Means you need to switch back source files for both as they don't offer Bullseye repository.

- 2.2.4 Software migrations
There is a larger block of commands. One of them would reinstall `systemd-timesyncd`

> `apt -y install fdisk systemd-timesyncd`

Psych85, 2022-12-20:
Many thanks guys. Adjusted the sources and now just running the update. All going smoothly like a wet seals back. Thank you for your help and support and keep up the great work!

closed 2022-12-20 by Psych85