MichaIng/DietPi #5997: Only a question

MDAR, 2022-12-20:
I would like to run a script to setup an odroid C4

Everything seems to be fine.

However

Adding in things like

```dietpi-software install 0 105 122 122 201``` 
doesn't work within a script

The terminal just reports that ```dietpi-software``` can't be found

I assume that is intentional

Is there a work around, so "is that how it is"

Either way is fine, as long as I know

Thanks a million

Joulinar, 2022-12-20:
Usually software installation during first boot can be specified within dietpi.txt. 

But for your questions. You need to call `dietpi-software` with full path `/boot/dietpi/`

MDAR, 2022-12-20:
> Usually software installation during first boot can be specified within dietpi.txt.
> 
> But for your questions. You need to call `dietpi-software` with full path `/boot/dietpi/`

Thanks that makes a lot of sense.

I'll add the software IDs to dietpi,txt

closed 2022-12-20 by MDAR