MichaIng/DietPi #5998: Update to 8.12.1 fails on UEFI x86_64

dequeant, 2022-12-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bullseye
- 
- Kernel version | `uname -a`
- Linux Dellpi 5.10.0-19-amd64 #1 SMP Debian 5.10.149-2 (2022-10-21) x86_64 GNU/Linux
- x86_64 UEFI
- Power supply used | ATX PSU
- SD card used | N/A SATA SSD

#### Steps to reproduce
run dietpi-update

#### Expected behaviour
Update from 8.11.2 to 8.12.1

#### Actual behaviour
Update fails with the following message:
│ APT upgrade
│  - Command: apt-get -y upgrade
│  - Exit code: 100
│  - DietPi version: v8.11.2 (MichaIng/master) | HW_MODEL: 21 | HW_ARCH: 10 |
│ DISTRO: 6
│  - Image creator: DietPi Core Team
│  - Pre-image: from scratch
│  - Error log:
│ Reading package lists...
│ Building dependency tree...
│ Reading state information...
│ Calculating upgrade...
│ The following packages have been kept back:
│   linux-image-amd64
│ 0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
│ 1 not fully installed or removed.
│ After this operation, 0 B of additional disk space will be used.
│ Setting up grub-pc (2.06-3~deb11u5) ...^M
 Installing for i386-pc platform.^M
│ grub-install: warning: this GPT partition label contains no BIOS Boot
│ Partition; embedding won't be possible.^M
│ grub-install: error: embedding is not possible, but this is required for
│ cross-disk install.^M
│ You must correct your GRUB install devices before proceeding:^M
│ ^M
│   DEBIAN_FRONTEND=dialog dpkg --configure grub-pc^M
│   dpkg --configure -a^M
│ dpkg: error processing package grub-pc (--configure):^M
│  installed grub-pc package post-installation script subprocess returned
│ error exit status 1^M
│ Errors were encountered while processing:^M
│  grub-pc^M
│ E: Sub-process /usr/bin/dpkg returned an error code (1)

#### Extra details
 Reference bug report: 608001d8-b931-4053-b0e7-f401967c680b

MichaIng, 2022-12-21:
Many thanks for reporting.

You btw do not use an UEFI image, but a BIOS image. The GRUB package needs to know where to flash GRUB to. We ship our BIOS images with `/dev/sda` as stored system/GRUB drive, which is correct in 95% of cases. In your case, based on your bug report upload, however, a data drive is having this name and the system drive is `/dev/sdb` instead. The GRUB package isn't smart enough to loop through other drives, or the one of the root filesystem at least, to look where to flash GRUB to, so you need to tell it interactively:
```sh
dpkg --configure grub-pc
```
When being asked, enter or select `/dev/sdb`. Afterwards, the update should go through.

dequeant, 2022-12-23:
> 

Thanks.  That worked a treat.  I could have sworn I used the UEFI image, but I also remember having issues with it, so it looks like I ended up going with the bios image.  Apologies for any confusion..... its been a while!

Joulinar, 2022-12-23:
Going to close the issue for now. Feel free to reopen if needed.

closed 2022-12-23 by Joulinar