MichaIng/DietPi #6001: There seems to be a serious problem with the functioning of the hostapd for WiFi HotSpot

abassign, 2022-12-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=12
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
`
- Distro version | `bullseye 0`
- Kernel version | `uname -a`
- SBC model | `Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`
- Power supply used | 5V 5A 
- SD card used |Transcend Micro UI HC - but I also tried with other cards

#### Additional Information (if applicable)
- Software title | the problem is highlighted when trying to start "hostapd" configured with software id = 60 (HotSpot)
- Was the software title installed freshly or updated/migrated | They are all fresh installations, I download the image, I do the basic configuration (password and IT keyboard)
- Can this issue be replicated on a fresh installation of DietPi? | They are all very fresh installations ... as I'm setting up a hotspot for my electronics lab.

#### Steps to reproduce

I make a premise: I'm a computer scientist who has been using Dietpi Raspberry machines for 5 years.
I have installed DietPi on dozens of Raspberry's needed for my R&D lab.
I have installed 5 Accespoint WiFi HotSpots in recent years, the last installation took place in July 2022 (6 months ago), the installation did not report any problems.
The procedure is to activate the software with id=60 with behind-software and that's it.

Download the image from your site, I tried with three different Raspberry:
a) an old 2 ver. 1.1 with WiFi Tp-Link on USB. I downloaded the DietPi_RPi-ARMv6-Bullseye.img image downloaded on 2022-12-21
b1) a newer Raspberry 3 Model B+ with DietPi_RPi-ARMv7-Bullseye.img image downloaded on 2022-12-21
b2) always on Raspberry 3 Model B+ with DietPi_RPi-ARMv8-Bullseye.img image downloaded on 2022-12-21
c) Raspberry 4 with 4GB with DietPi_RPi-ARMv8-Bullseye.img image downloaded on 2022-12-21

To date (I've been trying all day) I've never had any problems.

#### Expected behaviour

The installation up to the last time has not given any problems and has always activated an excellent HotSpot connection.

#### Actual behaviour

`● isc-dhcp-server.service - LSB: DHCP server
     Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
     Active: failed (Result: exit-code) since Wed 2022-12-21 18:41:15 CET; 25ms ago
       Docs: man:systemd-sysv-generator(8)
    Process: 1773 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)
        CPU: 81ms

Dec 21 18:41:13 DietPi dhcpd[1788]: before submitting a bug.  These pages explain the proper
Dec 21 18:41:13 DietPi dhcpd[1788]: process and the information we find helpful for debugging.
Dec 21 18:41:13 DietPi dhcpd[1788]: 
Dec 21 18:41:13 DietPi dhcpd[1788]: exiting.
Dec 21 18:41:15 DietPi isc-dhcp-server[1773]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Dec 21 18:41:15 DietPi isc-dhcp-server[1793]:  failed!
Dec 21 18:41:15 DietPi isc-dhcp-server[1794]:  failed!
Dec 21 18:41:15 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Dec 21 18:41:15 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Dec 21 18:41:15 DietPi systemd[1]: Failed to start LSB: DHCP server.
Setting up libnetfilter-conntrack3:arm64 (1.0.8-3) ...
Setting up iptables (1.8.7-1) ...
` 
`
DietPi-Software
─────────────────────────────────────────────────────
 Step: Finalising install

[  OK  ] DietPi-Software | systemctl daemon-reload
2022-12-21 18:41:26 [ INFO ] DietPi-RAMlog | Storing /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store...
2022-12-21 18:41:26 [  OK  ] DietPi-RAMlog | Stored /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store.
[ SUB1 ] DietPi-Services > dietpi_controlled 
[  OK  ] DietPi-Services | dietpi_controlled : isc-dhcp-server
[  OK  ] DietPi-Services | dietpi_controlled : hostapd
[  OK  ] DietPi-Services | dietpi_controlled : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed

[  OK  ] DietPi-Survey | Purging survey data
[ SUB1 ] DietPi-Services > restart 
[FAILED] DietPi-Services | restart : isc-dhcp-server
[FAILED] DietPi-Services | restart : hostapd
[  OK  ] DietPi-Services | restart : cron
` 
He says he has problems on both modules: "isc-dhcp-server" and
"hostapd", but actually it's the "hostapd" module that doesn't work.

#### Extra details

I enter additional data:
`
dietpi@DietPi:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 009: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 008: ID 04f2:3f41 Chicony Electronics Co., Ltd HP Wired Desktop 320K Keyboard
Bus 001 Device 007: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
```
dietpi@DietPi:~$ sudo lsmod
Module                  Size  Used by
xt_state               16384  0
xt_conntrack           16384  2
nft_counter            16384  6
nft_chain_nat          16384  2
xt_MASQUERADE          16384  2
nf_nat                 49152  2 nft_chain_nat,xt_MASQUERADE
nf_conntrack          139264  4 xt_conntrack,nf_nat,xt_state,xt_MASQUERADE
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
nft_compat             20480  4
nf_tables             221184  29 nft_compat,nft_counter,nft_chain_nat
nfnetlink              20480  2 nft_compat,nf_tables
brcmfmac              335872  0
brcmutil               24576  1 brcmfmac
cfg80211              868352  1 brcmfmac
rfkill                 36864  2 cfg80211
r8188eu               598016  0
rpivid_hevc            57344  0
v4l2_mem2mem           40960  1 rpivid_hevc
videobuf2_dma_contig    24576  1 rpivid_hevc
videobuf2_memops       20480  1 videobuf2_dma_contig
videobuf2_v4l2         32768  2 rpivid_hevc,v4l2_mem2mem
videobuf2_common       69632  5 videobuf2_dma_contig,videobuf2_v4l2,rpivid_hevc,v4l2_mem2mem,videobuf2_memops
videodev              262144  4 videobuf2_v4l2,videobuf2_common,rpivid_hevc,v4l2_mem2mem
mc                     61440  5 videodev,videobuf2_v4l2,videobuf2_common,rpivid_hevc,v4l2_mem2mem
joydev                 24576  0
raspberrypi_hwmon      16384  0
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
nvmem_rmem             16384  0
fuse                  135168  1
ip_tables              32768  0
x_tables               49152  5 xt_conntrack,nft_compat,xt_state,ip_tables,xt_MASQUERADE
ipv6                  561152  18
```

```
sudo journalctl -u hostapd
dic 21 19:23:11 DietPi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 1111.
dic 21 19:23:11 DietPi systemd[1]: Stopped Access point and authentication server for Wi-Fi and Ethernet.
dic 21 19:23:11 DietPi systemd[1]: Starting Access point and authentication server for Wi-Fi and Ethernet...
dic 21 19:23:11 DietPi hostapd[3375]: Configuration file: /etc/hostapd/hostapd.conf
dic 21 19:23:11 DietPi hostapd[3375]: nl80211: Driver does not support authentication/association or connect commands
dic 21 19:23:11 DietPi hostapd[3375]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
dic 21 19:23:11 DietPi hostapd[3375]: nl80211 driver initialization failed.
dic 21 19:23:11 DietPi hostapd[3375]: wlan0: interface state UNINITIALIZED->DISABLED
dic 21 19:23:11 DietPi hostapd[3375]: wlan0: AP-DISABLED
dic 21 19:23:11 DietPi hostapd[3375]: wlan0: CTRL-EVENT-TERMINATING
dic 21 19:23:11 DietPi hostapd[3375]: hostapd_free_hapd_data: Interface wlan0 wasn't started
dic 21 19:23:11 DietPi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
dic 21 19:23:11 DietPi systemd[1]: hostapd.service: Failed with result 'exit-code'.
dic 21 19:23:11 DietPi systemd[1]: Failed to start Access point and authentication server for Wi-Fi and Ethernet.

```

I hope I have given enough information, obviously I'm still with my new HotSpot and I don't update the old HotSpots done in the previous months to avoid possible blocks.
One note that upset me a lot:
I tried installing a Raspberry OS version:
"2022-09-22-raspios-bullseye-arm64-lite.img" on the Raspberry 3 Model B+ that I had already successfully used as a HotSpot.
I HAD THE SAME PROBLEM, always seems related to the hostapd module.

I also tried with various 5V power supplies that I use with the Raspberry in the lab.

Joulinar, 2022-12-21:
Just tested on my RPi4B and it's working without issues.

`isc-dhcp-server`, error message during install is a litte bit expected, but service should be working after reboot.

`hostapd`, did you manually enable onboard Wi-Fi before trying to install Hotspot?

abassign, 2022-12-21:
@Joulinar 
Hi, indeed the "isc-dhcp-server" module starts regularly after the reboot, at least apparently, while the hostapd remains failed.
If, on the other hand, I activate WiFi as a client, everything works perfectly and I can connect to any HotSpot. I think that's why until now, after 6 months since I last activated the HotSpot, I never noticed the problem.
I've been looking all day on the net but haven't found anything specific, I think it would be important to try with a raspberry to see if you can reproduce my problem.

MichaIng, 2022-12-22:
This seems to be the main problem:
```
nl80211: Driver does not support authentication/association or connect commands
```
AFAIK, this is the issue with these RTL8188CU* chips, where we used to ship even a dedicated hostapd build with "rtl871xdrv" driver. However, the chip is rarely used nowadays, and the hostapd build from RPi repo used to solve it differently internally already, making maintenance efforts just for a few cases of these old incompatible USB WiFi dongles reasonless.

However, RPi onboard WiFi is Broadcom, still I see the `r8188eu` module loaded. On RPi 2 the dongle seems to be the problematic chip, on RPi 3 and 4, you do use onboard WiFi, don't you? There it should work, doesn't it?

It seems the RPi repo is also not providing the own/patched hostapd build anymore, only for Stretch: https://archive.raspberrypi.org/debian/pool/main/w/wpa/
Not sure whether this changed recently, but might be an explanation why it did work before. So no one finds it reasonable to build and host extra builds for this old chip => you need to replace it if needed for hostapd 😉.

abassign, 2022-12-22:
@MichaIng Indeed this could be the reason, but the RTL8188CU chip is present in all the Raspberry I have in the lab both the PI3 mode B+ and the various 4 ... But the boring thing is that this chip is also present in the USB WiFi dongle of the Tp-Link with IC 8853A https://industrycanada.co/8853A-WN725N which I attempted to use. It could be a typically European product distribution problem. But, for example, the Raspberry 4 (due to its complete absence on the European market for months) was bought from China 6 months ago and has the same problem.
I'm curious if all Raspberry pre-2200 use the same chip for internal WiFi.
In the meantime, I get a newer WiFI dongle USB to see if the problem can be solved.
It could be useful to have a note on compatibility to allow us users to get the necessary material.

MichaIng, 2022-12-22:
The RPi 3 and 4 use BCM43438 (aka CYW43455) chip (on all revisions), not RTL8188CU*. So it should work fine with them. So far I see only the USB adapter with RTL8188EUS, on RPi 2, being problematic.

abassign, 2022-12-22:
@MichaIng 
To better answer your questions, I redid a new installation, by re-downloading the file and using the Raspberry PI 3 B+ which it already did from HotSpot until a week ago, when I decided to update the software. But the same thing also happens for another Raspberry Pi 3 B+ (2017) that I have and a Raspberry 4 purchased 6 months ago.

Execution of the start with the SD with DietPi version:
FILE:   DietPi_RPi-ARMv8-Bullseye.img DATE:   Tue Dec 20 05:06:40 UTC 2022

Login to start installation as root
I don't enter the new root and global password

Serial consolle can re-enabled at any time, respond [No]

The DietPi-Software menu il open I select the option Search Software
In the request: "Please enter a software title, ID or ... " I insert the number 60
In the request: "Please use the spacebar..." I push the space and I confirm the option

Start the check. In the pop-up "The following criteria must be met for installation of WiFi Hotspot to succeded" the voice "Ethernet online: PASSED" and the voice "WiFI adapter detected PASSED" and "PASSED:  Criteria met. Good to go" I push CR for confirmation.

After I push on Install selection

The pop-up write: "DietPi in now ready to apply your ..." I put the cursor on <OK> and I confirm the choice.
The installation of the modules proceeds on the Debian Bullseye repository. But while installing I see the error message:
```
	● isc-dhcp-server.service - LSB: DHCP server
	      Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
	      Active: failed (Result: exit-code) since Wed 2022-12-21 18:41:15 CET; 25ms Aug
	        Docs: man:systemd-sysv-generator(8)
	     Process: 1773 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)
	         CPU: 81ms 
```

Now I go ahead to the pop-up that ends with the phrase Would you like to join .. I answer the default "0: Opt OUT and purse uploaded data" I confirm with the <OK> key.

I go back to the tab login as root, now I do some checks.

In order to better follow the messages, I enter via SSH terminal on the test Raspberry 3 B+.

```
dietpi@DietPi:~$ lsusb 
Bus 001 Device 007: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter 
Bus 001 Device 006: ID 04f2:3f41 Chicony Electronics Co., Ltd HP Wired Desktop 320K Keyboard 
Bus 001 Device 004: ID 0409:005a NEC Corp. HighSpeed Hub 
Bus 001 Device 005: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC)  
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub 
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
```
dietpi@DietPi:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:c5:3c:1a brd ff:ff:ff:ff:ff:ff
    inet 192.168.20.123/24 brd 192.168.20.255 scope global dynamic eth0
       valid_lft 84736sec preferred_lft 84736sec
    inet6 fe80::ba27:ebff:fec5:3c1a/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 28:ee:52:03:59:e8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.1/32 scope global wlan0
       valid_lft forever preferred_lft forever
5: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:90:69:4f brd ff:ff:ff:ff:ff:ff
```
 
I see that the wlan0 has its network address 192.168.42.1/32 as I have always seen in the other installations I have done in previous years and up to 6 months ago.
[... 101 lines cut ...]
Dec 22 13:27:06 DietPi systemd[1]: Failed to start Access point and authentication server for Wi-Fi and Ethernet.

```

As I have already written, the Raspberry in use is a 3B+ from 2017, while the Raspbeery 4 (which I have not illustrated in this test, but gives the exact same messages).

I don't know what other tests I can do, but I think it would be important to do a parallel test with a Raspberry 3B+ or with a 4. I also tried version 7 which is usable for Raspberry 3B+, but I had exactly the same negative results.

As I wrote to you also the tp-link dongle I used (bought less than a year ago USB 2 type) uses exactly the same chip as I show you:

```
Bus 001 Device 009: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 007: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 005: ID 04f2:3f41 Chicony Electronics Co., Ltd HP Wired Desktop 320K Keyboard
Bus 001 Device 004: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 006: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

They are in fact indistinguishable
This is why I wonder if it is correct to consider the chip obsolete, given that it is still sold on Amazon and is present in the Raspberry 3 and 4.

Write to me if you think more tests are needed, there is no problem for me to go ahead with the tests.

abassign, 2022-12-22:
> The RPi 3 and 4 use BCM43438 (aka CYW43455) chip (on all revisions), not RTL8188CU*. So it should work fine with them. So far I see only the USB adapter with RTL8188EUS, on RPi 2, being problematic.

In may Raspberry Pi 3B+ an 4 i found this:
Bus 001 Device 007: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

At this point I would not like there to be different supplies for different geographical areas, the doubt is that other devices, like my dongle that I tested yesterday, in Italy (or southern Europe?) use these chips too. A very strange fact that I can't understand, even if there are quite precise international rules in the world of WiFi.
The knock thing is that if I reported the problem I wouldn't want to be the only one in Eoropa using a HotSpot with SW DietPi

Joulinar, 2022-12-22:
Can you share following from your RPi4? 

```
journalctl -b | grep brcmfmac
```

For me it returns `brcmfmac43455` as used WiFi chip

```
root@DietPi4:~# journalctl -b | grep brcmfmac
Dec 22 16:11:39 DietPi4 kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Dec 22 16:11:39 DietPi4 kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Dec 22 16:11:39 DietPi4 kernel: usbcore: registered new interface driver brcmfmac
Dec 22 16:11:40 DietPi4 kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Dec 22 16:11:40 DietPi4 kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
Dec 22 16:11:40 DietPi4 kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
root@DietPi4:~#
```

MichaIng, 2022-12-23:
```console
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 28:ee:52:03:59:e8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.1/32 scope global wlan0
       valid_lft forever preferred_lft forever
5: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:90:69:4f brd ff:ff:ff:ff:ff:ff
...
Bus 001 Device 009: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 007: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```
So you initially had a USB WiFi adapter attached to the RPi 3+ already, and when you attached the TP-Link, two USB WiFi adapters (with same chip) were attached.

Surprising is that the initial USB WiFi adapter seems to be interface `wlan0` while onboard then is `wlan1`, which is usually the other way round in which case onboard WiFi would be used. The first USB adapter is probably attached via USB hub or such, that you forgot about it? Easiest solution would be to remove it, so onboard WiFi is used. Otherwise, the hotspot can be as well configured to use the `wlan1` interface.

Some steps can be taken/derived from here, but let me wrap this up when required: https://dietpi.com/forum/t/dietpi-as-wifi-router/14679/33?u=michaing

abassign, 2023-01-10:
I think I have solved the problem, it was related to the fact that the DHCP server of the network where the router was installed was not reachable.
The DHCP server, for the installation of the isc-dhcp-server module requires to know a DHCP server in the network, if this, for various reasons, is not known, the installation does not proceed correctly.
It was enough to ping the DHCP server (which is not that of the router of the network that hosts the Raspberry with Dietpi) to rebuild the internal table and obtain the correct functioning of the service. This problem doesn't happen if the DHCP server has the same address as the router, which is common in small networks. In my case, being a corporate network, the DHCP server has a different IP.

MichaIng, 2023-01-10:
> The DHCP server, for the installation of the isc-dhcp-server module requires to know a DHCP server in the network, if this, for various reasons, is not known, the installation does not proceed correctly.

Not sure whether I understand correctly? The WiFi hotspot is meant to run on a dedicated network, otherwise you'd have two DHCP servers conflicting with each other. When WiFi clients connect to the WiFi hotspot, only DietPi's `isc-dhcp-server` is meant to answer to clients DHCP requests. The router's DHCP server must not be on the same network, otherwise clients may get conflicting IP addresses.

abassign, 2023-01-11:
Thank you for the note, it is the same issue that I discussed with the CED
manager of the company where DietPi WiFi routers are to be installed. This
company has a DHCP configuration that has taken a lot of pain due to the
fact that they have hundreds of operating machines in production all
connected and there is a feeling that there is something wrong. For this
they positioned the DHCP no longer with the address of the router
(192.168.20.201 in my subnet) but with a single DHCP with the address
192.168.1.2) The DHCP positioned my DietPi router at the address
192.168.20.126 during the WiFi router installation (id: 60).
Since after two weeks of hiatus, due to the Christmas holidays, we resumed
activity and the WiFi router that I had configured before the holidays WAS
WORKING .. I deduced the following things:

1 - It's not a problem with the WiFi card (in fact I also tried with other
types of Raspberry) which confirmed the problem, and before December the
Raspberry I'm testing worked normally as a WiFi router.
2 - The problem occurs during installation on the module
isc-dhcp-server.service.
3 - I also encountered the same problem by making a WiFi router with the OS
distributed by Rasberry.
4 - I reconfigured the working Router again (by uninstalling the 60
package, restarting it and then reinstalling the 60 package) and the same
problem occurred again.

So I have a good chance that the problem is with our company's DHCP, so I
need to talk to the network manager (DHCP machine 192.168.1.2 is a Windows
Server machine). The problem package is very likely to be the
isdc-dhcp-server and therefore I can shift the problem to whoever manages
this package.
Obviously, if I find a solution, I can report what I discovered in this
forum, although I believe it is a problem already reported by others who
operate in more complex network environments.

Greetings
Adriano Bassignana

abassign, 2023-01-11:
> Our WiFi hotspot package sets up the network at 192.168.42.0/24 subnet,
> which seems to not collide with any of the companies other subnets
> 192.168.20.0/24 and 192.168.1.0/24. Probably it's easiest to leave it
> like that?
>
> The problem package is very likely to be the
> isdc-dhcp-server and therefore I can shift the problem to whoever manages
> this package.
>
> It is extremely unlikely that the isc-dhcp-server package is the issue
> (that you found a bug in this package), but a configuration issue within
> the network.
>
> The problem which shows up obvious to me is that you still have a USB WiFi
> adapter attached to this RPi which does not support hostapd (RTL8188EUS
> chip), that's all. Unplug this USB WiFi adapter, restart the system and all
> issues should be gone.

Hi, at first I thought it was a good hypothesis but there are some points
that exclude it:
1. Until mid-December, the Raspberry 3+ worked without problems with its
internal WiFi in the company network 192.168.20.0. Then I did the Hot-Spot
software update and it no longer worked.
2. In my house, with another router and another network configuration
(192.168.0.0), the same device works fine.
3. I also did the tests with the 2019 Raspberry 4 and the exact same
problem.
4. Strangely yesterday it worked perfectly within the corporate network
192.168.20.0 but as soon as I redid the installation (to understand better)
it didn't work anymore.
So I have to completely rule out hardware-type problems, and focus on the
fact that the corporate DHCP doesn't have the same network address as the
default-gateway (but the address: 192.168.1.2), my feeling is that the DHCP
is the problem, it is no coincidence that the problem is highlighted during
installation, as already explained.
Today I learned that the corporate DHCP 192.168.1.2 has been active for
more than two years and therefore I should have intercepted the problem
immediately.
The only coincidence is precisely related to the software update. The same
problem also occurs with the software released by the OS distributed by
Raspberry, this makes me believe that it is a problem with the DHCP
management package.

abassign, 2023-01-12:
> As long as you still see this device in your lsusb output, it's a fact,
> not a hypothesis:
> Which of the two adapters gets wlan0 (used by the WiFi hotspot) and which
> one wlan1 might be a matter of randomness at boot, so that it sometimes
> works (if onboard WiFi gets wlan0 interface) and sometimes not. The MAC
> address can be used to identify which adapter is behind which interface:

Hello,
I found who was causing the problem, I discovered that in the monitor used
as a display for the Raspberry setup in the USB, placed next to the USB for
the keyboard, there was a TpLink USB Dongle which was the one you
highlighted:
Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Removed the dongle everything went fine!
In reality, the drive for its management is not present, but its presence
put the management of the hot-spot in crisis. Evidently there are parts of
code that, even if the drive is not present, do something that is not
satisfied and therefore lead to this problem.

I want to thank you for your assistance and patience.
On the other hand, I'll be more careful of traps like these that can lead
to long searches for problems that are actually due to forgotten pieces of
hardware.

I think at this point we can close the incident. But in parallel I ask
which drive can I use in dietPi to be able to use this device correctly?

Greetings
Hadrian

Joulinar, 2023-01-12:
going to close the issue for now.

closed 2023-01-12 by Joulinar

abassign, 2023-01-12:
Great, thanks!
:)

MichaIng, 2023-01-12:
Great that you found it. Quite some coincidence that 1. there was an overseen USB WiFi adapter attached and 2. that it was one if the (rare) models which do not support hotspot usage.

> But in parallel I ask
which drive can I use in dietPi to be able to use this device correctly?

Not sure if I understand correctly, you mean which drive**r** for RTL8188EUS to use it as hotspot? I am not aware of any maintained driver which does. There were some in the past, used by Raspberry Pi provided `hostapd` package, but all I can find now is very outdated, likely breaks to compile with recent toolkits and/or does not work with current kernel, which is likely why Raspberry Pi removed it from their repo:
- https://github.com/jekader/hostapd-rtl
- https://github.com/PlantDaddy/hostapd-rtl8188
- https://github.com/jenssegers/RTL8188-hostapd
- ... forget it, no possible or not worth the hassle 😄