MichaIng/DietPi #6002: Add a start delay to Chromium to account for HTTP Server boot up time

MDAR, 2022-12-22:
#### Is your feature request related to a problem? Please describe:

- Aligning Chromium start up with the availability of the web server, primarily when using some HA software, like openHAB
- A tiny irritation that I have solved by starting Chromium from Node-RED when that is up and can check the service state of openHAB
- Or by simply starting Chromium from Node-RED, when that is the app generating the HTML.
- However when running Chromium Autostart from the DietPi config option, it is always too quick and therefore can't fetch the page to load

#### Describe the solution you'd like:

- What I have done to cure this is to add a delay / sleep before the last exec command in the autostart file.

#### Additional context

- The lines I have added to `chromium-autostart.sh`

```
# start up delay
DELAY=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_DELAY=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
```

and at the end, just before the `exec` line

```
echo Will try starting Chromium with this command after a delay
echo $STARTX $FP_CHROMIUM $CHROMIUM_OPTS $URL
echo waiting for openHAB to start - $DELAY delay
sleep $DELAY
```

in the Chromium section of `dietpi.txt` I have added
```SOFTWARE_CHROMIUM_DELAY=3m```

Could this be added to the dietpi-config > autostart setup menu for Chromium (or others) ?

MichaIng, 2022-12-22:
Thanks for sharing your solution.

Isn't Chromium trying to reload the page automatically a few times, bevor it gives up? I know this from Opera (Chromium-based) at least.

Instead of a fixed delay, which may or may not work reliable, wouldn't it be better to have it waiting for a particular service or process, or if the process starts listening too long after its start, for a particular TCP port to be listened to?
```sh
ss -tln | grep ':1234'
```

MDAR, 2022-12-22:
> Isn't Chromium trying to reload the page automatically a few times, bevor it gives up?

Nop, just seems to try once and give the "no can do matey" screen

>Instead of a fixed delay, which may or may not work reliable

Depends on how long the delay is..... 30m would be excessive... but effective

I've noted that 3 minutes is more than enough for openHAB to get it's act together.. which I'm more than happy to set as a set time

> or if the process starts listening too long after its start, for a particular TCP port to be listened to?

I didn't even know that was an option

However...

For openHAB, port 8080 becomes live rather quickly with Jetty, but the apps and sub pages can take their sweet time
So a call to specific page will fail, even if port 8080 is up

MichaIng, 2022-12-23:
I see, so in this case its hard or impossible to find or check for the required condition. A static delay then is the only option.

MDAR, 2022-12-23:
> A static delay then is the only option

I could work out how to trigger it directly from openHAB if I need to.

I'm only thinking it is a simple solution for anyone else.

MichaIng, 2022-12-27:
I guess you are right. There are too many possible triggers based on individual apps to implement anything generic but a static delay.