MichaIng/DietPi #6003: dietpi-update resets the hostname to "DietPi"

infinitytec, 2022-12-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | CORE=8 SUB=12 RC=1
- Distro version | bullseye
- Kernel version | 5.15.76-v7+
- SBC model | RPi 2 Model B
- Power supply used | Probably a CanaKit one
- SD card used | I can't remember

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Set the hostname using `hostname newhostname`
2. Wait for an update to become available
3. Run `dietpi-update`
4. Check the hostname with `hostname`
5. At this point I am seeing it often reset to `DietPi`

#### Expected behaviour

- The hostname should stay to what it was manually set to

#### Actual behaviour

- Dietpi-update is resetting the hostname to the OOB value

Joulinar, 2022-12-22:
Hmm tested on six different device and none of them reset the hostname.

infinitytec, 2022-12-22:
Were they already up-to-date? I don't think I have the problem if no updates are run.

Joulinar, 2022-12-22:
My entire test lab started at DietPi v8.11 and I did the normal update to DietPi v8.12

infinitytec, 2022-12-22:
I may have not had it happen every time. It's been happening to me for a while (over a year) now across various installs... I'll keep an eye on it and see if it happens again.

For the record I also have Pi-hole and Unbound installed.

Thanks for looking into this.

MichaIng, 2022-12-23:
The only place where the hostname from `dietpi.txt` is set, is in the firstboot script: https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash
This however runs once on first boot of DietPi only, updates have no mechanism to call this script at all.

If you see this happening again, it would be interesting to know if `/etc/hostname` and `/etc/hosts` both reflect the change back to "DietPi", or `/etc/hostname` only, or even none of them but the output of the `hostname` command only.

And the devices you observed this, were they every rebooted after you changed the hostname? The only other idea I have is that the change was not written to disk, but kept in cache or some overlay filesystem and was hence lost on reboot. But of course that would mean that every other change (at least on the same filesystem) was lost as well, which you should have recognised.

I'm currently not sure whether Pi-hole has an option to set the hostname?

MichaIng, 2023-01-10:
`dietpi-config` sets the hostname via `hostname` command, to `/etc/hostname` and `/etc/hosts`, i.e. all relevant places. Feel free to report back it it is reset in any of those places.

omltcat, 2023-05-29:
In my case, just using `hostname NEW_HOSTNAME` command is not changing `/etc/hostname`, although running `hostname` again will return NEW_HOSTNAME. This will survive logout/login but not reboot.

Setting it in `dietpi-config` seems to do it permanently.

MichaIng, 2023-05-29:
Yes it's three things:
1. The `hostname` command sets is for the current boot session, not permanent.
2. `/etc/hostname` is where the kernel loads it from on boot, hence this makes it permanent.
3. `/etc/hosts` sets it as local DNS override, so that the hostname resolves to localhost.