MichaIng/DietPi #6004: Readarr install fails (installer attempts to install Prowlarr instead, and fails)

Mattheinrichs, 2022-12-22:
#### Details:
- Date           | Thu Dec 22 17:34:23 GMT 2022
- Bug report     | a5367a62-e844-45db-a47a-0f6453960314
- DietPi version | v8.11.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `mv Readarr /opt/readarr`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:
Attempt to install Readarr

#### Expected behaviour:

#### Actual behaviour:
The installer drops in Prowlarr instead - there is no Readarr folder to mv so the mv command fails

mv: cannot stat 'Readarr': No such file or directory

Joulinar, 2022-12-22:
Pls update to latest version of DietPi

Mattheinrichs, 2022-12-22:
That solved it

closed 2022-12-22 by Mattheinrichs