MichaIng/DietPi #6009: `CONFIG_CAN_GS_USB` is not set

clee, 2022-12-24:
#### Required Information
- DietPi version | 8.12.1-master
- Distro version | bullseye
- Kernel version | Linux quartz64 6.1.0-rc1 #1 SMP PREEMPT Wed Nov  9 01:08:14 UTC 2022 aarch64 GNU/Linux
- SBC model | Quartz64 (aarch64)
- Power supply used | 5V 3A DFRobot
- SD card used | SanDisk ultra

#### Steps to reproduce
1. Plug in a USB CAN adapter

#### Expected behaviour

- The `gs_usb` kernel module should load 
- I should see a `can0` network device when I run `ip link list`

#### Actual behaviour

- There is no `gs_usb` kernel module, and `zgrep GS_USB /proc/config.gz` reports `# CONFIG_CAN_GS_USB is not set`

Joulinar, 2022-12-24:
hmm we don't do any kernel development. Probably an upstream issue? What do you think @MichaIng

clee, 2022-12-24:
I think this is more of a kernel configuration issue; the module exists, it's just not being built/included here.

MichaIng, 2022-12-27:
Another limitations of Peter Geis' Quartz64 kernel. WireGuard and some BPF related modules are missing as well. Since he does not react to related merge requests, I guess we either need to compile the kernel ourselves, or use e.g. Debian's. Armbian does not (yet) support it.

MichaIng, 2023-02-11:
FYI: #6151

MichaIng, 2023-02-18:
The new set of images has this module available: https://dietpi.com/downloads/images/testing/
I'm still testing those before moving to stable downloads, but so far all works pretty good.

closed 2023-02-19 by MichaIng