MichaIng/DietPi #6013: Compose / Docker encounters error before installation completes - Rock 5B DietPi

tggjifdhn, 2022-12-27:
#### Details:
- Date           | Tue Dec 27 23:15:16 GMT 2022
- Bug report     | 96ea17fe-387e-412a-b3b1-28ed595dab4d
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | ROCK 5B (aarch64) (ID=78)
- Kernel version | `Linux DietPi 5.10.110-rockchip-rk3588 #22.11.2 SMP Sun Dec 11 21:44:12 CET 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y --allow-change-held-packages install docker-ce`
- Exit code      | 100
- Software title | DietPi-Software

#### Expected behaviour:
Docker and compose installing without errors.

#### Actual behaviour:
DietPi encounters an error before installation completes.

#### Extra details:
Perusing Armbian's Rock 5B page they have instructions for docker. I wonder if this is required here? : 
https://www.armbian.com/rock-5b/ 
> In case you need to run Docker:
update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy

#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
docker-ce is already the newest version (5:20.10.22~3-0~debian-bullseye).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up docker-ce (5:20.10.22~3-0~debian-bullseye) ...
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2022-12-27 23:15:16 GMT; 41ms ago
TriggeredBy: ● docker.socket

Joulinar, 2022-12-27:
Can you share following

```
journalctl -u docker.service
```

tggjifdhn, 2022-12-28:
Sure! Here's the output:
https://pastebin.com/f3mUHeaU

Joulinar, 2022-12-28:
Did you already set following?

```
update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy
```

If not, do so and try to install again.

MichaIng, 2022-12-28:
Strange is that this Linux version should support nftables. However, even that it's an Armbian build, it's still based on Radxa's vendor kernel sources where you cannot rely on the kernel version.

`dietpi-software` should however adjust iptables automatically. To check:
```sh
ls -l /etc/alternatives/iptables
```

tggjifdhn, 2022-12-28:
Manually setting 
`update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy`
allowed the installation to conclude successfully. 

Thank you!

MichaIng, 2022-12-28:
Great that it helped.

Did you install Docker and/or Compose via `dietpi-software` or manually? If via `dietpi-software` (since it should have tested and in case applied legacy alternative automatically), could you show the output of:
```sh
iptables-nft -L
echo $?
```

tggjifdhn, 2022-12-28:
Let me know if you need anything else:

`dietpi@DietPi:/usr/local/bin$ sudo iptables-nft -L`
`# Warning: iptables-legacy tables present, use iptables-legacy to see them`
`iptables v1.8.7 (nf_tables): Could not fetch rule set generation id: Invalid argument`

MichaIng, 2022-12-28:
Ah, I think I understand the issue now: The `docker-ce` package install itself fails already, so the config step to apply legacy `iptables` is not even reached. Hum, now I'm confused how to solve this, since `iptables` itself is installed as a dependency of the Docker package, so we cannot configure it prior to installing the Docker package 🤔. Probably it works to mask the service before installing the package, so it is not tried to be started, hopefully without having the install failing because of it.

MichaIng, 2022-12-28:
This should fix it, but as I'm on vacation I cannot test it currently: https://github.com/MichaIng/DietPi/pull/6014/files
If you find time to uninstall Docker + iptables and install it freshly with these changes applied, that would be great. Otherwise I can do in second January week.

tggjifdhn, 2022-12-28:
I'm not entirely sure how to go about and test (inexperienced). I tried mounting the DietPi_ROCK5B-ARMv8-Bullseye image and editing the lines in dietpi-software manually however the file is read only and when attempting to change I get "Error setting permissions: Read-only file system"

Joulinar, 2022-12-28:
just install a fresh image and complete basic install as usual. Once done do following to switch to testing branch for this issue

```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=docker' /boot/dietpi.txt
```

This should update the system. Do a reboot afterwards and try to install Docker again.

tggjifdhn, 2022-12-28:
Installed without a hitch on a fresh installation with the above instructions!
`G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=docker' /boot/dietpi.txt
dietpi-update 1`

Joulinar, 2022-12-28:
means Docker is installed now?

MichaIng, 2022-12-29:
Seems so, I merged the PR 🙂.

tggjifdhn, 2022-12-29:
> means Docker is installed now?

Yes

MichaIng, 2022-12-30:
Great, I'll mark this as solved.

closed 2022-12-30 by MichaIng

isarrider, 2023-01-02:
Ah - very well...
I saw the same for my Rock 5 here (I link it for reference):
https://github.com/MichaIng/DietPi/discussions/5247
Awesome that it is fixed...