MichaIng/DietPi #6017: Make your own distribution script is failing on OrangePi 5 (rockchip 3588)

rtmpgh, 2022-12-30:
#### Details:
- Date           | Thu Dec 30 00:00:17 EST 2021
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=)
- Kernel version | `Linux orangepi5-30 5.10.110-rockchip-rk3588 #1.0.6 SMP Fri Dec 16 09:18:31 CST 2022 aarch64 aarch64 aarch64 GNU/Linux`
- Distro         |  (ID=)
- Command        | `7zr a -m0=lzma2 -mx=9 -spf -x!/var/log/journal/* .7z CMD_OUT.txt G_EXEC_ERROR_REPORT /var/log /boot/config.txt /boot/cmdline.txt /boot/boot.ini /boot/uenv.txt /boot/uEnv.txt /boot/armbianEnv.txt /boot/boot.cmd /boot/dietpi.txt /boot/dietpi /var/lib/dietpi /var/tmp/dietpi /run/dietpi /etc/bash.bashrc /etc/bashrc.d /root/.bashrc /etc/profile /etc/profile.d /root/.profile /etc/rc.local /etc/X11/xorg.conf /etc/X11/xorg.conf.d /etc/asound.conf /etc/network/interfaces /etc/network/interfaces.d /etc/wpa_supplicant/wpa_supplicant.conf /etc/fstab /etc/sysctl.conf /etc/sysctl.d /etc/init.d /etc/systemd/system /lib/systemd/system /usr/local/lib/systemd/system /etc/apt/sources.list /etc/apt/sources.list.d /etc/apt/preferences /etc/apt/preferences.d /etc/apt/apt.conf.d`
- Exit code      | 127
- Software title | DietPi-BugReport
#### Steps to reproduce:

1. ... bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-installer')"
2. ...
#### Expected behaviour:

- ... i would hope this script would actually work 
#### Actual behaviour:

- ...  this script is failing because it keeps telling me 
-  Err:5 https://repo.huaweicloud.com/docker-ce/linux/ubuntu focal Release                                              │
                                  │   Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid          │
                                  │ certificate.  Could not handshake: Error in the certificate verification. [IP: 199.91.74.174 443]                    │
                                  │ Ign:6 https://deb.debian.org/debian bullseye-backports InRelease                                                     │
                                  │ Err:7 https://deb.debian.org/debian bullseye Release                                                                 │
                                  │   Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid          │
                                  │ certificate.  Could not handshake: Error in the certificate verification. [IP: 146.75.38.132 443]                    │
                                  │ Err:8 https://deb.debian.org/debian bullseye-updates Release                                                         │
                                  │   Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid          │
                                  │ certificate.  Could not handshake: Error in the certificate verification. [IP: 146.75.38.132 443]                    │
                                  │ Err:9 https://deb.debian.org/debian-security bullseye-security Release                                               │
                                  │   Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid          │
                                  │ certificate.  Could not handshake: Error in the certificate verification. [IP: 146.75.38.132 443]                    │
                                  │ Err:10 https://deb.debian.org/debian bullseye-backports Release                                                      │
                                  │   Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid          │
                                  │ certificate.  Could not handshake: Error in the certificate verification. [IP: 146.75.38.132 443]    
#### Extra details:

- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 858: 7zr: command not found
```

Joulinar, 2022-12-30:
> Err:5 https://repo.huaweicloud.com/docker-ce/linux/ubuntu focal Release │

What base image did you used? Was at an Ubuntu one? We don't support Ubuntu. You need to use a Debian image.

rtmpgh, 2022-12-30:
Ok I will go create a new image using a Debian base

rtmpgh, 2022-12-31:
after downloading and flashing the Debian version from https://drive.google.com/drive/folders/1F2uc8v_EQnvsNrevDihwoymOJlFgM-dZ

this is what the script produces 

#### Details:
- Date           | Sat Dec 31 14:40:19 EST 2022
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=78)
- Kernel version | `Linux orangepi5-31 5.10.110-rockchip-rk3588 #1.0.2 SMP Fri Dec 9 10:21:25 CST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install linux-image-legacy-rockchip-rk3588 linux-dtb-legacy-rockchip-rk3588 linux-u-boot-rock-5b-legacy u-boot-tools armbian-firmware`
- Exit code      | 100
- Software title | DietPi-PREP
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
u-boot-tools is already the newest version (2021.01+dfsg-5).
linux-dtb-legacy-rockchip-rk3588 is already the newest version (22.11.2).
linux-image-legacy-rockchip-rk3588 is already the newest version (22.11.2).
The following packages were automatically installed and are no longer required:
  alsa-utils apt-file apt-transport-https apt-utils aptitude aptitude-common
  autoconf automake autotools-dev avahi-autoipd bash-completion bc
  bind9-dnsutils bind9-host bind9-libs binutils binutils-aarch64-linux-gnu
  binutils-common bison bluetooth bluez bluez-tools bridge-utils bsdextrautils
  btrfs-progs build-essential bzip2 ca-certificates can-utils chrony
  command-not-found console-setup console-setup-linux cpp cpp-10 cpufrequtils
  cracklib-runtime crda cron curl dbus dbus-user-session dctrl-tools
  debconf-utils device-tree-compiler dialog dirmngr distro-info-data dkms
  dnsutils dosfstools dpkg-dev ethtool evtest expect f2fs-tools f3
  fake-hwclock fbset fdisk figlet file flex fontconfig fontconfig-config
  fonts-cantarell fonts-dejavu-core fping fuse g++ g++-10 gcc gcc-10
  gir1.2-glib-2.0 gir1.2-packagekitglib-1.0 git git-man gnupg gnupg-l10n
  gnupg-utils gnupg2 gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm
  groff-base haveged hdparm hostapd hping3 html2text htop i2c-tools ifenslave
  ifupdown iotop iozone3 iperf3 iproute2 iptables iputils-arping iputils-ping
  isc-dhcp-client iso-codes iw jq kbd keyboard-configuration keyutils less
  libapt-pkg-perl libasan6 libasound2 libasound2-data libassuan0 libatomic1
  libatopology2 libbinutils libbluetooth3 libboost-iostreams1.74.0 libbpf0
  libbrotli1 libbsd0 libc-dev-bin libc-l10n libc6-dev libcaca0 libcairo2
  libcap2-bin libcbor0 libcc1-0 libcpufreq0 libcrack2 libcrypt-dev
  libctf-nobfd0 libctf0 libcurl3-gnutls libcurl4 libcwidget4 libdaemon0
  libdatrie1 libdbus-1-3 libdigest-sha-perl libdns-export1110 libdpkg-perl
  libdrm-common libdrm2 libdw1 libedit2 libelf1 liberror-perl libestr0
  libevent-2.1-7 libexpat1 libexporter-tiny-perl libfastjson4 libfdisk1
  libfdt1 libfftw3-single3 libfido2-1 libfontconfig1 libfreetype6 libfribidi0
  libfstrm0 libfuse2 libgcc-10-dev libgdbm-compat4 libgdbm6
  libgirepository-1.0-1 libglib2.0-0 libgomp1 libgpm2 libgraphite2-3
  libharfbuzz0b libhavege2 libi2c0 libicu67 libip6tc2 libiperf0
  libisc-export1105 libisl23 libitm1 libiw30 libjansson4 libjq1 libksba8
  libldap-2.4-2 liblist-moreutils-perl liblist-moreutils-xs-perl liblmdb0
  liblocale-gettext-perl liblognorm5 liblsan0 liblzo2-2 libmagic-mgc libmagic1
  libmaxminddb0 libmd0 libmm-glib0 libmnl0 libmpc3 libmpdec3 libmpfr6
  libncurses6 libncursesw6 libndp0 libnetfilter-conntrack3 libnetplan0
[... 32 lines cut ...]
  wireless-tools wpasupplicant xkb-data xxd xz-utils zsh zsh-common
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  armbian-firmware linux-u-boot-rock-5b-legacy
0 upgraded, 2 newly installed, 0 to remove and 1 not upgraded.
2 not fully installed or removed.
Need to get 0 B/10.4 MB of archives.
After this operation, 2048 B of additional disk space will be used.
(Reading database ... 38634 files and directories currently installed.)
Preparing to unpack .../armbian-firmware_22.11.2_all.deb ...
Unpacking armbian-firmware (22.11.2) ...
dpkg: error processing archive /tmp/apt/archives/armbian-firmware_22.11.2_all.deb (--unpack):
 trying to overwrite '/lib/firmware/BCM4345C0.hcd', which is also in package orangepi-firmware 1.0.2
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../linux-u-boot-rock-5b-legacy_22.11.2_arm64.deb ...
Unpacking linux-u-boot-rock-5b-legacy (22.11.2) ...
dpkg: error processing archive /tmp/apt/archives/linux-u-boot-rock-5b-legacy_22.11.2_arm64.deb (--unpack):
 trying to overwrite '/usr/lib/u-boot/LICENSE', which is also in package linux-u-boot-orangepi5-legacy 1.0.2
Errors were encountered while processing:
 /tmp/apt/archives/armbian-firmware_22.11.2_all.deb
 /tmp/apt/archives/linux-u-boot-rock-5b-legacy_22.11.2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.

MichaIng, 2023-01-01:
Many thanks for your report.

You selected "ROCK 5B" as target SBC right? This isn't gonna work. Please try with "Generic Device".

danielpinto8zz6, 2023-01-05:
I was able to use the script with "Generic Device", it worked

Joulinar, 2023-01-05:
Ok I will close the issue for now. Feel free to reopen if needed.

closed 2023-01-05 by Joulinar

danielpinto8zz6, 2023-01-05:
Just a question, is there any documentation/guideline to add support to a new device oficially?

danielpinto8zz6, 2023-01-07:
Hey @MichaIng , and how can I test the implementation?

MichaIng, 2023-01-08:
You could run `dietpi-build` with respective arguments and environment variables from any DietPi, Debian or Ubuntu Jammy system, like done in the `dietpi-build.yml`.

danielpinto8zz6, 2023-01-08:
Thank you, the Opi5 needs some specific patches, where can I add them? Or it extracts some base image?

MichaIng, 2023-01-10:
The preferred method is to build images from scratch via `debootstrap` (done by default in `dietpi-build`) and apply patches in `dietpi-installer`.

I btw recommend to use Armbian's mainline kernel and U-Boot packages, like we do in most other cases. There shouldn't be many patches required then, like for ROCK 5B and most others. They currently use the vendor kernel as well.

danielpinto8zz6, 2023-01-12:
Hey @MichaIng , I'm currently working on Opi5 dietpi image, based on armbian,
I have a doubt, https://github.com/danielpinto8zz6/DietPi/commit/4604308f57add7e7612a757c9335158b13b274be#diff-6497f4d3c710aad72255bf4d0850cb45a098831470f935a7225d4d2b70ea9f54R117 how do we get the root_size and boot_size correctly?
Thank you

MichaIng, 2023-01-12:
The values you use should be a good starting point:
- 128 MiB for boot is more than sufficient for the Armbian kernel/dtb packages, given that max three of them (`legacy`, `current` and `edge`) can be installed concurrently. And good to align with our other images.
- The root partition should be as small as possible, as it affects the final image size as well (the file creation+removal done by APT installs and other `dietpi-installer` tasks create holes in the data layout), but as large as required to not fail with insufficient disk space at some point. The values we use are based on some empiric trail&error. Since it is very similar to ROCK 5B, using the same size should be fine.

danielpinto8zz6, 2023-01-12:
@MichaIng Oh great, thank you
I can generate an image from the build script?

danielpinto8zz6, 2023-01-12:
I'm having this message when building the image with dietpi-build, is this normal?
[image]

MichaIng, 2023-01-12:
Yes that is expected whenever an image/disk is resized and the step "Moving GPT backup partition table" is to fix this broken backup header/partition table.

danielpinto8zz6, 2023-01-12:
My only problem now is to apply the patches to the kernel, https://github.com/armbian/build/tree/master/patch/kernel/rockchip-rk3588-legacy
Opi5 requires some, rock 5b apparently doesn't

MichaIng, 2023-01-12:
The current Armbian `rockchip-rk3588` kernel package does not contain these patches yet?

danielpinto8zz6, 2023-01-12:
nope that I know

MichaIng, 2023-01-13:
A right, the latest kernel package is older than 2 weeks (when the patches have been merged): https://fi.mirror.armbian.de/apt/pool/main/l/linux-5.10.110-rockchip-rk3588/

I wonder if there is an unpublished version of the kernel package installed on the provided images, or whether the files have been manually placed for the provided Armbian images: https://mirrors.dotsrc.org/armbian-dl/orangepi5/archive/

I'd say with image generation we should wait until the kernel packages have been updated with the patches, everything else is unclean hassle.

danielpinto8zz6, 2023-01-13:
Ok, let's wait then, 
Thank you for you help @MichaIng