MichaIng/DietPi #6020: uninstall mopidy error

joybbs, 2023-01-01:
#### Details:
- Date           | Sun Jan  1 10:20:34 CST 2023
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi 5.10.0-20-amd64 #1 SMP Debian 5.10.158-2 (2022-12-13) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `pip3 uninstall -y Mopidy-MusicBox-Webclient`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 9, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main_parser.py", line 9, in <module>
    from pip._internal.build_env import get_runnable_pip
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/build_env.py", line 20, in <module>
    from pip._internal.cli.spinners import open_spinner
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/spinners.py", line 9, in <module>
    from pip._internal.utils.logging import get_indentation
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 29, in <module>
    from pip._internal.utils.misc import ensure_dir
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/misc.py", line 42, in <module>
    from pip._internal.locations import get_major_minor_version
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/locations/__init__.py", line 67, in <module>
    from . import _distutils
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/locations/_distutils.py", line 20, in <module>
    from distutils.cmd import Command as DistutilsCommand
ModuleNotFoundError: No module named 'distutils.cmd'

joybbs, 2023-01-01:
I have solved

MichaIng, 2023-01-01:
I guess some Python 3 package/module has been manually uninstalled before Mopidy?

Joulinar, 2023-01-01:
Should we close the issue than?

MichaIng, 2023-01-03:
Marking it as closed. Feel free to reopen if required.

closed 2023-01-03 by MichaIng