MichaIng/DietPi #6023: Kernel Panic on Installing Bulleye on Nanopi Neo Plus2

bench2012, 2023-01-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- <Can't do that because it never finish the install. But the image is ARMv8 Bullseye"
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- <DietPi_NanoPiNEOPlus2-ARMv8-Bullseye>
- Kernel version | `uname -a`
- <From boot log (serial console) 247.3-7+deb11u1>
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- <From uboot log FriendlyARM NanoPi NEO Plus2>
- Power supply used | (EG: 5V 1A RAVpower)
- <5V 2A>
- SD card used | (EG: SanDisk ultra)
- <Samsung EVO 32G>

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- <N/A>
- Was the software title installed freshly or updated/migrated?
- <Internal EMMC have FriendlyARM Ubuntu 20.04, which booted ok and working fine>
- Can this issue be replicated on a fresh installation of DietPi?
- <Oh yes, failed every time>

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Create an boot SD card from img file using Ether 
2. Boot and start installing using serial port
3. It will fail at some point (quite random)

#### Expected behaviour

- Finish the install and boot as least from the SD card.

#### Actual behaviour

- Kernel panic 

#### Extra details

- Pastbin link of the console log:
https://pastebin.com/H6fQTmBV

MichaIng, 2023-01-02:
Many thanks for your report.

Is it possible to continue the first run setup after the crash (power cycle)?

Probably you can monitor some general things like CPU temperature, RAM usage and auch on dedicated SSH sessions.

Also what is thr max CPU frequency?
```bash
cpu
```
I read one comment on Armbian forum from a user stating that his particular board runs stable at 1.1 GHz only. This can be limited in `dietpi-config` performance options.

Voltage (insufficient PSU) or bad SD card could be also the reason, so if you have alternatives to test with, that would be a good idea as well.

bench2012, 2023-01-02:
Hi,

Thanks for your reply. I did power cycle it and it did booted up. But once I reboot again it crash with a kernel panic. I power cycle and it will come backup. I try setting the max. cpu freq. to 1.08GHz, but does the same, as soon as I reboot, it came backup with a kernel panic when it restart.

I try with another PSU, 5.1v, 2A and still the same result. CPU Temp. is a cool 35C, so should not be a thermal issue.

As I say, the image from FriendlyARM (ubuntu 20.04 LTS) works (on SD and emmc), but with an older kernel (4.14).

I have try different SD (I org.use an old unbranded 8G, but using a 32G Samsung EVO now) and it is the same.

I don't want to power cycle it all the time when reboot, it doesn't do the file system much good.

This is really wired and any pointers are greatly appreciated. 

Many thx.

MichaIng, 2023-01-03:
Ah, do I understand it correctly that it crashes only when you `reboot` but boots and runs stable when you power cycle?

I remember this power cycling being/was an issue with quite some boards and Armbian kernel/bootloader, although in all other cases boot halts in bootloader stage, so kernel is not loaded yet => no kernel panic.

Fixing this is outside of our abilities, I'm afraid. We do no kernel development but focus on userland tools. We might try with a different kernel or bootloader build. Basically I want to test the official Debian kernel with some SBCs, especially those where Debian also ships U-Boot for (which are not that many), but not sure when I find time for this.

For now, instead of `reboot`, a `poweroff` with power cycle should at least lead to a clean shutdown without potential damage to the filesystem.

Another thing you can do is to test with the Armbian Bullseye image: https://www.armbian.com/nanopi-neo-plus2/
Upgrade the kernel to latest version
```sh
apt update
apt full-upgrade
```
If it has the same issue, you can report it at the Armbian bug tracker: https://www.armbian.com/bugs/
The board has a maintainer, so in theory someone should reply, but there is no guarantee. Also, do not mention DietPi, since Armbian does not like us to forward bug reports, even if those are clearly kernel/bootloader bugs, unrelated to DietPi specifics.

Ah @StephanStS if you find time, can you replicate the issue on your NEO Plus2 with our latest image?

StephanStS, 2023-01-03:
I did a test with my NanoPi NEO Plus 2: I did the minimal basic installation with no changes in the first run setup using the actual image downloaded from dietpi.com.
I could reboot it via `reboot` a couple of times and the NEO Plus 2 came up every time without any issues.

MichaIng, 2023-01-04:
Thanks for testing. So it's either a different board revision or bad luck with hardware or SD cards. Or the OS on internal eMMC plays a role? Probably the bootloader from eMMC is used (accidentally) on reboot, causing the issue? For verifying this, the first MiB of the eMMC would need to be erased (breaking Ubuntu until being reflashed), or DietPi flashed to eMMC.

bench2012, 2023-01-04:
Hi,

I basically have the same problem with Armbian as well and have reported it in the bug report, but have yet to hear from them, I suppose it is still new year for everyone. I haven't try the min. version and is it the current version you #StephanStS tested? Will certainly give it a go. Not sure about wiping the emmc, as I am not too comfortable with dd a running SD onto the emmc, unless there is another way to do it. 

One thing I did observe with Friendlyarm's own ubuntu is once I install onto the emmc, I can no longer access the SD card. I do understand it will try to boot off the SD card if first, but even when I let it boot to emmc and reinsert the SD card, the system did not detect it.

I do prefet dietpi than Armbian, as it is more compact with less "bloatware" and also more responsive.

I have dietpi running on a Nanopi Neo 2 with no problem, but it have no wireless connection and I don't want to use USB (I have other plan for it). It would be nice if I can run dietpi on the Plus2 as well.

Thanks for you help anyway. Will see if I get any more luck with the min. version.

Bench2012

StephanStS, 2023-01-04:
@bench2012: Maybe this blog post (https://dietpi.com/blog/?p=1236) help a bit for using the dd command.

MichaIng, 2023-02-12:
Any news on this?

Here is a similar bug report on Armbian forum, bug looks like it was a broken eMMC only: https://forum.armbian.com/topic/20522-nanopi-neo-plus2-weird-issues-and-kernel-oops-with-current-kernel/

bench2012, 2023-02-13:
No, haven't really had time to  mass around with it lately. I have already came across the armbian topic and looks like quite a few people have the same issue. May be I wait a bit and see if there is any update. As I say, it works fine with Friendlyarm's legacy ubuntu kernel, so not sure it is a HW issue, just not more uptodate kernel. May be the old kernel is more  tolerating to broken storage, who knows?

Any way, will keep look out for a solution.

Thank for all the help anyway.

MichaIng, 2023-03-18:
I'll mark this as closed. Feel free to reopen when this becomes relevant to you again and the issue persists.

closed 2023-03-18 by MichaIng

bench2012, 2023-07-14:
Hi,

Time flies and we are back again. I got a working Detipi setup on a Nanopi Neo 2, that is until it decide to die on me, so I am left with the Nanopi Neo Plus 2 again. I actually got it working on old Armbian Buster, until I update/upgrade to the latest kernel and the Kernel panic starts again. 

I try Dietpi Buster and it boot up ok and then it update to latest kernel, it kernel panic at boot.

The really funny thing is, in desperation, I put the Nanopi Neo 2 SD card on the Plus 2 and it boot up fine! Of course it did not detect all the Plus2 devices like Wifi and emmc, but everything else was fine (I have a DAC board and the LCD shield on it). So that is something screwy about the latest kernel build for the Plus 2. 

Any help would be appreciated.

Below is the kernel panic log:
[  OK  ] Started Entropy Daemon based on the HAVEGE algorithm.
         Starting Netwo[    5.986920] Unable to handle kernel paging request at virtual address ffff80003eff2364
rk Time Synchron[    5.996369] Mem abort info:
ization...
[    6.000481]   ESR = 0x0000000096000006
[    6.005599]   EC = 0x25: DABT (current EL), IL = 32 bits
[    6.010913]   SET = 0, FnV = 0
[    6.013967]   EA = 0, S1PTW = 0
[    6.017114]   FSC = 0x06: level 2 translation fault
[    6.021993] Data abort info:
[    6.024873]   ISV = 0, ISS = 0x00000006
[    6.028707]   CM = 0, WnR = 0
[    6.031674] swapper pgtable: 4k pages, 48-bit VAs, pgdp=000000004134b000
[    6.038369] [ffff80003eff2364] pgd=100000007ffff003, p4d=100000007ffff003, pud=100000007fffe003, pmd=0000000000000000
[    6.048977] Internal error: Oops: 96000006 [#1] SMP
[    6.053850] Modules linked in: cpufreq_dt g_serial libcomposite realtek dwmac_sun8i mdio_mux
[    6.062292] CPU: 2 PID: 360 Comm: systemd-udevd Not tainted 5.15.93-sunxi64 #23.02.2
[    6.070024] Hardware name: FriendlyARM NanoPi NEO Plus2 (DT)
[    6.075675] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    6.082626] pc : cmp_map_id+0x0/0x78
[    6.086207] lr : bsearch+0x50/0xb8
[    6.089607] sp : ffff8000098bb930
[    6.092914] x29: ffff8000098bb930 x28: 00000000fffffff6 x27: 2f2f2f2f2f2f2f2f
[    6.100042] x26: 0000000000000001 x25: ffff8000098bb9ac x24: ffff800008118f60
[    6.107170] x23: ffff8000093c8cb4 x22: 000000000000000c x21: 00000000047ae1e4
[    6.114297] x20: ffff80003eff2364 x19: 0000000008f5c3c7 x18: 0000000000000000
[    6.121424] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[    6.128551] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[    6.135678] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
[    6.142805] x8 : ffff00000353abb8 x7 : 0000000000000000 x6 : 0000000000000001
[    6.149932] x5 : ffff000001c1ca68 x4 : ffff800008118f60 x3 : 000000000000000c
[    6.157060] x2 : 0000000008f5c3c8 x1 : ffff80003eff2364 x0 : ffff8000098bb9ac
[    6.164188] Call trace:
[    6.166630]  cmp_map_id+0x0/0x78
[    6.169855]  map_id_range_down+0xd0/0x100
[    6.173861]  make_kuid+0x14/0x20
[    6.177086]  generic_permission+0x104/0x2f0
[    6.181264]  kernfs_iop_permission+0x7c/0xf0
[    6.185529]  inode_permission+0x180/0x260
[    6.189532]  link_path_walk.part.41+0x2c8/0x3c0
[    6.194056]  path_lookupat.isra.43+0x3c/0x158
[    6.198406]  path_openat+0xa14/0x1128
[    6.202063]  do_filp_open+0x88/0x100
[    6.205633]  do_sys_openat2+0x21c/0x2e8
[    6.209468]  do_sys_open+0x58/0xa0
[    6.212866]  __arm64_sys_openat+0x24/0x30
[    6.216871]  invoke_syscall+0x44/0x108
[    6.220617]  el0_svc_common.constprop.3+0x84/0xf8
[    6.225314]  do_el0_svc+0x24/0x98
[    6.228624]  el0_svc+0x20/0x50
[    6.231676]  el0t_64_sync_handler+0x90/0xb8
[    6.235853]  el0t_64_sync+0x180/0x184
[    6.239515] Code: 17ffffd9 12800000 17ffffe4 00000000 (29400823)
[    6.245599] ---[ end trace df9196fad785315d ]---
[    6.558483] hrtimer: interrupt took 7000 ns
[   65.982477] rcu: INFO: rcu_sched self-detected stall on CPU
[   65.988044] rcu:     0-...!: (14998 ticks this GP) idle=91f/1/0x4000000000000002 softirq=2234/2242 fqs=0
[   65.997341]  (t=15000 jiffies g=333 q=845)
[   66.001431] rcu: rcu_sched kthread starved for 15000 jiffies! g333 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=3
[... 549 lines cut ...]
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
EHCI failed to shut down host controller.
   Loading Ramdisk to 496d9000, end 49fff121 ... OK
   Loading Device Tree to 0000000049668000, end 00000000496d8fff ... OK

Starting kernel ...

[    1.629793] debugfs: Directory '1c22c00.codec' with parent 'H3 Audio Codec' already present!
Loading, please wait...
Starting version 241
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... done.
Begin: Will now check root file system ... fsck from util-linux 2.33.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk0p1] fsck.ext4 -a -C0 /dev/mmcblk0p1
/dev/mmcblk0p1: clean, 19833/482144 files, 476964/1992704 blocks
done.
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... done.

Joulinar, 2023-07-16:
I'm not sure we're in a position to help with this kernel panic. Because we don't do own kernel development for this device. Also, you are on Debian Buster, which became oldstable with the release of Debian Bookworm last month. You could try asking for support on the Armbian forum, since you have the same behaviour on regular Armbian. However, you should avoid any reference to DietPi, as the developers of Armbian don't necessarily like us ;) And I'm not sure if they will even take a look because of the old Debian version..

MichaIng, 2023-07-16:
> I put the Nanopi Neo 2 SD card on the Plus 2 and it boot up fine! Of course it did not detect all the Plus2 devices like Wifi and emmc

Actually both images share the same SoC and kernel, with device trees for both of them, so you should be able to use all hardware features with the NEO2 image, at least all of them which work with the NEO Plus2 image.

But they have a different bootloader, so I guess booting from eMMC does not work with the NEO2 image, but it should be detected after booting from SD card.

That they share exactly the same kernel build also makes me wonder how it can have the kernel panic with one image but not with the other. In case, aside of the bootloader, both images are completely identical. Did you boot the NEO Plus2 image via SD card of eMMC?

bench2012, 2023-07-17:
> Actually both images share the same SoC and kernel, with device trees for both of them, so you should be able to use all hardware features with the NEO2 image, at least all of them which work with [...]

Hi,

I actually boot successfully from SD card and install onto EMMC using Armbian Buster. DietPi Buster boot as well, but it die when it done all the update and reboot with the new kernel. Apart from being an "old" distribution with out dated kernel, it seems to be fine, that is until I update to the latest kernel(v5.10<) and firmware. 

As I have already done some stuff DietPi on my Neo2 (not the Plus2) with the Bullseye and my Neo2 just die, I am stuck with the Plus2. I check the boot loader log (uboot) and they are different between (version) the Neo2 and the Plus2, the uboot actually say it is for an Plus2 or a Neo2. I am not sure if they compile a new version of the uboot with Plus2 profile/DTB or not. As the Neo2 is almost identical with the Plus2, except for the Wifi and EMMC, one would thought  they wouldn't, but then this happen. 

Anyway, will try this again with Armbian. I can see from their forum they are having some trouble with Bookworm on the H5 kernel ie. Neo2, Plus2 etc. but I am not going for the latest and greatest, I just need something recent-ably new and work.   

As I say, it boot happily using the Neo2 SD with DietPi on it, just no Wifi and EMMC, so I just wonder if anybody else have the same issue (I can't find anything on yours or Armbian's forum). 

Bench