MichaIng/DietPi #6028: Issue with reloading openssh service during installation

danielshub, 2023-01-05:
This is going to be a crappy bug report, but I am hoping it is an easy enough to track down. In a clean install I opted to use openssh instead of dropbear. I am assuming this is what causes the issue. The installation fails when trying to run `systemctl reload ssh`. The error message is "ssh.service is not active". I dropped into a sub shell and ran `systemctl start ssh` and then retry the original command and everything works.

It may be worth adding a `systemctl is-active` before trying to reload services

MichaIng, 2023-01-05:
Thanks for reporting. OpenSSH should be actually started during DEB package install. Which SBC and Debian version are you using, basically could you fill the bug report template?

MichaIng, 2023-02-12:
Closing this issue due to outstanding reply. Feel free to reopen if the problem persists.

closed 2023-02-12 by MichaIng