MichaIng/DietPi #6029: Prowlarr

LvsDark, 2023-01-05:
DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Prowlarr: Indexer Manager & Proxy for PVR

curl: (22) The requested URL returned error: 403
[FAILED] DietPi-Software | Automatic latest Prowlarr download URL detection failed.
[FAILED] DietPi-Software | "https://github.com/Prowlarr/Prowlarr/releases/download/v0.4.10.2111/Prowlarr.develop.0.4.10.2111.linux-core-arm64.tar.gz" will be used as fallback, but a newer version might be available.
[FAILED] DietPi-Software | Please report this at: https://github.com/MichaIng/DietPi/issues

Joulinar, 2023-01-05:
Maybe a temp. issue. Can you check if following is working fine

```
curl -sSfL 'https://api.github.com/repos/Prowlarr/Prowlarr/releases' | mawk -F\" "/\"browser_download_url\": .*linux-core-arm64\.tar\.gz\"/{print \$4}" | head -1
```

LvsDark, 2023-01-06:
curl: (22) The requested URL returned error: 403

LvsDark, 2023-01-06:
Although, i got it to install after rebooting and trying again.  but this does still pop up : curl -sSfL 'https://api.github.com/repos/Prowlarr/Prowlarr/releases' | mawk -F\" "/\"browser_download_url\": .*linux-core-arm64\.tar\.gz\"/{print \$4}" | head -1

LvsDark, 2023-01-06:
randomly tried it again and it worked: 
dietpi@rpi:~$ curl -sSfL 'https://api.github.com/repos/Prowlarr/Prowlarr/releases' | mawk -F\" "/\"browser_download_url\": .*linux-core-arm64\.tar\.gz\"/{print \$4}" | head -1
https://github.com/Prowlarr/Prowlarr/releases/download/v1.0.1.2220/Prowlarr.master.1.0.1.2220.linux-core-arm64.tar.gz

Mustve been a temp issue. Thanks@!

closed 2023-01-06 by LvsDark

LvsDark, 2023-01-16:
I am not and at the time was not using ant public or large Lan/Internet. Just my personal private network with regular traffic.