MichaIng/DietPi #6030: [dietpi-vpn] ProtonVPN not able to select FREE Server

black00019, 2023-01-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
                               G_DIETPI_VERSION_SUB=12
                               G_DIETPI_VERSION_RC=1
                               G_GITBRANCH='master'
                               G_GITOWNER='MichaIng'
                               G_LIVE_PATCH_STATUS[0]='not applied'
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
- SBC model | RPi 3 Model B+ (aarch64)
- Power supply used | 5V 2A
- SD card used | Kingstone 128GB

#### Additional Information (if applicable)
- Software title | dietpi-vpn - ProtonVPN
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. I update the server list of ProtonVPN from dietpi-vpn
2. Not show FREE server, but i see only paid server

#### Expected behaviour
- I want select the account typology (FREE, PAID) and select server associated

#### Actual behaviour
- See paid account server

#### Extra details
- https://dietpi.com/forum/t/dietpi-vpn-in-7-1/5261/7

black00019, 2023-01-26:
Hi
the change when it is passed on DietPi?

MichaIng, 2023-01-26:
We have a looong ToDo and wish list, so I cannot give you an ETA. If anyone else does a start, I'll however give a helping hand quickly.

MichaIng, 2026-10-26:
#8399 should fix it. At least I do see free servers in the newly used server list. And the name is shown in the selection menu as well now, which is prefixed with "FREE" in case of a free server.

closed 2026-10-26 by MichaIng