MichaIng/DietPi #6031: Pine H64 DietPi Image is Currently Missing the Kernel

DremOSDeveloperTeam, 2023-01-05:
At the time of writing, the DietPi image for the Pine H64 is missing a kernel and, by extension, cannot boot. This goes for both the the [Bullseye](https://dietpi.com/downloads/images/DietPi_PINEH64-ARMv8-Bullseye.7z) and [Bookworm](https://dietpi.com/downloads/images/DietPi_PINEH64-ARMv8-Bookworm.7z) images.

Here are some [screenshots of the issue](https://imgur.com/a/wbkyz7C). Apologies for using Imgur instead of attaching them in this issue directly, but for some reason I can't upload images to Github at the moment.

MichaIng, 2023-01-06:
What the... Thanks for reporting. I'll fix it in some hours.

MichaIng, 2023-01-07:
Solved with: https://github.com/MichaIng/DietPi/commit/a861eff
Images are rebuilt and will be available in ~30 minutes here: https://dietpi.com/downloads/images/testing/

DremOSDeveloperTeam, 2023-01-07:
Excellent, thanks. Closing this issue as the problem seems to be solved!

closed 2023-01-07 by DremOSDeveloperTeam