MichaIng/DietPi #6032: Dietpi Dashboard

LvsDark, 2023-01-06:
#### Details:
- Date           | Thu Jan  5 16:58:05 PST 2023
- Bug report     | 3e0b1e65-621c-41c6-8401-67252e9e0c89
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid N2 (aarch64) (ID=15)
- Kernel version | `Linux rpi 4.9.312-arm64 #1 SMP PREEMPT Wed Nov 23 00:32:45 CET 2022 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `chmod +x /opt/dietpi-dashboard/dietpi-dashboard`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1.  Attempting to install or reinstall package 200, DietPi Dashboard
#### Expected behaviour:

- should install dietpi dashbaord stable version.
#### Actual behaviour:

- ...
#### Extra details:
Prior to this issue, during a reinstallation of the dietpi dashboard, i accidently tapped my key bored and cancelled the mid process, now i get this error everytime i try to "install it, or reinstall it.
- ...
#### Additional logs:
```
chmod: cannot access '/opt/dietpi-dashboard/dietpi-dashboard': No such file or directory
```

Joulinar, 2023-01-06:
pls can you share the whole output of 

```
dietpi-software install 200
```

You might need to scroll up a bit if hit by the issue.

LvsDark, 2023-01-16:
unfortunetly i fixed the issue  before i was able to get another screenshot, i just manually created the directory that was said was missing, i then attempted to install and it worked.

ravenclaw900, 2023-01-16:
`/opt/dietpi-dashboard/dietpi-dashboard` should actually be the executable file, so did you create the parent directory (`/opt/dietpi-dashboard`)? However, if the dashboard works and you can access it, I think that we can close this.

LvsDark, 2023-01-16:
Yes, created the parent directory, then the installation completed successfully, then i uninstalled it again, and reinstalled it again to see if i would get the same error, and it worked just fine. Thanks guys!

closed 2023-01-16 by LvsDark