MichaIng/DietPi #6036: HomeAssistant Update Failure "gcc"- DietPi or HA Problem ?

whyisthisbroken, 2023-01-06:
### Creating a bug report/issue

#### Required Information
Latest DietPi Beta

#### Additional Information (if applicable)
HomeAssistant

#### Steps to reproduce
Updating Home Assistant from 2022.12.8 -> 2023.1

#### Expected behaviour
Updating without Problems - like the last ones...

#### Actual behaviour

```
Building wheels for collected packages: ciso8601, home-assistant-bluetooth
  Building wheel for ciso8601 (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for ciso8601 (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-310
      creating build/lib.linux-aarch64-cpython-310/ciso8601
      copying ciso8601/__init__.pyi -> build/lib.linux-aarch64-cpython-310/ciso8601
      copying ciso8601/py.typed -> build/lib.linux-aarch64-cpython-310/ciso8601
      running build_ext
      building 'ciso8601' extension
      creating build/temp.linux-aarch64-cpython-310
      gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DCISO8601_VERSION=2.3.0 -DCISO8601_CACHING_ENABLED=1 -I/home/homeassistant/.pyenv/versions/3.10.9/include/python3.10 -c isocalendar.c -o build/temp.linux-aarch64-cpython-310/isocalendar.o
      error: command 'gcc' failed: No such file or directory
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for ciso8601
  Building wheel for home-assistant-bluetooth (pyproject.toml) ... done
  Created wheel for home-assistant-bluetooth: filename=home_assistant_bluetooth-1.9.2-cp310-cp310-manylinux_2_31_aarch64.whl size=73884 sha256=5d1af2320c4845a51cb2f89494b9ceca565ba656e2840fba44f1b885b781c23e
  Stored in directory: /tmp/pip-ephem-wheel-cache-zlm30c51/wheels/1b/45/f8/d15781d785e0954f95d7fa40a868d3af255d4a5dc6096e5777
Successfully built home-assistant-bluetooth
Failed to build ciso8601
ERROR: Could not build wheels for ciso8601, which is required to install pyproject.toml-based projects

```
#### Extra details

Is this a DietPi Problem or a HomeAssitant failure ?
Dont what to do with "gcc" - any ideas ?

Joulinar, 2023-01-06:
`gcc` is a compiler and needed to build different tools/apps. Depending on the used functions, it might be needed. You can install it quite easily

```
apt install gcc
```

whyisthisbroken, 2023-01-06:
Okay, the solution "apt install gcc" not working directly. it try it again, and gives me more errors...also a reboot doesnt help.
So i do a "dietpi-software reinstall 157" and the the problem is gone. 
dont know if missing gcc or something in pip from HA was the probem...

anyway... it works now :) thanks

closed 2023-01-06 by whyisthisbroken