MichaIng/DietPi #6041: ZRamSwap

ConsoleXXVII, 2023-01-06:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Have the possibility to create a compressed swap space in RAM to "increase" the free memory (actually allow to compress the RAM used by the less-used application to free memory, at the same time storing them in RAM to maximize the speed to make them usable again when needed). 
- Implement in a better way what the option AUTO_SETUP_SWAPFILE_LOCATION=zram in /boot/dietpi.txt  (automatic  base installation at first boot) allows to do.
- Make it available as an option in dietpi-drive_manager.

#### Describe the solution you'd like:

- Leverage zramswap tool in the zram-tools package.
The config file /etc/default/zramswap allows the user to set up the working environment  for zramswap. In the file can be selected the compress algorithm and the zramswap size (either as a size or a percentage of the RAM). A "zramswap restart" is needed to apply the changes.
-   The setup process looks similar to the one for /etc/dphys-swapfile. Maybe the scripts for configure that file can be used as a base.
- The default compression algorithm (lzo) isn't the best: lz4 has the best speed, zstd the best compression (in x86 systems I use lz4, I've never tested them in a SBC).
- As size I use 50% of the RAM but with some test can be increased (Fedora introduced it at 50%, now should be more; Google Chrome OS should be at 67%).
-  The package already handle the coexistence with swap file or swap partition if someone want to use both (first uses the zramswap after the swapfile).

#### Additional context

- ZRamSwap reduces the writes to the sdcard.
- ZRamSwap still work in RAM so the total available memory will be less than RAM+zramswap, but still more than only the RAM (compression does the trick).
- ZRamSwap should be an alternative option to the swapfile, NOT a replacement, NOT a default (need tests).
- Does exist an alternative to zramswap that compress the swap in a swap-file or swap-partition instead of in the RAM: it's call zswap. Here total memory will be RAM+zswap. I prefer zramswap because it does not tear the sdcard.

MichaIng, 2023-01-07:
You're right, I totally missed that the zram swap space cannot be selected from `dietpi-drive_manager`. You can however do:
```sh
/boot/dietpi/func/dietpi-set_swapfile 1 zram
```
`1` here creates a swap space in zram with a size of 50% of physical RAM.

But indeed this needs to be made available in our menus as well. Currently not sure whether `dietpi-drive_manager` is the right place, since it is more for tasks done on a specific drive, i.e. add/move swap file to a physical drive. Maybe we should instead re-enable the dedicated swap space menu in `dietpi-config`. It currently redirects to the drive manager, but since vast part of the code is done in the backend script, it wouldn't be much overhead to add it to the config script as well, with a way to select a drive or zram, and the size. There we can also add an algorithm option, if zram was selected.

appleguy, 2023-01-12:
I'd just run into this myself. While I found the set_swapfile options eventually (by reading the PR that added zram), it would've been great to find it in the very user-friendly menus.

This distro is amazing BTW, so keep up the phenomenal work! I would love to see zswap support too, e.g. allow retaining the possibility of paging out the compressed data.

MichaIng, 2023-01-12:
Since I needed to lookup again, to clarify what `zswap` is:
- It is a secondary compressed-memory cache for the actual swap device.
- So before data is swapped from RAM to disk, it is compressed to free up RAM space, and swapped to disk latest if required.
- It is a pure kernel feature which can be enabled via `echo 1 > /sys/module/zswap/parameters/enabled` and configured via other tunables in this dir.
- `zram`-based swapping on the other hand turns the swap space itself into a compressed RAM disk. In theory both behave similar until RAM space is exhausted despite data being compressed up to the configured limit/threshold. `zswap` at this point starts to swap to the disk swap space, while with `zram`-based swapping there is no disk swap space.

So indeed both features make sense depending on use case:
- `zswap` in case there are frequently small swap needs, covered by compression entirely, and rare large swap needs, which can be covered by a disk swap space only.
- `zram`-based swapping if there are smaller swap needs only, which can be covered by compression in RAM.
- Both do only make sense if the system runs close to memory exhaustion, so that regular smaller swap needs are given. If there are instead rare large swap needs only, e.g. due to daily/weekly/... maintenance tasks, which cannot be covered with compression in RAM only, it doesn't make much sense, in favour of a regular disk swap space.
- That said, compression and decompression costs CPU usage/time and hence potentially also slows down the process which triggers the swap need and that way increases the time of high RAM usage and the risk of process/task overlaps. So IMO generally such condition (running the system close to RAM limit) should be better avoided, when possible, instead of using any compressed RAM swapping.

appleguy, 2023-01-13:
+1, that sounds right and is a helpful overview for all readers to be on the same page!

zswap IMO is especially valuable for the RPi use case because of the slow performance of SD card storage. As such, given a quad-core CPU even in my old RPi 2 v1.1, it is likely that the compression overhead is worth it for the reduced I/O volume (at least for fast algorithms like lz4). In other cases, zstd is an excellent choice.

Overall these sound like features that fit the ethos of DietPi extremely well, enabling users to maximize the utility and service life of low-end hardware by providing convenient access to Linux features that can fit different use cases (good example with the occasional main't tasks pushing into swap only infrequently, while e.g. lz4 compression allows more common operations to never wait on / thrash the SD and stay in RAM)