MichaIng/DietPi #6043: Chromium install X Server - xinit unable to connect

therourke, 2023-01-07:
I have been trying to run Chromium on my Dietpi, Raspberry Pi 4. I want to show the browser on autostart on the HDMI output.

I keep getting the [same errors](https://imgur.com/a/zYdXVtA).

`xinit: unable to connect to x server : connection refused`

I have tried autobooting root and Dietpi, I have tried with default login and password and my own password, same outcome.

What am I doing wrong?

Joulinar, 2023-01-07:
How did you installed Chromium browser?

therourke, 2023-01-07:
Through dietpi dietpi-software menu

therourke, 2023-01-07:
Seems I can get it to run only in display modes: vc4-fkms-v3d or vc4-kms-v3d.

I would like to output to a CRT, so this mode's resolution is crazy high. Anyway to have the mode work and change the resolution from the default 1920x1080?

Joulinar, 2023-01-07:
You mean for the browser window or the whole screen?

therourke, 2023-01-07:
I want the whole screen to be lower resolution. I can get the chromium browser to fill 1920x1080 but I can't reduce that 1920x1080 to anything else. Tried tweaking dietpi.txt and config files. Nothing makes a difference.

Joulinar, 2023-01-07:
Did you tried `dietpi-config` display settings? Maybe a reboot is required to activate.

therourke, 2023-01-07:
Yes. As I said above, the only config that works in dietpi-config are the two v3d modes. Once I have changed options in config and dietpi.txt files I have rebooted each time. I can change resolution of chromium window, but nothing else makes any difference, even manually changing display modes.

MichaIng, 2023-01-08:
Easiest currently is to apply the wanted resolution first and add the dtoverlay manually afterwards.

MichaIng, 2023-02-12:
I'll mark this as closed. Issue is basically known, we need to redo the resolution menu for RPi to allow KMS with other resolutions than 1080p. As workaround, see my comment.

closed 2023-02-12 by MichaIng