MichaIng/DietPi #6044: Dietpi Update Failure  8.9.2 -> 8.12.1

cpsjones, 2023-01-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bullseye 1

- Kernel version | `uname -a`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi Zero W (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)

USB fitted mains adaptor

- SD card used | (EG: SanDisk ultra)

SanDisk ultra??

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
N/A

- Was the software title installed freshly or updated/migrated?
- N/A
- 
- Can this issue be replicated on a fresh installation of DietPi?
- N/A
- 

- Bug report ID | `echo $G_HW_UUID`

2d590dd7-b3c2-496c-b012-d1ebb2dfae20

#### Steps to reproduce

Run dietpi-update

#### Expected behaviour

Update without error

#### Actual behaviour

Errors as per bug report
#### Extra details

Tried to update as part of trying to work out why audio playing had suddenly failed (alsa etc.)

Joulinar, 2023-01-09:
Can you share any kind of error message?

MichaIng, 2023-01-10:
Pasting from bug report:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
Get:3 https://download.opensuse.org/repositories/home:/jcorporation/Raspbian_11  InRelease [1534 B]
Err:3 https://download.opensuse.org/repositories/home:/jcorporation/Raspbian_11  InRelease
  The following signatures were invalid: EXPKEYSIG AA09B8CCE895BD7D home:jcorporation OBS Project <home:jcorporation@build.opensuse.org>
Fetched 1534 B in 4s (375 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://download.opensuse.org/repositories/home:/jcorporation/Raspbian_11  InRelease: The following signatures were invalid: EXPKEYSIG AA09B8CCE895BD7D home:jcorporation OBS Project <home:jcorporation@build.opensuse.org>
W: Failed to fetch https://download.opensuse.org/repositories/home:/jcorporation/Raspbian_11/InRelease  The following signatures were invalid: EXPKEYSIG AA09B8CCE895BD7D home:jcorporation OBS Project <home:jcorporation@build.opensuse.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Same issue as reported here: https://github.com/jcorporation/myMPD/issues/947

Solve it with:
```sh
curl -sSfL 'https://download.opensuse.org/repositories/home:/jcorporation/Debian_11/Release.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-mympd.gpg --yes
```

We will update the key as pre-patch with upcoming DietPi release.

MichaIng, 2023-01-10:
Added pre-patch: https://github.com/MichaIng/DietPi/commit/f4b5752

closed 2023-01-10 by MichaIng