MichaIng/DietPi #6050: Open Beta v8.13 | Please help testing and hardening our upcoming release

MichaIng, 2023-01-10:
RC version | v8.13.0
---------- | -------
v8.13.0 PR | #6016
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2023-01-14**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.13
____
### Beta v8.13.0
_(2023-01-10)_

#### New images
- NanoPi R5S/R5C | Our NanoPi R5S image has been verified to work fine on NanoPi R5C as well, and the hardware model has been hence internally renamed to indicate this.

#### Enhancements
- General | Reduced log level for kernel logs to console to omit warnings. This matches to what the `quiet` kernel command-line parameter results in and helps to prevent messy display with DietPi-CloudShell, Pi-hole PADD and similar. Many thanks to `@Falconcrest` for reporting a related issue: https://dietpi.com/forum/t/pi-hole-padd-problem-under-dietpi-v8-12-1/15299
- DietPi-Software | myMPD: On fresh installs, plain HTTP is now disabled completely but myMPD is accessible via HTTPS on port 1333 only. The myMPD CA for the default self-signed certificate is added to the systems trusted store to allow MPD fetching web radio playlists from myMPD.
- DietPi-Software | vaultwarden: Updated to latest version 1.27.0. The update can be applied via reinstall: `dietpi-software reinstall 183`
- DietPi-Software | NoMachine: Updated to latest version 8.2.3. The update can be applied via reinstall: `dietpi-software reinstall 30`
- DietPi-Software | Transmission: The `cache-size-mb` setting is not set to anymore on fresh installs. It was set to 10% of RAM size, which is often too much for a pure download cache. It now defaults to 4 MiB instead. If you experience inconsistent download speeds on an existing install, you may edit or remove the setting in `/etc/transmission-daemon/settings.json`. Many thanks to @bbsixzz for bringing up this topic: https://github.com/MichaIng/DietPi/issues/5069
- DietPi-Software | UrBackup: The backup path can now be pre-configured with a new `SOFTWARE_URBACKUP_BACKUPPATH` setting in `dietpi.txt`.
- DietPi-Software | Amiberry: Updated to version 5.5.1, including LibSDL2 v2.26.2. The update can be applied via reinstall: `dietpi-software reinstall 108`

#### Bug fixes
- NanoPi M2/T2/Fire2 | Updated our image with a device tree for the actual NanoPi M2 and a fixed device tree to allow boot on NanoPi 2 Fire. Many thanks to @NewbieOrange for reporting this issue: https://github.com/MichaIng/DietPi/issues/5555
- PINE H64 | Resolved an issue where our recent PINE H64 images did not contain a kernel. Many thanks to @DremOSDeveloperTeam for reporting this issue: https://github.com/MichaIng/DietPi/issues/6031
- DietPi-LetsEncrypt | Resolved an error with Nginx when multiple `server_name` directives were present in `/etc/nginx/sites-available/default` due to e.g. a repeating `dietpi-letsencrypt`/Certbot execution or manually added vhosts. Many thanks to `@hemertje` for reporting this issue: https://dietpi.com/forum/t/nextcloud-problems-questions/15415/13
- DietPi-Software | Readarr: Resolved an issue where the installation failed because of a wrong download URL. Many thanks to @lambolighting for reporting this issue: https://github.com/MichaIng/DietPi/issues/5992
- DietPi-Software | Docker: Resolved an issue where the install failed if the kernel did not support nftables. Since `iptables` is installed as dependency along with the Docker package, it cannot be configured first to use the legacy API where needed. The Docker service is now masked before the package is installed, to prevent it from starting during the package install, before we can configure iptables. Many thanks to @tggjifdhn for reporting this issue: https://github.com/MichaIng/DietPi/issues/6013
- DietPi-Software | myMPD: Resolved an issue where APT updates failed since the myMPD APT repository key expired on 2023-01-01. Many thanks to @cpsjones for reporting this issue: https://github.com/MichaIng/DietPi/issues/6044

MichaIng, 2023-01-13:
#### Beta v8.13.1 has been merged: #6054

MichaIng, 2023-01-14:
#### Beta v8.13.2 has been released: #6061

MichaIng, 2023-01-14:
### DietPi v8.13 has been released: #6053
Many thanks to all testers ❤️!

closed 2023-01-14 by MichaIng

mrbluecoat, 2023-01-21:
A bit late to the party but just wanted to thank Team DietPi for a stellar NanoPi R5C image!