MichaIng/DietPi #6052: Usb HDD never spinning down

teodurini84, 2023-01-11:
Hello, I've connected a usb disk to my RPi 3 Model B (DietPi v8.12.1). I've just realized that the disk is never spinning down, problem that was not present with raspbian. am I the only one facing this problem?

MichaIng, 2023-01-11:
Whether and how spindown works highly depends on the drive, whether it supports APM and what it does with it.

Can you show the output of:
```sh
hdparm -B 127 /dev/sda
```
Adjust `/dev/sda` in case another drive is having the issue.

And does manual spindown work?
```sh
hdparm -S 120 /dev/sda
```

MichaIng, 2023-01-16:
Confusing, first it states "setting APM", then "not supported". Usually I see only one or the other, but not both. "SG_IO: bad/missing sense data" however looks like a related issue. Nasty that it shows up as well when trying to set the spindown time. The `SG_IO` error btw seems to mean that the drive didn't respond to the command: https://superuser.com/questions/1213715/hdparm-error-sg-io-bad-missing-sense-data

Probably this gives some hint:
```sh
hdparm -I /dev/sdb
```

teodurini84, 2023-01-29:
hello @MichaIng , sorry for the late reply

/dev/sdb:

ATA device, with non-removable media
Standards:
        Likely used: 1
Configuration:
        Logical         max     current
        cylinders       0       0
        heads           0       0
        sectors/track   0       0
        --
        Logical/Physical Sector size:           512 bytes
        device size with M = 1024*1024:           0 MBytes
        device size with M = 1000*1000:           0 MBytes
        cache/buffer size  = unknown
Capabilities:
        IORDY not likely
        Cannot perform double-word IO
        R/W multiple sector transfer: not supported
        DMA: not supported
        PIO: pio0

MichaIng, 2023-01-29:
Okay, no info at all about APM, standby/spindown capabilities. `hdparm -S 120 /dev/sdb` does not spin it down immediately, does it? Does one of the following work?
```sh
hdparm -y /dev/sdb
hdparm -Y /dev/sdb
```
See here another case where other tools were required for some drives which do not support standard tools: https://dietpi.com/forum/t/hdd-not-spinning-down-after-installing-another-one/15641

MichaIng, 2023-02-12:
Let me know if you want to further investigate. As a matter of fact, spindown via hdparm config just does not work with all drives. But there are other methods and tools as posted/linked above.

closed 2023-03-18 by MichaIng