MichaIng/DietPi #6055: md1 Raid array can not be seen after Dietpi update

ndg2806, 2023-01-12:
Hello Dietpi Community.
after Dietpi update Raid discs are not seen or accessible. here i run Nextcloud. now Nextcloud weblink is not working as well.
this has occured after trying to update Dietpi OS using "apt update", which eventually navigated to update from Buster to Bullseye.
during this process php was uninstalled. then i have reinstalled php.

i remember, names of discs as **sda, sdb** and **sdc**.

i see **/dev/md1**, and output of **/mnt/** as below;

[image]

[image]

output from "fdisk";
root@DietPi:~# **fdisk -l**
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 119.1 GiB, 127865454592 bytes, 249737216 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2fed7fee

Device         Boot  Start       End   Sectors   Size Id Type
/dev/mmcblk0p1        8192    532479    524288   256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 249737215 249204736 118.9G 83 Linux

....still scanning and command has not exited!

output from fsck;
root@DietPi:~# **fsck /dev/md1**
fsck from util-linux 2.33.1

....still scanning and command has not exited!

my worry is: did i loose data on discs? if not what can be done?
I am not Linux expert, but mere a USER. I can provide more information if someone can guide me through detailed navigation to find root cause of this issue, with set of commands.
Thank you in advance and Happy New Year 2023.

Cheers,
Deepak

Joulinar, 2023-01-12:
By design our update process can't update a system from Buster to Bullseye. This is something a user need to do manually. `apt update` is not a DietPi command. It's the standard Debian package manager. Probably the software used to build the raid got a version update? How did you create the raid?

Following should show all available drives

```
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

If possible, try to avoid doing screen prints. Try to copy content directly from SSH terminal.

As well can you check available services

```
dietpi-services status
```

ndg2806, 2023-01-12:
I do not remember exactly since it has been more than 1 - 1.5 years. but i do remember using "mdadm".

root@DietPi:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[ INFO ] DietPi-Services | redis-server          inactive (dead)
[ INFO ] DietPi-Services | mariadb               inactive (dead)
[ INFO ] DietPi-Services | php8.0-fpm            inactive (dead) since Wed 2023-01-11 20:38:53 GMT; 15h ago
[ INFO ] DietPi-Services | lighttpd              inactive (dead)
[ INFO ] DietPi-Services | cron                  inactive (dead)
[  OK  ] DietPi-Services | dropbear              active (running) since Wed 2023-01-11 20:33:52 GMT; 15h ago
[ INFO ] DietPi-Services | vncserver             inactive (dead)
[ INFO ] DietPi-Services | dietpi-vpn            inactive (dead)
[  OK  ] DietPi-Services | dietpi-ramlog         active (exited) since Wed 2023-01-11 20:33:52 GMT; 15h ago
[  OK  ] DietPi-Services | dietpi-preboot        active (exited) since Wed 2023-01-11 20:33:52 GMT; 15h ago
[  OK  ] DietPi-Services | dietpi-postboot       active (exited) since Wed 2023-01-11 20:34:03 GMT; 15h ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor   inactive (dead)

Joulinar, 2023-01-12:
I don't see any external disk attached at all. Are you sure your disks are online? What about kernel error messages

```
dmesg -l err,crit,alert,emerg
```

ndg2806, 2023-01-12:
Discs are mounted on RPi HAT.

root@DietPi:~# dmesg -l err,crit,alert,emerg
[    2.260533] systemd-journald[117]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    2.312464] systemd-journald[128]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    2.359483] systemd-journald[139]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    2.391780] systemd-journald[142]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    2.429246] systemd-journald[143]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    2.497544] systemd-fsck[137]: fsck terminated by signal PIPE.
[    3.261742] vc4_hdmi fef00700.hdmi: Could not register sound card: -517
[    3.446947] rpi-sense 1-0046: Read from reg 240 failed
[    4.906614] systemd-logind[377]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    5.200106] systemd-logind[420]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    5.233639] systemd-logind[422]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    5.265849] systemd-logind[423]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[    5.298049] systemd-logind[424]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   17.876496] systemd-logind[1024]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   17.910368] systemd-logind[1028]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   17.957557] systemd-logind[1034]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   18.006167] systemd-logind[1049]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   18.041250] systemd-logind[1054]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   20.318183] blk_update_request: I/O error, dev sda, sector 16 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 0
[   20.318194] md: super_written gets error=-5
[   20.318203] md/raid:md1: Disk failure on sda, disabling device.
               md/raid:md1: Operation continuing on 2 devices.
[   20.528208] blk_update_request: I/O error, dev sdb, sector 16 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 0
[   20.528216] md: super_written gets error=-5
[   20.768204] blk_update_request: I/O error, dev sdb, sector 264192 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   22.028176] blk_update_request: I/O error, dev sdc, sector 264192 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   22.048152] blk_update_request: I/O error, dev sdc, sector 264192 op 0x0:(READ) flags 0x4000 phys_seg 1 prio class 0
[   43.442633] systemd-logind[1629]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   43.445342] systemd[1]: Failed to start Login Service.
[   43.474086] systemd-logind[1630]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   43.505648] systemd-logind[1631]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   43.546256] systemd-logind[1634]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   43.577818] systemd-logind[1636]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   74.324526] systemd-logind[1929]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   74.327241] systemd[1]: Failed to start Login Service.
[   74.356751] systemd-logind[1930]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   74.388310] systemd-logind[1931]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   74.420092] systemd-logind[1932]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[   74.452346] systemd-logind[1933]: Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' failed at ../src/basic/time-util.c:55, function now(). Aborting.
[  197.940901] systemd-udevd[947]: Spawned process '/sbin/mdadm --incremental --export /dev/sdc --offroot /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1:1.0-scsi-0:0:0:0 /dev/disk/by-id/usb-WDC_WD10_JFCX-68N6GN0_1234567890123- [... cut ...]
[  489.448172] INFO: task mdadm:950 blocked for more than 368 seconds.
[  489.448181]       Tainted: G         C        5.10.63-v7l+ #1459
[  489.448188] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  489.448645] INFO: task mdadm:1253 blocked for more than 368 seconds.
[  489.448652]       Tainted: G         C        5.10.63-v7l+ #1459
[  489.448658] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[42463.940776] systemd[1]: Timed out waiting for device /dev/disk/by-uuid/50c5f3dc-d04a-4483-ae3d-e3f522654dda.

Joulinar, 2023-01-12:
there seems to be I/O errors on all your disk

```
[ 20.318183] blk_update_request: I/O error, dev sda, sector 16 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 0
[ 20.318194] md: super_written gets error=-5
[ 20.318203] md/raid:md1: Disk failure on sda, disabling device.
md/raid:md1: Operation continuing on 2 devices.
[ 20.528208] blk_update_request: I/O error, dev sdb, sector 16 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 0
[ 20.528216] md: super_written gets error=-5
[ 20.768204] blk_update_request: I/O error, dev sdb, sector 264192 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 22.028176] blk_update_request: I/O error, dev sdc, sector 264192 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 22.048152] blk_update_request: I/O error, dev sdc, sector 264192 op 0x0:(READ) flags 0x4000 phys_seg 1 prio class 0
```

mdadm is not able to get the raid up. 

```
68N6GN0_1234567890123-0:0' [950] timed out after 2min 59s, killing
[ 197.940965] systemd-udevd[153]: sdc: Worker [947] processing SEQNUM=1711 killed
[ 197.941994] systemd-udevd[153]: sdc: Worker [947] failed
[ 200.940905] systemd-udevd[1252]: Spawned process '/sbin/mdadm --detail --export /dev/md1' [1263] timed out after 2min 59s, killing
```

have a look/check if all disk correctly powered/conencted.

Joulinar, 2023-01-12:
did you upgrade your system manually from Stretch > Buster > Bullseye?

```
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Err:2 https://packages.sury.org/php stretch InRelease
```

ndg2806, 2023-01-12:
All discs are connected on HAT and working.

Rpi is mounted with Quad SATA HAT from radxa.
https://wiki.radxa.com/Dual_Quad_SATA_HAT

i did Firmware update using https://forum.radxa.com/t/quad-sata-hat-assembly-and-troubleshooting/3879

root@DietPi:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

then sudo apt-get install i2c-tools

output from sudo i2cdetect -y 1
root@DietPi:~# sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- 51 -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

ndg2806, 2023-01-12:
> did you upgrade your system manually from Stretch > Buster > Bullseye?
> 
> ```
> Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
> Err:2 https://packages.sury.org/php stretch InRelease
> ```
> 
> On actual DietPi, we switched away from `sury.org` PHP repository back to Debian default

is it possible to downgrade version to stretch? if yes let me know how?

ndg2806, 2023-01-12:
> did you upgrade your system manually from Stretch > Buster > Bullseye?
> 
> ```
> Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
> Err:2 https://packages.sury.org/php stretch InRelease
> ```
> 
> On actual DietPi, we switched away from `sury.org` PHP repository back to Debian default

yes, i did perform manual update, but only valid reason is error during "apt update"
Err:2 https://packages.sury.org/php stretch InRelease
  403  Forbidden [IP: 169.150.247.40 443]

MichaIng, 2023-01-12:
> is it possible to downgrade version to stretch?

No, distro downgrades are naturally not possible since packages of older distros cannot know which config/setup adjustments have been done for (packages of) newer distros. Only a backup would do, which is/was generally advised before any distro upgrade 😉.

You should go through the steps we collected in our upgrade guides:
- https://dietpi.com/docs/usage/
- https://dietpi.com/blog/?p=811#2.2.2-change-the-package-sources

Not sure what you already did, but obviously the Stretch suite of the sury PHP repo is still there, which gives the error, and you have PHP 8.0 installed from it, while Debian Bullseye natively ships PHP 7.4. So depending on what you did intentionally and what happened automatically (e.g. PHP meta packages pulling latest PHP like on rolling release distros), be careful on applying PHP related steps. E.g. if you need PHP 8.0, you need to keep the repo, but update the suite from "stretch" to "bullseye" to solve the error.
```sh
sed -i 's/stretch/bullseye/' /etc/apt/sources.list.d/dietpi-php.list
```
But if you go this path, be aware that this might lead to another upgrade to PHP 8.1, so that you need to adjust e.g. the Lighttpd config again (if you ever did it for PHP 8.0) to get it working.

Does the HAT's LCD work?

ndg2806, 2023-01-13:
> > is it possible to downgrade version to stretch?
> 
> No, distro downgrades are naturally not possible since packages of older distros cannot know which config/setup adjustments have been done for (packages of) newer distros. Only a backup would do, which is/was generally advised before any distro upgrade 😉.
> 
> You should go through the steps we collected in our upgrade guides:
> 
> * https://dietpi.com/docs/usage/
> * https://dietpi.com/blog/?p=811#2.2.2-change-the-package-sources
> 
> Not sure what you already did, but obviously the Stretch suite of the sury PHP repo is still there, which gives the error, and you have PHP 8.0 installed from it, while Debian Bullseye natively ships PHP 7.4. So depending on what you did intentionally and what happened automatically (e.g. PHP meta packages pulling latest PHP like on rolling release distros), be careful on applying PHP related steps. E.g. if you need PHP 8.0, you need to keep the repo, but update the suite from "stretch" to "bullseye" to solve the error.
> 
> ```shell
> sed -i 's/stretch/bullseye/' /etc/apt/sources.list.d/dietpi-php.list
> ```
> 
> But if you go this path, be aware that this might lead to another upgrade to PHP 8.1, so that you need to adjust e.g. the Lighttpd config again (if you ever did it for PHP 8.0) to get it working.
> 
> Does the HAT's LCD work?

HATs OLED display not working.
during initial boot all 3 LEDs flash, remains for 1 minute then gets off.

ndg2806, 2023-01-13:
yesterday spontaneously i did "apt upgrade", and stuck at 66%... see below

root@DietPi:~#  apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  bsdmainutils cpp-8 libapt-inst2.0 libapt-pkg5.0 libayatana-appindicator3-1 libcroco3
  libdbusmenu-glib4 libdbusmenu-gtk3-4 libdns-export1104 libegl1-mesa libevent-2.1-6 libfam0 libfl2
  libhiredis0.14 libhogweed4 libidn11 libindicator3-7 libip4tc0 libisc-export1100 libisl19
  libjson-c3 libldap-common libllvm9 libncursesw5 libnettle6 libperl5.28 libprocps7 libreadline7
  librsvg2-2 librsvg2-common libunique-1.0-0 libusb-0.1-4 multiarch-support ncal perl-modules-5.28
  php8.1-redis usb.ids
Use 'apt autoremove' to remove them.
The following NEW packages will be installed:
  bsdextrautils cpp-10 dropbear libatopology2 libbpf0 libctf-nobfd0 libctf0 libdeflate0
  libdns-export1110 libevent-2.1-7 libfdt1 libgdk-pixbuf-2.0-0 libgdk-pixbuf-xlib-2.0-0 libicu67
  libip4tc2 libisc-export1105 libisl23 libjson-c5 libllvm11 liblzf1 libmd0 libperl5.32 libprocps8
  libvpx6 libvulkan1 libyaml-0-2 libz3-4 logsave mailcap media-types mksh ncal pci.ids
  perl-modules-5.32 php8.1-cli php8.1-opcache php8.1-phpdbg php8.1-readline php8.2-cli php8.2-common
  php8.2-igbinary php8.2-opcache php8.2-phpdbg php8.2-readline php8.2-redis python3-smbus
  python3-wheel rng-tools-debian
The following packages have been kept back:
  cpp-8 galera-3 gcc-8-base libpolkit-agent-1-0 libpolkit-gobject-1-0 mariadb-server policykit-1
  raspberrypi-sys-mods
The following packages will be upgraded:
  accountsservice adwaita-icon-theme alsa-utils apt apt-utils base-files base-passwd bash
  bash-completion bc binutils binutils-arm-linux-gnueabihf binutils-common bsdmainutils bsdutils
  bzip2 console-setup console-setup-linux coreutils cpio cpp crda cron dash dbus dbus-x11
  dconf-editor dconf-gsettings-backend dconf-service debconf debianutils device-tree-compiler
  diffutils distro-info-data dmsetup dosfstools dpkg dropbear-bin dropbear-run e2fsprogs ethtool
  fake-hwclock fdisk findutils firefox-esr firmware-atheros firmware-brcm80211 firmware-iwlwifi
  firmware-linux-free firmware-misc-nonfree firmware-realtek flashrom fontconfig fontconfig-config
  fonts-dejavu-core gawk gettext-base git git-man glib-networking glib-networking-common
  glib-networking-services grep groff-base gsettings-desktop-schemas gtk-update-icon-cache gzip
  hdparm hostname htop ifupdown init init-system-helpers install-info iproute2 iputils-ping
  isc-dhcp-client iw jq kbd keyboard-configuration kmod libaccountsservice0 libacl1 libaio1
  libapparmor1 libargon2-1 libasound2 libasound2-data libassuan0 libatk-bridge2.0-0 libatk1.0-0
  libatk1.0-data libatomic1 libatspi2.0-0 libattr1 libaudit-common libaudit1 libavahi-client3
  libavahi-common-data libavahi-common3 libayatana-appindicator3-1 libayatana-ido3-0.4-0
  libayatana-indicator3-7 libbinutils libblas3 libblkid1 libbsd0 libbz2-1.0 libcairo-gobject2
  libcairo2 libcap-ng0 libcap2 libcap2-bin libcolord2 libcom-err2 libconfig-inifiles-perl libcroco3
  libcryptsetup12 libcups2 libdatrie1 libdb5.3 libdbi-perl libdbus-1-3 libdbus-glib-1-2
  libdbusmenu-glib4 libdbusmenu-gtk3-4 libdconf1 libdebconfclient0 libdevmapper1.02.1 libdrm-amdgpu1
  libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2 libedit2 libegl-mesa0 libegl1 libegl1-mesa
  libelf1 libepoxy0 liberror-perl libevdev2 libext2fs2 libfam0 libfdisk1 libffi6 libfl2
  libfontconfig1 libfontenc1 libfribidi0 libftdi1-2 libgbm1 libgcrypt20 libgdbm-compat4 libgdbm6
  libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgfortran5 libgl1 libgl1-mesa-dri libglapi-mesa
  libgles2 libglu1-mesa libglvnd0 libglx-mesa0 libglx0 libgmp10 libgomp1 libgpm2 libgraphite2-3
  libgssapi-krb5-2 libgtk-3-0 libgtk-3-common libgtk2.0-0 libgtk2.0-common libgudev-1.0-0
  libharfbuzz0b libhiredis0.14 libice6 libicu65 libidn11 libidn2-0 libinput-bin libinput10 libiw30
  libjpeg62-turbo libjq1 libjson-glib-1.0-0 libjson-glib-1.0-common libk5crypto3 libkeyutils1
  libkmod2 libkrb5-3 libkrb5support0 liblapack3 liblcms2-2 libldap-2.4-2 libldap-common
  liblightdm-gobject-1-0 libllvm9 liblocale-gettext-perl liblua5.1-0 liblz4-1 liblzma5 libmnl0
  libmpc3 libmpfr6 libmtdev1 libncurses6 libncursesw5 libncursesw6 libnewt0.52 libnghttp2-14
  libnpth0 libonig5 libpam-modules libpam-modules-bin libpam-runtime libpam-systemd libpam0g
  libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libparted2 libpci3 libpciaccess0 libpcre2-8-0
  libpcre3 libpcsclite1 libpipeline1 libpixman-1-0 libpng16-16 libpopt0 libproxy1v5 libpsl5
  libraspberrypi-bin libraspberrypi0 librest-0.7-0 librsvg2-2 librsvg2-common librtmp1
  libsamplerate0 libsasl2-2 libsasl2-modules-db libseccomp2 libsemanage-common libsemanage1
  libsensors-config libsensors5 libsepol1 libsigsegv2 libslang2 libsmartcols1 libsnappy1v5
  libsoup-gnome2.4-1 libsoup2.4-1 libsqlite3-0 libss2 libssh2-1 libssl1.1 libsystemd0
  libtext-charwidth-perl libtext-iconv-perl libtext-wrapi18n-perl libthai-data libthai0 libtiff5
  libtinfo5 libtinfo6 libtomcrypt1 libtommath1 libuchardet0 libudev1 libunique-1.0-0 libunistring2
  libunwind8 libusb-1.0-0 libutempter0 libuuid1 libvte-2.91-0 libvte-2.91-common libwacom-common
  libwacom2 libwayland-client0 libwayland-cursor0 libwayland-egl1 libwayland-server0 libwebp6
[... cut ...]
   22 root      20   0       0      0      0 S   0.0   0.0   0:00.45 ksoftirqd/2
   23 root      20   0       0      0      0 I   0.0   0.0   0:00.98 kworker/2:0-events
   25 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/3
   26 root      rt   0       0      0      0 S   0.0   0.0   0:00.05 migration/3
   27 root      20   0       0      0      0 S   0.0   0.0   0:00.39 ksoftirqd/3
   30 root      20   0       0      0      0 S   0.0   0.0   0:00.00 kdevtmpfs
   31 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 netns

Joulinar, 2023-01-13:
Your system is in the middle of a RPI kernel update. Definitely wait, otherwise you could break it.

ndg2806, 2023-01-13:
> Your system is in the middle of a RPI kernel update. Definitely wait, otherwise you could break it.
- but this is going on for last 10-12 hours
> 
> > HATs OLED display not working.
> > during initial boot all 3 LEDs flash, remains for 1 minute then gets off.
> 
> This doesn't sounds healthy.

MichaIng, 2023-01-13:
So the Bullseye upgrade was never finished. Did you go through the steps I linked to cleanup obsolete APT configs? Also I see (saw before the shortening) that PHP 8.1 and PHP 8.2 have been pulled alongside PHP 8.0 and probably other PHP versions from the sury repo, as feared. This should be cleaned up. I suggest to first assure that kernel and bootloader are successfully upgraded, then cleanup APT and finish the Bullseye upgrade. The webserver can be configured with PHP 7.4 afterwards via reinstalls:
```sh
apt update
apt install raspberrypi-bootloader raspberrypi-kernel
```
The above must finish with success before you continue, otherwise the system cannot boot anymore.
```
rm -f /etc/apt/sources.list.d/dietpi-php.list
rm -f /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm -f /etc/apt/preferences.d/dietpi-{php,openssl,xrdp}
rm -f /etc/mysql/mariadb.conf.d/97-dietpi.cnf
sed -i 's/ ui$//' /etc/apt/sources.list.d/raspi.list
rm -f /etc/apt/sources.list.d/dietpi-{php,wireguard}.list
rm -f /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm -f /etc/apt/preferences.d/dietpi-{php,openssl,xrdp,wireguard,kodi}
rm -f /etc/apt/sources.list.d/pivpn-bullseye.list
rm -f /etc/apt/sources.list.d/pivpn-bullseye-repo.list
rm -f /etc/apt/preferences.d/pivpn-limit-bullseye
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```
I might fail at some point if some package downgrades might need to be applied.

ndg2806, 2023-01-13:
> 

root@DietPi:~# apt update
Err:1 https://packages.sury.org/php stretch InRelease
  403  Forbidden [IP: 138.199.37.225 443]
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:3 https://packages.sury.org/php buster InRelease [6837 B]
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
E: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  403  Forbidden [IP: 138.199.37.225 443]
E: The repository 'https://packages.sury.org/php stretch InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease is not valid yet (invalid for another 15h 3min 5s). Updates for this repository will not be applied.
E: Release file for https://packages.sury.org/php/dists/buster/InRelease is not valid yet (invalid for another 21h 0min 26s). Updates for this repository will not be applied.
root@DietPi:~# apt install raspberrypi-bootloader raspberrypi-kernel
^Citing for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 1748 (apt)... 176s

because;

root@DietPi:~# ps -edaf | grep -i apt
root      1748     1  0 Jan11 ?        00:04:52 apt upgrade
root     16093 16061  0 13:32 pts/3    00:00:00 grep -i apt

Joulinar, 2023-01-13:
there are at laest 3 issues now

1. Your system time is out of sycn `(invalid for another 21h 0min 26s)`. Fix would be
```
/boot/dietpi/func/run_ntpd 1
```
2. You need to correct the source file as stated above by @MichaIng before trying again to update.
3. there is a stuck update process. Probably due to lost SSH connection. You could try to kill the process and run the update again.

MichaIng, 2023-01-13:
Seems to:
```sh
kill -9 1748
```
> You need to correct the source file as stated above by @MichaIng before trying again to update.

I think it's better to do this after kernel/bootloader have been upgraded, to it has a higher chance this succeeds quickly.

closed 2023-01-25 by ndg2806

Joulinar, 2023-01-25:
@ndg2806 as I can see, you closed the issue. Do you like to share any solution or what you have done?