MichaIng/DietPi #6056: Next cloud installation

jmta, 2023-01-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.12.1
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.15.84-v8+
- SBC model |RPi3
- Power supply used | 5V 2A
- SD card used | SSD over usb

#### Additional Information (if applicable)
- Software title | Nextcloud
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi?
Untested
- Bug report ID | e59e4ee3-d929-4d0e-b014-d5deb0795a19

#### Steps to reproduce
1. Use dietpi-software and select Nextcloud for installation 

#### Expected behaviour
NextCloud to be installed 

#### Actual behaviour
- rm -rf /car/www/nextcloud/data fails

Joulinar, 2023-01-12:
can you share the full error message? Actually, there seems to be an issue with the NC archive available on NC download server. But this would be different from your issue.

jmta, 2023-01-12:
The error is that the folder does not exists. I've added the output of ls in the parent folder which shows no folder there. 

Let me know if there is anything specific you are looking for. I'll dig it out

Joulinar, 2023-01-12:
if you it by the issue, you should see the blue error screen. There you can select an option to change command. Do this and replace the whole command with `true`.

jmta, 2023-01-12:
_ [FAILED] File does not exist or cannot be written to │
│ by current user                                      │
│                                                      │
│ Please verify the existence of the file $3           │
│         /var/www/nextcloud/config/config.php         │
│                                                      │
│ Retry with proper permissions or apply the setting   │
│ manually:                                            │
│         'mysql.utf8mb4' => true,       _

MichaIng, 2023-01-12:
Can you please share the output of the whole installation process until now? There must have been an issue with the previous `ncc maintenance:install` step already, if neither the `data` dir, nor the `config.php` do exist.

jmta, 2023-01-12:
I've dropped true in as a replacement for the further issue and the install completes (with further warnings). I'll see how things go and see if I can search for the config file myself to add the options

Joulinar, 2023-01-12:
your NC is working now? You can login into web interface?

jmta, 2023-01-12:
Step: Installing Nextcloud: File sync, sharing and collaboration platform

[ INFO ] DietPi-Software | Existing Nextcloud installation found, will NOT overwrite...
[ INFO ] DietPi-Software | Installing required PHP modules
[ INFO ] DietPi-Software | APT install for: php7.4-intl, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
php7.4-intl is already the newest version (7.4.33-1+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[  OK  ] DietPi-Software | APT install for: php7.4-intl
[ INFO ] DietPi-Software | Enabling required PHP modules
[  OK  ] DietPi-Software | phpenmod ctype curl dom gd intl mbstring pdo_mysql posix simplexml xmlreader xmlwriter zip fileinfo opcache apcu redis exif json
[ INFO ] DietPi-Software | Apply PHP override settings for Nextcloud.
[  OK  ] DietPi-Software | phpenmod dietpi-nextcloud
[ INFO ] DietPi-Software | Lighttpd webserver found, enabling Nextcloud specific configuration.
[  OK  ] DietPi-Software | Desired setting in /etc/lighttpd/lighttpd.conf was already set:      "mod_access",
[  OK  ] DietPi-Software | Desired setting in /etc/lighttpd/lighttpd.conf was already set:      "mod_setenv",
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.conf/dps_114/lighttpd.nextcloud.conf -o lighttpd.nextcloud.conf
[ INFO ] DietPi-Software | Updating file: /etc/lighttpd/conf-available/99-dietpi-nextcloud.conf.dietpi-new
[  OK  ] DietPi-Software | mv lighttpd.nextcloud.conf /etc/lighttpd/conf-available/99-dietpi-nextcloud.conf.dietpi-new
[  OK  ] DietPi-Software | lighty-enable-mod rewrite dietpi-nextcloud
[  OK  ] DietPi-Software | systemctl restart mariadb
[  OK  ] DietPi-Software | systemctl restart redis-server
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/nextcloud_data

 on *.* to tmp_root@localhost identified by 'H2G6BOfT2RlsJmUFpVcriym9KBpwCv' with grant option;
[  OK  ] DietPi-Software | Nextcloud ncc install
[  OK  ] DietPi-Software | mysql -e drop user tmp_root@localhost;
[FAILED] DietPi-Software | rm -R /var/www/nextcloud/data
[ INFO ] DietPi-Software | Executing alternative command: true
[  OK  ] Alternative command execution | Completed
grep: /var/www/nextcloud/config/config.php: No such file or directory
sed: can't read /var/www/nextcloud/config/config.php: No such file or directory
[FAILED] DietPi-Software | sed -i s|'http://localhost'|'http://localhost/nextcloud'| /var/www/nextcloud/config/config.php
[ INFO ] DietPi-Software | Executing alternative command: sed -i s|'http:/
sed: -e expression #1, char 9: unterminated `s' command
[FAILED] Alternative command execution | Exited with error
Press any key to return to error handler menu...
[ INFO ] DietPi-Software | Executing alternative command: true
[  OK  ] Alternative command execution | Completed
[ INFO ] DietPi-Software | Enabling Redis for transactional file locking.
[  OK  ] DietPi-Software | Current setting in /etc/redis/redis.conf will be preserved: unixsocket /run/redis/redis-server.sock
[  OK  ] DietPi-Software | Desired setting in /etc/redis/redis.conf was already set: unixsocketperm 770
[  OK  ] DietPi-Software | usermod -aG redis www-data
[  OK  ] DietPi-Software | systemctl restart redis-server
Composer autoloader not found, unable to continue. Check the folder "3rdparty". Running "git submodule update --init" will initialize the git submodule that handles the subfolder "3rdparty".Composer autoloader not found, unable to continue. Check the folder "3rdparty". Running "git submodule update --init" will initialize the git submodule that handles the subfolder "3rdparty".Composer autoloader not found, unable to continue. Check the folder "3rdparty". Running "git submodule update --init" will initialize the git submodule that handles the subfolder "3rdparty".Composer autoloader not found, unable to continue. Check the folder "3rdparty". Running "git submodule update --init" will initialize the git submodule that handles the subfolder "3rdparty".Composer autoloader not found, unable to continue. Check the folder "3rdparty". Running "git submodule update --init" will initialize th

jmta, 2023-01-12:
I just grabbed everything from specifically the nc installation. Ncc install appears to pass fine

MichaIng, 2023-01-12:
> [ INFO ] DietPi-Software | Existing Nextcloud installation found, will NOT overwrite...

So this was no fresh installation?

Joulinar, 2023-01-12:
was there already an attempted before to install NC? And did it fail as well? Because there is a step missing to download NC archive. Means there was a install detected alredacy.

jmta, 2023-01-12:
Ah yes there was a previous install which failed. I assumed fresh was no successful previous install, and updating to the latest DietPi moved me further along.

For reference the output of ncc install 

dietpi@DietPi:~$ ncc install
Composer autoloader not found, unable to continue. Check the folder "3rdparty". Running "git submodule update --init" will initialize the git submodule that handles the subfolder "3rdparty".

Joulinar, 2023-01-12:
as stated above, currently NC has some issues at their download server. It seems the archive is corrupted. This caused the first install to fail and now your issues afterwards on further tries.

jmta, 2023-01-12:
Will try again in the morning, is there something additional I should do to clear any remnants from this issue?

MichaIng, 2023-01-12:
Ah yes makes sense. My unrelated install attempt a few minutes before your post also extracted parts of the archive until the CRC error occurred. With this incomplete archive the `ncc install` cannot succeed.

So I'd say
```sh
rm -R /var/www/nextcloud
```
And let's wait until tomorrow, or some hours depending on your timezone, before trying again. Hopefully the faulty download has been fixed until then.

Joulinar, 2023-01-12:
reported upstream https://github.com/nextcloud/server/issues/36128

jmta, 2023-01-13:
Issue now resolved

closed 2023-01-13 by jmta