MichaIng/DietPi #6058: [Feature request] Simple Git over SSH server

thanhminhmr, 2023-01-13:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- Problem: I want to backup/mirror Git repositories from GitHub/GitLab/Bitbucket or similar services. I also want to host a few small personal repositories from my very-low-end SBC as a mirror upstream.

#### Describe the solution you'd like:
- A DietPi style script with menu to: 
  + Setup a Git over SSH server.
  + Create/Clone repositories.
  + Setup cron jobs to sync with upstream repositories.

#### Describe alternatives you've considered:
- Personally, I'm currently use Gitea to do so. It works perfectly, but Gitea was overkill for the job and also use >70MiB (sometimes hovering around 100MiB) on my 512MiB Pi Zero 2 W.

#### Additional context
- [This guide here](https://www.inmotionhosting.com/support/website/git/git-server/) shows how to set it up (but I failed to follow it correctly and had to restore the Pi from a backup).

MichaIng, 2023-01-14:
Thanks for your suggestion.

The issue following the guide probably was the `ssh-copy-id` as the `openssh-client` is not pre-installed on DietPi?`

Since following this guide requires to do steps on server and on client(s) side, if is not so helpful to add a script/tool for this. And how things are done on each side, highly depends on the OS/system setup. Also all the Git steps themselves depend on what you actually want to do, whether you want to clone a repo, push one or other.

The major part here is to setup key authentication, which is actually optional for convenience and security, the latter only if you disable/lock the password for the `git` user, or password authentication in general on the SSH server, which the guide both does not handle. And the statement "Remember that password authentication is not allowed on your server by default." is wrong for all OpenSSH and Dropbear server implementations I know, and at least our VPS when we did set it up some years ago. It is likely true for "inmotion hosting" which wrote this guide, for the local console login, but for SSH it depends on the OS you install, probably true for the images they offer.

I think it is better served if we offered two things:
- Proper user management GUI, allowing to create login users and system/service users, enable/disable password login, adding/removing SSH keys, managing `sudo` permissions and such in the long-term.
- SSH server management GUI, allowing to enable/disable password-login for all users or for root user only, basically a small GUI for the `SOFTWARE_DISABLE_SSH_PASSWORD_LOGINS` `dietpi.txt` setting, which applies this directly to the installed SSH server (the setting applies on SSH server installs/reinstalls and first boot only).

We could then add a HowTo to our docs somewhere (not sure where it fits best) about your particular use case: Creating a system user (without password in the first place, blocking shell login attempts, which are AFAIK not required for remote Git operations) on the server, adding SSH keys, and example commands to push or pull from the client(s).

thanhminhmr, 2023-01-15:
After reading your answer and some more digging, I realize that Git over SSH works exactly like the name suggest: do Git things over SSH (duh!). So for now I settled with a simple setup that only require Git and a SSH server.

1. To mirror a project from the internet to my DietPi, just SSH in then `git clone --mirror`.
2. To update those mirrors, create a `crontab` to `git remote update` them.
3. To create a new repository in my DietPi, just SSH in, create a directory (for example at `/home/dietpi/git/new-project`) and `git init` inside it.
4. To clone a repository from my DietPi to my computer, just do `git clone dietpi@dietpi.local:/home/dietpi/git/new-project`.

Maybe a script can automate the setup for a "proper" server like in [this guide](https://git-scm.com/book/en/v2/Git-on-the-Server-Setting-Up-the-Server), but maybe it is not worth the effort.

MichaIng, 2023-01-15:
Yes exactly. The purpose/benefit is that a Git client can connect to a Git server via SSH, so the repository does not need to be accessible via HTTP.

There is one pitfall when connecting from a DietPi system via SSH to a Git server: Since no `openssh-client` is installed OOTB, no `ssh` command is available. One can do `GIT_SSH=$(which dbclient)` to tell Git to use the Dropbear SSH client instead. Or one can install the `openssh-client` package, of course.