MichaIng/DietPi #6060: Reinstalling Homer fails

JappeHallunken, 2023-01-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.13.0 beta
- Distro version | bullseye
- Kernel version | 5.15.84-v8+ #1613
- SBC model |  RPi3b+

#### Steps to reproduce
If I do `dietpi-software reinstall 205` (reinstall Homer) it throws an error:
`rm: cannot remove '/mnt/dietpi_userdata/homer_backup': No such file or directory`

I'm not complete sure what is going on here, according to the docs there should be an backup in this location, which is not true in my case.

Joulinar, 2023-01-13:
I will have a look to our code later the evening.

MichaIng, 2023-01-13:
Here is the related code:
```sh
			# Backup existing instance
			if [[ -d '/var/www/homer' ]]
			then
				G_DIETPI-NOTIFY 2 'Existing Homer instance found, backing up to /mnt/dietpi_userdata/homer_backup ...'
				[[ -d '/mnt/dietpi_userdata/homer_backup' ]] || G_EXEC rm -R /mnt/dietpi_userdata/homer_backup
				G_EXEC cp -a /var/www/homer /mnt/dietpi_userdata/homer_backup
				G_DIETPI-NOTIFY 2 'Removing old assets with hash in file name ...'
				G_EXEC rm -Rf /var/www/homer/{workbox-*.js,resources}
			fi
```
- `||` needs to be `&&`, so that the dir is tried to be removed if it exists, and not if it does not exist 😄.

MichaIng, 2023-01-13:
Fixed with: https://github.com/MichaIng/DietPi/commit/756c774

closed 2023-01-13 by JappeHallunken