MichaIng/DietPi #6063: DietPi-Drive_Manager: Ability to format logical volumes (lvm) resp. RAID disks

StephanStS, 2023-01-14:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
In case of dealing with LVMs (via package lvm2) or RAIDs (via package mdadm), dietpi-drive_manager cannot be used to format the whole LVM/RAID, because it wipes the disk and generates a new GPT partition table.
With this, previous `lvmcreate ...` or `mdadm create ...` will be destroyed.

A sample screenshot in case of a RAID is:
[image]
RAID /dev/md0 consists of /dev/sda .. /dev/sdd.

A sample screenshot in case of an LVM is:
[image]
LVM /dev/mapper/vg0-lvm1 consists of /dev/sda1, /dev/sda2.

Remark: In case of a RAID, the raid disks need not to have partition tables like shown in the example above. In case of an LVM partitions are necessary to build a volume group.

#### Describe the solution you'd like:
DietPi-Drive_Manager detects these type of drives and skips generating the partition table. 
I assume it is an issue in lines 582ff within dietpi-drive_manager.

The solution would help to use the comfort of the dietpi-drive_manager in case of RAIDs/LVMs.

#### Describe alternatives you've considered:
Actually, a format of such disk groups must be done outside of dietpi-drive_manager "by hand", i.e. e.g. `mkfs.ext4 /dev/md0`.

MichaIng, 2023-01-15:
Many thanks for your suggestion.

Partly discussed here: #2995
But the basis is support in `dietpi-drive_manager` indeed. It's great that the "filesystem type" is shown as "linux_raid_member" resp. "LVM2_member", which can be used to identify these member disks and show e.g. different info and a warning when selecting them.

For the RAID/LVM itself: Formatting the "partition" instead of the whole "drive" works, right?

I think as a first step, we should simply remove the option for formatting the "drive" if an LVM or RAID is detected. Long-term, of course it would be nice to have LVM and RAID management included, which allows to pick drives to create an VG>LV from, respectively a RAID of chosen type. But that is a large task. For now we should just assure that the drive manager cannot destroy existing LVs and RAIDs, respectively warns users when they e.g. choose to format an LVM/RAID member, and show info about members and LVs/RAIDs properly.

StephanStS, 2023-01-15:
> For the RAID/LVM itself: Formatting the "partition" instead of the whole "drive" works, right?

Unfortunately, not: The RAID/LVM format menue entry does not give the option of formatting a partition:
[image]
It seems that dietpi-drive_manager interpretes /dev/md0 as a disk and not as a partition.

That would be a first step in combination with a warning about the RAID/LVM destroying if formatting the whole disk.

MichaIng, 2023-01-15:
> It seems that dietpi-drive_manager interpretes /dev/md0 as a disk and not as a partition.

Ah yes this makes sense, since it does not match a known naming pattern with partition.