MichaIng/DietPi #6065: Docker and (or) Portainer Installation failed

omavoss, 2023-01-15:
#### Details:
- Date           | Sun Jan 15 22:36:40 CET 2023
- Bug report     | 3b3e8746-5a2b-4e97-b0f4-b13bd8640c2b
- DietPi version | v8.13.2 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RasPi4 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `docker volume create portainer_data`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
```
Please help, thank you.

Joulinar, 2023-01-15:
Docker is running fine?

```
journalctl -u docker.service
systemctl status docker.service
```

MichaIng, 2023-01-15:
Might be related to #6014.
The service was stopped before as well, but the socket unit was active. Probably, if the service is masked before install, the socket unit isn't enabled either 🤔.

Joulinar, 2023-01-15:
yes, you're right. The socket unit is not running.

```
root@DietPi4:~# systemctl status docker.socket
● docker.socket - Docker Socket for the API
     Loaded: loaded (/lib/systemd/system/docker.socket; enabled; vendor preset: enabled)
     Active: inactive (dead)
   Triggers: ● docker.service
     Listen: /run/docker.sock (Stream)
root@DietPi4:~#

```

Joulinar, 2023-01-15:
workaround would be following. Needs to be started on a sub shell. 

```
systemctl start docker.socket
```

MichaIng, 2023-01-15:
Verified and fixed with: https://github.com/MichaIng/DietPi/commit/e9cf357

We can ship a live patch as well.

MichaIng, 2023-01-15:
Live patch merged: #6066

closed 2023-01-15 by MichaIng