MichaIng/DietPi #6067: IPv4 dropped when enabling IPv6 as well

mews-se, 2023-01-15:
- DietPi version | every one I can remember
- SBC model | Different VM:s

#### Additional Information (if applicable)
When IPv6 is included in network config after a couple of days online it seems like it reverts to it being the priority, and dropping the IPv4 address. When trying to restart networking the fail message "failed to bring up eth0" appears.

#### Steps to reproduce
Select IPv6 if applicable, don't change anything manually. Wait. After a while the box will seem to be "offline" since it only has a IPv6 address

#### Expected behaviour
Both addresses should run in parallel

#### Actual behaviour
The IPv4 is lost until IPv6 is disabled again.

Joulinar, 2023-01-16:
Are you using DHCP? What is the lease time for DHCP request? Does your system loose IPv4 connection after same amount of time? 

Personally, I run all my system IPv4/IPv6 and never experienced an issue with loosing IPv4.

mews-se, 2023-01-16:
No I use static IP for both 4 and 6, which makes it even more interesting. I have only noticed this on different kind of VPS:es.

I will spin up another one to try and get the exact error message!

MichaIng, 2023-01-16:
How did you configure your network at all (since `dietpi-config` cannot be used to configure a static IPv6 address)? Obviously there is something conflicting and altering the interface state.

In plain DietPi, there is no mechanism which could remove an active IPv4 address from an interface. Either you configure a static IP address, which is applied once at boot via `ifup` and never touched again, or you use DHCP, in which case the `isc-dhcp-client`/`dhclient` regularly requests an address/renewal from the DHCP server/router. If no DHCP server is available at all, you wouldn't get any address right from the start. If it is not reachable after you got an address once, the old address would stay in place. A removal can be done only by other tools, which are not pre-installed on DietPi.

mews-se, 2023-01-16:
> How did you configure your network at all (since `dietpi-config` cannot be used to configure a static IPv6 address)? Obviously there is something conflicting and altering the interface state.
> 

Thanks, I was just about to ask how it is supposed to be configured (since, as you say, there is no way to use dietpi-config for IPv6 except turn it off or on).

I configured the 4 as static, and didn't touch anything for the 6 (which makes it DHCP?) And it's ifup that creates the error message for the 4.

Could it be that by not manually doing anything, the static 4 also reverts to DHCP?

Joulinar, 2023-01-16:
no, if you configure a STATIC IP it should stay like this. IPv6 is autoconfigured by the system itself using SLAAC. Something we don't influence from our tooling.

MichaIng, 2023-01-16:
Statically configuring IPv6 is generally difficult, since the prefix must match the public prefix for remote hosts to be accessible via IPv6. So if you do not have a static public IPv6 prefix given by your ISP, this is not gonna work.

DHCPv6 does exist, but isn't commonly used, often confused with SLAAC and not supported by all routers.

IPv6 is commonly setup via SLAAC only, in which case the system assigns a local-link address (LLA) based on the adapters MAC address to be able to communicate with the router. The router then (and regularly) sends router advertisements (RA) which contain a global unicast address (GUA) based on the public IPv6 prefix of the router, for communicating with remote hosts, a gateway, DNS and optionally a unique local address (ULA) for LAN-internal communication, which is however usually (recommended to be) disabled. This however all does not interfere with IPv4 at all, which is handled completely independent.

> And it's ifup that creates the error message for the 4.

Where does this error show up? It should run only once at boot, where you can see logs via `journalctl -u ifup@eth0`. If this fails, you wouldn't have any IPv4 address right from the start, while your said it drops after a couple of days. Or is it when you run it manually? That would be expected if the interface as been at least partly configured already, you'd then need to use `ifup --force eth0`.

mews-se, 2023-01-16:
It shows up when trying to restart networking. 

When logging in via console i noticed with 'ip addr' that only the 6 address was available and that the system couldnt connect to the outside or resolve any dns. Via dietpi-config trying to restart networking this error message pops up. Of course I forgot to copy it, but I have spun up a new vps with the same config to try to replicate it

mews-se, 2023-01-16:
Ok here's something interesting...

So the dietpi-config for network adapter is as following:
[image]

and 'ip addr'
[image]

Could the lifetime set to forever for the 6 but not for the 4 be the culprit, since it is static - it shouldn't need to try to renew.
[image]

MichaIng, 2023-01-16:
There seems to be a conflicting interface configuration since even that static IP is set, `dhclient` still runs and consequently fails. Can you show the output of:
```sh
for i in /etc/network/interfaces{,.d/*}; do echo "$i:"; cat "$i"; done
```

mews-se, 2023-01-16:
`/etc/network/interfaces:
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 65.108.153.244
netmask 255.255.255.255
gateway 172.31.1.1
#dns-nameservers 1.1.1.1 1.0.0.1

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 185.12.64.1 185.12.64.2
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
/etc/network/interfaces.d/50-cloud-init:
# This file is generated from information provided by the datasource.  Changes
# to it will not persist across an instance reboot.  To disable cloud-init's
# network configuration capabilities, write a file
# /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg with the following:
# network: {config: disabled}
auto lo
iface lo inet loopback

auto eth0
iface eth0 inet dhcp

# control-alias eth0
iface eth0 inet6 static
    address 2a01:4f9:c012:4c6f::1/64
    dns-nameservers 2a01:4ff:ff00::add:2 2a01:4ff:ff00::add:1
    gateway fe80::1
`

MichaIng, 2023-01-16:
Your VPS provider shipped the image with `/etc/network/interfaces.d/50-cloud-init`, which enables DHCP for IPv4 and a static IPv6 address (which is then obviously fine/correct in this case). Remove these parts of that file and make it persistent according to the comment:
```
auto lo
iface lo inet loopback

auto eth0
iface eth0 inet dhcp
```
```sh
echo 'network: {config: disabled}' > /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg
```
Or you use the network capabilities of your VPS GUI and remove `/etc/network/interfaces` + avoid using `dietpi-config` for network setup instead.

Joulinar, 2023-01-16:
```
/etc/network/interfaces.d/50-cloud-init
```

This one is causing issues. It configure DHCP in `eth0` interface. As well it set STATIC IP for IPv6.

mews-se, 2023-01-16:
Of course, don't know how I actually missed that one.........

Well, as always - thanks for your quick help!

Joulinar, 2023-01-16:
I guess we can close this one for now?

mews-se, 2023-02-16:
Sorry to reopen this, but I'm not that Github-savvy :)

Is it possible to move this to a feature request since a lot of VPS-providers do this config.

Also I would like to add to be able to choose between Dropbear and OpenSSH at the start of the install (like choosing bullseye, bookwork and so forth). Since my guess is that most users prefer OpenSSH?

Joulinar, 2023-02-16:
pls open a new issue for feature request. 

OpenSSH could be selected during initial installation or via `dietpi.txt` before first boot.

mews-se, 2023-02-16:
Apologies! I will add the ssh-thing to feature requests since I find myself almost always "converting" regular debian to dietpi.

re: the above original problem. that was what i had in mind, to keep the discussion and solution linked to the feature request

Joulinar, 2023-02-16:
we could wait on @MichaIng and his opinion. 

But to be true, your VPS configuration is quite dedicated to your provider. Other might have other ways to configure network. Means, quite hard for us to support all variants.

MichaIng, 2023-02-18:
We had a [cloud-init](https://github.com/canonical/cloud-init) config in place as well on our VPS, so indeed it is not so uncommon when starting with an image from the VPS provider. I just checked back, and without the [cloud-init package](https://packages.debian.org/en/bullseye/cloud-init) installed, the config shouldn't be recreated, hence creating `/etc/cloud/cloud.cfg.d/99-disable-network-config.cfg` shouldn't be necessary. This change should do it: https://github.com/MichaIng/DietPi/commit/604ae6f

closed 2023-02-19 by MichaIng